e2nIEE/pandapower #113: Power flow initialization

lthurner, 2018-06-25:
Continuation from #110:

- Allow to define a flat start voltage magnitude as parameter in runpp
- Either:
         - Helper function(s) to create res_bus for initialization of result vector with init="results" or
         - Possibility to directly define initialization vector
- Is graph based initialization function implemented by @WinfriedL suited as a toolbox function?

lthurner, 2018-06-25:
Another suggestion that came up: does it make sense to initialize the voltage magnitude with the slack voltage instead of 1.0 pu. Either in init="auto" or providing a new option? Obviously the question is how to handle grids with multiple ext_grids in gens, a simple solution would be to just use the average of all voltage controlled elements? Would of course still be a very rough estimate, but might be better than just using 1.0pu?

lthurner, 2018-07-03:
I have introduced a parameter vm_start_pu into the runpp function. The parameter can be set to a float (e.g. 1.0) or an iterable of values with the length of net.bus to set an individual value for each bus. It can also be set to "auto", where the mean value of all ext_grids / gens are then considered as the starting value at all buses. The different initialization methods can lead to fewer newton-raphson steps as shown in the [test_vm_start_pu](https://github.com/lthurner/pandapower/blob/develop/pandapower/test/loadflow/test_runpp.py#L756).

When I implemented this I realised that what would actually be coherent would be to seperately initialize voltage angle and voltage magnitude, because right now the init parameters can be set to different values that sometimes affect one and sometimes both of these values. So my suggestion would be to deprecate the init parameter and instead introduce two parameters:

vm_start_pu:
- "result": voltage magnitude vector is taken from result table
- "auto": all buses are initialized with the mean value of all voltage controlled elements in the grid
- a float with which all voltage magnitudes are initialized 
- an iterable with a voltage magnitude value for each bus

vm_start_degree:
- "result": voltage angle vector is taken from result table
- "dc":  voltage angles are initialized from DC power flow
- "auto": voltage angles are initialized from DC power flow if angles are calculated or as 0 otherwise
- a float with which all voltage angles are initialized 
- an iterable with a voltage angle value for each bus

This would give complete control over the initialization of each value. Should also be pretty straightforward to implement, but I didn't do that yet as it would change default behaviour and break code, so we should discuss this first.

WinfriedL, 2018-07-04:
Just my thoughts on this matter:

Currently I am using a faked res_bus table for initialization . After some initial (probably unrelated) problems, that was relativly straightforward. So for me all of the above is definitly a 'nice-to-have', but currently not a 'must-have'. 

vm_start_pu:

> "auto": all buses are initialized with the mean value of all voltage controlled elements in the grid

I am very interested to test that out. From a quick look at the code it seems not completly suitable for my cases, because I have regularly several disconnected subnets in one powerflow. So the bus voltages should be initialized with the mean voltage magnitude per subnet.  That would require the topological toolbox and maybe too much for just an addition in run.py

> a float with which all voltage magnitudes are initialized

That would be a great addition. For that kind of initialization creating a full res_bus table is (while not difficult) more complex than needed.

> an iterable with a voltage magnitude value for each bus

That would be a little bit more convenient than creating a res_bus table. And from a user point of view clearer than running the powerflow with init=results even if there are no results yet. 

vm_start_degree:

>a float with which all voltage angles are initialized

Would that make sense? The only relevant part for the powerflow are the differences in voltage angles, so setting all to the same values should be equivalent to setting everything to zero? Or should the slack angle stay at `va_degree` and only the other buses should be initialized with the constant float?
I am not sure if there is a usecase for such a feature, maybe if one knows beforehand that there is a large angle difference between the slack bus and the rest of the network?

>an iterable with a voltage angle value for each bus

As  for vm_start_pu a bit more convient. I myself have at the moment no usecase, since the only way I know to get 'better' initial angles is by dc loadflow. But that is no reason not to have it. If this would be implemented for vm_start_pu, it is probably better to have it for vm_start_degree as well.

lthurner, 2018-07-09:
> Would that make sense? The only relevant part for the powerflow are the differences in voltage angles, so setting all to the same values should be equivalent to setting everything to zero? Or should [...]

Yes only the difference is relevant for the power flow, but if the end result is somewhere at 100° and the flat start initializes at 0° the NR-algorithm still starts at 0° and has to iterate to 100°. Starting from 100° might allow faster convergence. No idea if this really makes a difference in any case, but as you say there is no reason not to have it and give the user full control.

Further suggestions from the developers meeting today:

- parameter names: "init_vm_pu" and "init_va_degree" to keep "init" in the name for consistency
- consider initializing angle to minimum or mean value of slack angles instead of zero?
- for smooth transission and not breaking code: capture "init" in **kwargs, set the new parameters so that it defaults to the old behaviour and give a deprecation warning

lthurner, 2018-07-18:
I introduced init_vm_pu and init_va_degree now as discussed. init is still getting processed as before, an error is raised when both init and init_vm_pu or init_va_degree are set. This way
```
pp.runpp(net, init="results") 
```
is still possible, but it is also possible to do something like
```
 pp.runpp(net, init_va_degree="dc", init_vm_pu="results")
```

lthurner, 2018-09-10:
I moved init_va_degree and init_vm_pu to the kwargs as discussed in #122, so that init is the normal parameter to control the initialization behaviour, and init_vm_pu and init_va_degree are the specialized parameters that allow to control the initialization more precisely for advanced users.

I think with this solution we now landed on a good balance of simple usage and maximum control for the user.

closed 2018-09-10 by lthurner