e2nIEE/pandapower #114: Algorithms for ill-conditioned problems

lthurner, 2018-06-25:
In meshed networks, multiple solution might exist, and different solutions might be found depending on the starting values (see #110). Aside from smart choice of initial solution (see #113), other algorithms than Newton-Raphson might be better suited to deal with ill-conditioned problems, such as Levenberg-Marquart or Holomorphic embedding.

GridCal already has implementations of LM and HELM. Do these implementations work and can we adopt them for pandapower (also due to license)?

@ascheidl you have run some tests, can you report your findings?

lthurner, 2018-06-25:
@WinfriedL: Having a simple test case would be helpful for testing, are there simple test cases for this problem, where different restults are found for different starting solutions? Or does this problem only occurr in larger, more complex grids?

WinfriedL, 2018-06-25:
In principle this problem exists in all grids, but usually the voltage magnitudes of the 'wrong' solution should be far enough away from an initial flat start of 1 pu. I am not sure what part of my testgrid leads to the proximity of both solutions.
However, it should be possible to replicate the behaviour with a two-bus grid and a very low initial value for the voltage.

FlorianShepherd, 2018-06-26:
I implemented a version of the Iwamoto Newton Rapshon. This may help to improve convergence. The algorithm can be used by choosing 
`runpp(net, algorithm="iwamoto_nr")`

WinfriedL, 2018-07-05:
Great addition!

Unfortunatly it currently does not help in my testcase. And I believe it cannot work for any network with PV buses.
Because `dV = dVm * exp(1j * dVa)` includes PV buses, and for those dVm is zero, there is always at least some `dV==0` in 
https://github.com/lthurner/pandapower/blob/463311d24012ed7270312e27ba612fd267a965b5/pandapower/pf/iwamoto_multiplier.py#L15-L19
so one gets `iwa_multiplier=1.0` all the time.
Or am I missing something?

WinfriedL, 2018-10-17:
I have looked at the original Iwamoto papers and I think the Implementation was wrong. Or at least I did not understand how the implementation was related to Iwamoto paper. It somehow looked like a strange implementation of the formulas given by Milano (which also differ from the Iwamoto formulas)

I tried to implement the original Iwamoto:
https://github.com/WinfriedL/pandapower/blob/10c7bca902a95dcdfd25aa7243c7c8563adf9c34/pandapower/pf/iwamoto_multiplier.py#L22-L41

It seems to work, and in most cases improves the convergence behaviour. I have not yet done speed comparisons. 

It would be great if someone can look at the implementation for some obvious errors and maybe test it. It is not yet ready for a pull request, there is at least some cleanup necessary.

WinfriedL, 2018-10-18:
I have checked all formulas again, and I think they are correct. I have alos done some speed and convergence comparison. The iwamoto step is slower than a pure NR step, as expected. However, in most of my tests it converges faster and needs less iterations. So depending on the net and the tolerance, it is either a bit slower (if it needs the same amount of iterations) or faster (if it needs less iterations).

And at least in one testcase, the standard Newton-Raphson converged in an unstable solution, while the Iwamoto converged in the stable solution. But that might be coincidence and is probably not a general rule.

lthurner, 2018-10-19:
That sounds great, thanks a lot!

SteffenMeinecke, 2018-10-23:
> @WinfriedL: Having a simple test case would be helpful for testing, are there simple test cases for this problem, where different restults are found for different starting solutions? Or does this problem only occurr in larger, more complex grids?

If you still need small test cases:
- Data for the widespread 3 Ill-conditioned power systems (11 bus and 43 bus systems from iwamoto and 13 bus system from customs head) can be obtained at https://ieeexplore.ieee.org/document/4111178
- In addition, I found another paper that uses these 3 networks, as well as another 25 node network: https://www.sciencedirect.com/science/article/pii/0142061590900327 Maybe that's helpful too? (I suspect there will be some more and recent papers and test cases)

WinfriedL, 2018-10-23:
Thanks, we will look at the 11 bus system. However, it will take some time because the data format is not very "user friendly". Do you have any tools to convert a Ybus matrix into a pandapower network?

lthurner, 2020-03-11:
After GridCal has released an updated version of the HELM algorithm, I looked into this once more. Unfortunately, the iwamoto 11 bus testcase still doesn't converge with the HELM. But I looked into the source once more, and I have doubts the iwamoto network as it is now in pandapower is correct:

- The admittance values have no unit given in the paper, but I am assuming they are in per unit?
- The base power is not explicetly given in the paper (it is given for the 13bus network, but not for the 11bus)
- The bus powers where translated 1:1 to pandapower as MW values, even though they are given in pu. Once again, the base power is not given, so it is not clear if 0.128pu power means 0.128, 12.8MW or 128MW.
- The signing system for bus powers is not given in the paper, so it is not clear if its load or generation.
- Results for the voltage are also not given to allow validation

With that much information missing, I don't see how a network can be compiled with certainty... I would vote to remove the iwamoto network from pandapower as long as we are not sure it is correct

wangzhenassd, 2021-01-04:
Hi all,
are there any advances in this thread regarding the implementation of HELM algorithm? Does it still make sense to pandapower since iwamoto is already available?

AnkurArohi, 2022-03-25:
I tried to follow this thread,  I believe its extremely intresting. 
Do we have iwamoto impl also in pandapower?

vogt31337, 2024-11-21:
> Hi all, are there any advances in this thread regarding the implementation of HELM algorithm? Does it still make sense to pandapower since iwamoto is already available?

I have built a prototype, but it doesnot produce correct results, I think I made something wrong while converting the datastructures: https://github.com/e2nIEE/pandapower/pull/2067