e2nIEE/pandapower #116: Allow quadratic term in dc line losses

WinfriedL, 2018-06-25:
The current DC line model has constant losses (`loss_kw`) and power dependent losses (`loss_percent`). However, the DC line losses should (as far as I know) depend on the transmitted active power squared as well.
With an additional column in the dcline dataframe this could be implemented by changing the line in https://github.com/lthurner/pandapower/blob/f14386ceccd21ae24a2e51962767aced7a8fd80e/pandapower/powerflow.py#L156

I have tested an implementation and get approximatly the same results as Integral. Not exactly the same, since Integral does a full load flow calculation for the dc network with several contributions to the losses, but much better than only with linear losses. 

If you agree I will make a pull request with that implementation, I am not sure what to call that column, though. Maybe something like `loss_const_r`, since it basically depends on a constant resistance?

BTW, there seems to be an error in the documentation for the DC line losses, at least the formula is not the same as in the code...

lthurner, 2018-06-25:
> The current DC line model has constant losses (loss_kw) and power dependent losses (loss_percent). However, the DC line losses should (as far as I know) depend on the transmitted active power squared as well.

The reason quadatic losses are not considered is because it leads to quadratic equations in the OPF, which is more difficult to handle. And for the OPF the linear approximation is usually good enough. It would be a good addition to have quadratic losses also for more detailed power flow calculations.

> If you agree I will make a pull request with that implementation

Yes I agree, but the parameter would then be ignored in the OPF and this should be clearly stated in the documentation somewhere.

> I am not sure what to call that column, though. Maybe something like loss_const_r, since it basically depends on a constant resistance?

The naming convention in pandapower is to have the unit at the end of the parameter name, so if it is an impedance, maybe it should be loss_ohm? Or how exactly are the quadratic losses calculated? 

> BTW, there seems to be an error in the documentation for the DC line losses, at least the formula is not the same as in the code...

Where exactly?

WinfriedL, 2018-06-26:
> The reason quadatic losses are not considered is because it leads to quadratic equations in the OPF, which is more difficult to handle.

Makes sense, thanks for the explanation.

> The naming convention in pandapower is to have the unit at the end of the parameter name

I know. The problem is, the unit would be something like 1/kw, since multiplied by p_kw^2 should yield the losses in kw.
For example, one part is calculated by the resistance of the dc line itself and the (rated) voltage of the dc line:
`p_loss = R*I^2=R*p_kw^2/Ur^2`
Maybe one could define something in relation to `max_p_kw` to get a pu value?
`loss_const_r_percent= 100 * R/U^2 * max_p_kw`

>Where exactly?

The code calculates Pto as
`pto = - (pfrom * (1 - dctab.loss_percent / 100) - dctab.loss_kw) `

while in the documentation the formula is
`pto = - (pfrom -  dctab.loss_kw) *  (1 - dctab.loss_percent / 100)`

WinfriedL, 2018-07-04:
>> If you agree I will make a pull request with that implementation

> Yes I agree, but the parameter would then be ignored in the OPF and this should be clearly stated in the documentation somewhere

At the moment I don't use my initial implementation anymore. Due to some other issues I don't use pandapower dclines, but create my own pairs of generators and have an additional table to keep track of their setpoints (and losses). So while I still could do a pull request, I am not sure if that is a good idea if I don't use (and thereby test) that code?

julffers, 2020-11-06:
@lthurner any news regarding this issue? please reopen in that case.

closed 2020-11-06 by julffers