e2nIEE/pandapower #117: Constant dU for ideal phase shifters

WinfriedL, 2018-06-25:
Based on [entso-e](https://docstore.entsoe.eu/Documents/CIM_documents/Grid_Model_CIM/ENTSOE_CGMES_v2.4_28May2014_PSTmodelling.pdf), ideal phase shifters (symmetrical phase shifters ) can either be implemented with a constant angle shift per step, as it is implemented in pandapower right now, or with a constant voltage step. In the first case the shift angle is calculated as in 
https://github.com/lthurner/pandapower/blob/f14386ceccd21ae24a2e51962767aced7a8fd80e/pandapower/build_branch.py#L362-L364

In the second case the shift angle can be calulated by 
```
trafo_shift[phase_shifters] += (direction * 2* np.rad2deg(np.arcsin(tp_diff[phase_shifters] *
                                            trafo_df["tp_st_percent"].values[phase_shifters]/100/2)))
```
(I believe there is an error in the entso-e document, since the arctan should be an arcsin)

So there is nothing wrong with the current implementation, but at least Integral implements the second formula. I don't know about other tools.

Would it be possible to allow both implentations? Maybe use the first one if `tp_st_degree` is defined and the other one if `tp_st_percent` is defined?

The deviations are not huge (in my testcase about 0.05 to 0.1 degree difference in the resulting voltage angle), so I would understand if you only want to keep one implementation, especially since the current one seems to be more intuitive...

lthurner, 2018-07-09:
@rbolgaryn you have worked with the transformer phase shifters, can you have a look at this?

lthurner, 2018-09-12:
merged

closed 2018-09-12 by lthurner