e2nIEE/pandapower #118: Index problem in _branches_with_oos_buses ?

WinfriedL, 2018-06-28:
I think there is an issue in `_branches_with_oos_buses(net, ppc)` if lines are connected to oos buses on both sides and have non-continuous branch indices.

https://github.com/lthurner/pandapower/blob/a3c270da7f41931168716a6e1f49399ec80eb6cd/pandapower/build_branch.py#L759-L768

`line_is_idx[mask_and]` is an array of branch indices, but ppc["branch"] has an continuous index. 
This leads to an IndexError.

jhmenke, 2018-07-02:
I think an additional mapping via *net._pd2ppc_lookups["branch"]* could be required.
*line_is_idx* is initialized with the pandapower indices: *is_elements["line_is_idx"] = net["line"].index[net["line"].in_service.values]*

@FlorianShepherd ?

lthurner, 2018-07-02:
I think this issue should be adressed within the planned refactoring of the switch code (see #30).

To summarize, there are four basic cases:

1. A branch with an open switch at one side
2. A branch with an open switch at either side
3. A branch that is connected to an out of service bus at one side
4. A branch that is connected to an out of service bus at both sides

Internally, cases 1 and 3 are handled by creating an auxiliary bus and reconnecting the branch to the auxiliary bus. Cases 2 and 4 are handled by setting the branch out of service.

Even though cases 1/3 and 2/4 are basically the same case with regard to the internal modeling, they are adressed in different parts of the code. The [_switch_branches](https://github.com/lthurner/pandapower/blob/a3c270da7f41931168716a6e1f49399ec80eb6cd/pandapower/build_branch.py#L549) function handles switches (case 1 and 2) and the [_branches_with_oos_buses](https://github.com/lthurner/pandapower/blob/a3c270da7f41931168716a6e1f49399ec80eb6cd/pandapower/build_branch.py#L726) function handles out of service buses (cases 3 and 4). But since both functions do basically the same thing just once for OOS buses and once for switches, the functions are very similar, and it looks like some code was copy&pasted from the _switch_branches function to the _branches_with_oos_buses function. So when we refactor the switch branches functionality, my opinion is that these two functions should be merged into one to avoid duplicate code and make it more structured. This also has the advantage that "mixed" cases like an open switch at one side and an out of service bus at the other side, would be considered, which from my understanding would fail at the moment.

Of course none of this adresses the issue that there seems to be a bug. But keeping in mind that the switch code is supposed to be refactored anyway, I would prefer a small (and maybe not very sophisticated) bugfix, if that it is possible. Do we relaly need an additional lookup here? What works with two switches at both sides should also work with two out of service buses, so either we don't need a new lookup or the "open switches at both sides" case also has a bug? @ascheidl you wanted to look into the switch refactoring?

lthurner, 2018-07-03:
I was able to reproduce the problem and have written a small test. I also found a pretty simple solution - I just removed the code block that threw the error. In this code, branches with two out of service buses at each side were explicitely set out of service, but there is really no need to do that because the ext2int function (ppc-->ppci) automatically removes any branch that is connected to a out of service bus. So at least for the test I wrote everything now works as intended. @WinfriedL please report if this also solves your problem?

WinfriedL, 2018-07-03:
I have not tested in detail, at the moment I get an error in https://github.com/lthurner/pandapower/blob/1139dd846ec021066ac81dca4afd06992ff3e068/pandapower/build_branch.py#L773

It seems like `mask_to[mask_or] & ~mask_from[mask_or]` has shape 0, which makes sense, since I have only oos buses on both sides of a line, so mask_or is always False.

`ls_info[:, 0]` has shape 104, however, which is the number of oos buses connected to lines?

lthurner, 2018-07-03:
Yeah it didn't work in this case, fixed it and added it to the test, try again?

WinfriedL, 2018-07-03:
It seems to work now, but I have to do some more tests. 
If I use this functionality my powerflow does not converge anymore, but this might be unrelated.

Usually I run `set_isolated_areas_out_of_service()` before the powerflow, and this works as expected.
But since I determine unsupplied buses during network creation anyway, I wanted to get rid of this extra step. Just as a thought: `_branches_with_oos_buses(net, ppc)` only deals with lines, not other branches like transformers or impedances, right? Can this be a problem, having in_service branches connected to out of service buses?

WinfriedL, 2018-07-04:
> but the power flow should still converge.

Ok, thanks. In that case I have to look further which objects have to be set out of service apart from the buses.

WinfriedL, 2018-07-04:
The poblem are three winding transformers. If those are connected to out of service buses they should be ignored in the powerflow. But the star point bus is in_service, so there is an unsupplied subnet which cannot converge.

WinfriedL, 2018-07-04:
> But the star point bus is in_service

This can be solved by changing
https://github.com/lthurner/pandapower/blob/463311d24012ed7270312e27ba612fd267a965b5/pandapower/powerflow.py#L142-L144
to check if all three original buses are in service. I will make a pull request tomorrow.

closed 2018-07-09 by lthurner