e2nIEE/pandapower #123: Trafo plotting in colormaps

thomaswolgast, 2018-07-09:
Currently I'm plotting a pandapower network as colormap with colorbars for the line load and the voltage deviation. I use create_line_collection() and create_bus_collection(). That functions without problems.
Trafos are not plotted this way and I want to add them, if possible, without colorbar. As I did before with lines and buses, I wanted to use create_trafo_collection(). The creation of the collection functions without problems, but the following plotting fails [draw_collections] with an Attribute Error:
[image]

Also the corresponding tutorial is not available at the moment:
https://github.com/lthurner/pandapower/blob/8db0c33a8283836b2cd001acb20ff7945ca3b980/tutorials/plotting_structural.ipynb

Did I do something wrong? Or is the method broken? 

I also tried create_trafo_connection_collection(). That functions well, but the trafos are simply plottet as thick lines, not with a trafo symbol.

Are there possibilities to adjust the method in a way that the normal trafo symbol is plotted?
(like when the simple_plot() method is used)

lthurner, 2018-07-09:
Looks like you are passing a tuple instead of a collection to the draw_collections function. The trafo collection returns two seperate collections (one for the circles and one for the lines), so you have to catch them both when calling the variable like it is done here:

https://github.com/lthurner/pandapower/blob/8db0c33a8283836b2cd001acb20ff7945ca3b980/pandapower/plotting/collections.py#L672

@SteffenMeinecke you refactored the transformer collection functions, can you check if the changes are the reason the tutorial doesn't compile? Also are colormaps supported yet for this collection?

lthurner, 2018-07-09:
Great, but I would still like to keep this open to figure out what happened to the tutorial...

thomaswolgast, 2018-07-09:
That's strange... When I come from the pandapower docs, the tutorial works well:
https://github.com/lthurner/pandapower/blob/master/tutorials/plotting_structural.ipynb

lthurner, 2018-07-09:
That is because it links to the master branch. It's broken on develop, so apparently something changed since the last release

jhmenke, 2018-07-10:
Both plotting_structural and plotting_colormaps work for me without issues. (commit af09e1ccd80f7 on the develop)

lthurner, 2018-07-10:
Thats strange, maybe it was an issue with github and not with the code? Well in any case it works again, so this can be closed.

closed 2018-07-10 by lthurner