e2nIEE/pandapower #125: Keep out of service buses in create_graph()?

WinfriedL, 2018-07-10:
Currently buses which are out of service are removed from the graph in https://github.com/lthurner/pandapower/blob/1f08ae8ccef62f65088be4cd993c36a571bdc566/pandapower/topology/create_graph.py#L174

This is a problem if the connected lines are in service, since in that case networkx algorithms throw an error. It seems like networkx does not check for edges connected to removed nodes...

Is there any need to remove the buses?

lthurner, 2018-07-12:
> Is there any need to remove the buses?

Well the idea is to have only "active" buses in the graph, so that for example connected components searches do not go over out of service buses. So imho the solution would be to remove the edges connected to the out of service branches as well?

WinfriedL, 2018-07-12:
>Well the idea is to have only "active" buses in the graph, so that for example connected components searches do not go over out of service buses. 

Since you mention it... this means there is a possible source of error if there are nogobuses specified. Those are removed as well without removing the edges.

>So imho the solution would be to remove the edges connected to the out of service branches as well?

I am not sure if this is always the right fix. Suppose you add a line to the net to supply previously unsupplied buses and want to set these buses in service.  This would not be possible with the graph, since the buses are always removed.

I guess that there can be use cases where oos buses should be treated as
- nogobuses (like the current implementation)
- notravbuses (like your reply implies)
- normal buses (like in my example)

lthurner, 2018-07-16:
> I am not sure if this is always the right fix. Suppose you add a line to the net to supply previously unsupplied buses and want to set these buses in service. This would not be possible with the graph, since the buses are always removed.

Unsupplied buses and out of service buses are two seperate things. Unsupplied buses are set out of service internally by the connectivity check, but stay in service in the pandapower network. All in service buses would always be added to the graph by create_nxgraph, even if they are unsupplied.

WinfriedL, 2018-07-16:
> Unsupplied buses and out of service buses are two seperate things. Unsupplied buses are set out of service internally by the connectivity check in the power flow, but stay in service in the pandapower network.

I had to set all unsupplied elements out of service to get converged loadflows, so I used them as the same. But that was probably due to the problem fixed in c0dd7701a1aba6ae4c1a38b9770b86dc821fcac5.
If indeed unsupplied buses should still be in service the right fix would be to treat them as notravbuses?

I am not sure what exactly the intended difference between nogobuses and notravbuses is.

From the current implementation, nogobuses are removed but the edges connecting to it are still in the graph (which should be problematic), while notravbuses stay in the graph, but the edges are removed. So notravbuses become isolated nodes in the graph. Is this intended?

WinfriedL, 2018-07-17:
>But even in this case, if you then supply them with an additional line, you still have to set the buses in service again for them to be considered in the power flow. 

Yes, and currently I cannot do that with a graph search. But if I don't have to set them out of service anymore, there is no problem. I just need to leave them in service and let pandapower handle them internally. I just mentioned that because it was the reaseon why so far out of service and unsupplied meant the same for me. 

>Only the edge that goes from that bus to other buses is removed. So it is possible to get to that bus, but not to go anywhere from this bus. 

That took me some time to understand. This would imply that the graph search has different results depending on the direction the lines are created? That sounds kind of dangerous to me. If just by chance every edge has that bus as from_bus, **every**  edge is removed and it becomes an isolated node and cannot be reached in a graph search. But if every edge has it as to_bus, **no** edge is removed and notravbuses has no influence on the graph search...

Am I missing something? Is that intendend and Ok for the use case?

lthurner, 2018-07-17:
> That took me some time to understand. This would imply that the graph search has different results depending on the direction the lines are created? That sounds kind of dangerous to me. If just by [...]

No it does not depend on the from_bus / to_bus. In a directed graph there needs to be an edge for each direction so that each bus can be reached from the other bus. The add_edge function automatically adds the same edge in both directions. Lets say we have this net:

1---2---3

Then the nxgraph would have the following edges

mg[1][2]
mg[2][1]
mg[2][3]
mg[3][2]

If bus 2 would be a notravbus, the outgoing edges from bus 2 would be removed (regardless which of the buses is the to/from bus), which leaves:

mg[1][2]
mg[3][2]

Now if a connectivity search is started from bus 1, bus 2 will be found, but not bus 3, because there is no connection from bus 2 to 3. If bus 2 were a nogobus, there would be no edges at all in the graph and a connectivity search starting from bus 1 would only find bus 1.

WinfriedL, 2018-07-17:
> In a directed graph there needs to be an edge for each direction

But the Graph is undirected...?

lthurner, 2018-07-17:
> But the Graph is undirected...?

The undirected graph is implemented internally by networkx as a directed graph that has edges in both directions. That is why add_edges always adds both directions for the undirected graph, unless you remove one of those edges manually, which is what the notravbuses do.

WinfriedL, 2018-07-17:
Wow. Now I understand what you are doing and why it probably works. Looks a bit hacky and implicit to me ;-)

I wonder if this is save for all networkx functionality. For example during testing I looked at `mg.edges` which contained no edges starting from notravbuses but all edges ending at notravbuses. But this poperty does not know about the now undirected graph, so it does not contain the reversed edges. So there seems to be parts of networkx which are ok with this, but other parts might not work as expected. But I don't have any indication of a problem, so thanks for all the explanations.

It still leaves the question open how oos buses should be treated in the graph, but I guess the current implementation of treating them the same as nogobuses makes the most sense.

And since I also found my original problem, which was not connected to the graph afterall, there is no need to fix anything. Thanks and sorry for all the noise.

closed 2018-07-17 by WinfriedL