e2nIEE/pandapower #129: Bus IDs in plots

julianw00, 2018-07-12:
I am looking for a plot of a network where the bus ID (and maybe line ID) is depicted in the figure. 
- I know of the html-plot that can be created with pandapower.plotting.html(net, filename). However, the quality is quite bad so one can't properly see the values. Can the quality be enhanced somehow?
- res_plotly does not seem to always show the correct data when scrolling over the lines and busses

Any ideas?

Thanks a lot!

(btw the info in the doc about the html-export at https://pandapower.readthedocs.io/en/v1.5.1/plotting/html.html is no longer available)

FlorianShepherd, 2018-07-12:
We also discovered some issues with plotly lately. Have you tried plotting with the newest pandapower version on develop? 

If you still have issues with the develop version, please provide an example. That way we can have a look into it.

Thank you for the hint about the doc :+1:

lthurner, 2018-07-12:
> (btw the info in the doc about the html-export at https://pandapower.readthedocs.io/en/v1.5.1/plotting/html.html is no longer available)

This is already fixed in the develop version: https://pandapower.readthedocs.io/en/develop/plotting/html.html

jhmenke, 2018-08-06:
is this still an issue?

jhmenke, 2018-08-06:
>I know of the html-plot that can be created with pandapower.plotting.html(net, filename). However, the quality is quite bad so one can't properly see the values. Can the quality be enhanced somehow?

I just pushed a commit that enables zooming on the to_html-plot via the mouse wheel.

jhmenke, 2018-08-07:
I also found a bug in the plotly bus traces plotting. Please update to 1e4470c9bca and try again. The labels should be correct now.

jhmenke, 2018-08-09:
are the issues fixed? @julianw00

closed 2018-08-10 by jhmenke