e2nIEE/pandapower #130: LoadflowNotConverged using a individually designed distribution grid

Poldi1912, 2018-07-12:
Hello,

I have on major problem in my code to conduct a powerflow calculation in pandapower. Apparently, my problem has an issue that it does not converge. Hence, i would like to ask you for any possible help.

FYI: I already tried to set my transformer values fo pfe_kw and vscr_percent to zero, but unfortunately this did not help. 
I further used the internal diagnostic function in order to identify the issue but apparently non of these served as a solution to let the solver converge.

I deeply appreciate any help from your side and I am sorry for the long code but I thought this might help to follow up on my issue.

Here is the code that i implemented so far and the additional fault report:
`    
    import pandapower as pp
    import openpyxl as ox

    #Load input file as workbook
    wb = ox.load_workbook("C://Users//Paul Kessler//Documents//Studium//Master//Masterarbeit//E.ON 
    Masterarbeit//03_Analyse//03_Use cases//Use case A//Raw Data//Gridassetinformation.xlsx")
    #print (wb.sheetnames)
    net = pp.create_empty_network()

    #Create medium and low voltage Buses
    buses = wb['node']
    for m in range(3,buses.max_row+1):
        buses.name = buses.cell(row=m, column=1).value
        if (buses.cell(row=m, column=2).value == 22):
            pp.create_bus(net, name=buses.name, vn_kv=22, type='b')
            
        elif (buses.cell(row=m, column=2).value == 132):
            pp.create_bus(net, name=buses.name, vn_kv=120, type='b')
        else:
            pp.create_bus(net, name=buses.name, vn_kv=.42, type='b')
        
    #Create an external grid
    feeder = wb['network feeder']
    feeder.bus = feeder.cell(row=3, column=9).value
    feeder.s_sc_max_mva = feeder.cell(row=3, column=4).value
    feeder.s_sc_min_mva = feeder.cell(row=3, column=3).value
    feeder.rx_max = feeder.cell(row=3, column=8).value
    feeder.rx_min = feeder.cell(row=3, column=7).value
    feeder.bus = feeder.cell(row=3, column=9).value
    feeder.bus = feeder.cell(row=3, column=9).value
    bus = pp.get_element_index(net, "bus", feeder.bus)

    ext_grid = pp.create_ext_grid(net, bus, vm_pu=1.02, va_degree=0, name='External grid', 
                                  s_sc_max_mva=feeder.s_sc_max_mva, s_sc_min_mva=feeder.s_sc_min_mva,
                                  rx_max=feeder.rx_max, rx_min=feeder.rx_min)

    #Create Lines connected to the MV Buses
    mv_lines = wb['line']
    for i in range(3,mv_lines.max_row+1):
        mv_lines.from_bus = mv_lines.cell(row=i, column=14).value
        mv_lines.to_bus = mv_lines.cell(row=i, column=16).value
        mv_lines.length = mv_lines.cell(row=i, column=3).value
        mv_lines.r_ohm_per_km = mv_lines.cell(row=i, column=6).value
        mv_lines.x_ohm_per_km = mv_lines.cell(row=i, column=7).value
        mv_lines.c_nf_per_km = mv_lines.cell(row=i, column=8).value
        mv_lines.max_i_ka = mv_lines.cell(row=i, column=12).value
        from_bus = pp.get_element_index(net, "bus", mv_lines.from_bus)
        to_bus = pp.get_element_index(net, "bus", mv_lines.to_bus)    
    
        if (mv_lines.cell(row=i, column=4).value == 1):
            pp.create_line_from_parameters(net, from_bus, to_bus, length_km=mv_lines.length,
                                           r_ohm_per_km=mv_lines.r_ohm_per_km, 
                                           x_ohm_per_km=mv_lines.x_ohm_per_km, 
                                           c_nf_per_km=mv_lines.c_nf_per_km, 
                                           max_i_ka= mv_lines.max_i_ka,
                                           name='Cable-Line MV%s 22kV' % i, index=None, 
                                           type=None, geodata=None, 
                                           in_service=True, df=1., 
                                           parallel=1, g_us_per_km=0.)
[... 107 lines cut ...]

_____________ END OF PANDAPOWER DIAGNOSTIC _____________ 
C:\ProgramData\Anaconda3\lib\site-packages\pandapower\pf\makeBdc.py:61: RuntimeWarning: invalid value encountered in true_divide
  b = stat / branch[:, BR_X]                ## series susceptance
Traceback (most recent call last):

  File "<ipython-input-113-b3a6a2dff4e9>", line 1, in <module>
    runfile('C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes/GridmodelA.py', wdir='C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes')

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes/GridmodelA.py", line 144, in <module>
    pp.runpp(net, calculate_voltage_angles=True, algorithm='nr', check_connectivity=True, max_iteration=10, init="dc")

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\run.py", line 252, in runpp
    _powerflow(net, **kwargs)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 79, in _powerflow
    "{1} iterations!".format(algorithm, max_iteration))

LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!

jhmenke, 2018-07-13:
> C:\ProgramData\Anaconda3\lib\site-packages\pandapower\build_branch.py:392: RuntimeWarning: invalid value encountered in sqrt
x_sc = np.sign(z_sc) * np.sqrt(z_sc ** 2 - r_sc ** 2)

so r_sc is alrger than z_sc in this case. it may be a line, but likely a transformer parameter is at fault.
maybe use the debug mode to figure out which element exactly and then try to work your way backwards where r_sc and z_sc are calculated.

>C:\ProgramData\Anaconda3\lib\site-packages\pandapower\build_branch.py:271: RuntimeWarning: invalid value encountered in true_divide
zbc_triangle = zSum_triangle / za_star

za_star is zero. this also points to a transformer parameter value 

>C:\ProgramData\Anaconda3\lib\site-packages\pandapower\build_branch.py:274: RuntimeWarning: invalid value encountered in true_divide
y[tidx] = -2j / zbc_triangle

same here

>C:\ProgramData\Anaconda3\lib\site-packages\pandapower\pf\makeYbus_pypower.py:79: RuntimeWarning: invalid value encountered in true_divide
>Ysf = stat / (branch[:, BR_R] + 1j * branch[:, BR_X]) ## series admittance

both admittances are zero in this case. again, find out with debugging whether it's a line or transformer at fault (i suspect transformer) and work your way backwards. you can also create additional checks in the code before pandapower elements are created, e.g., using
assert mv_lines.r_ohm_per_km > 0
to make sure the input parameters are what you expect.

lthurner, 2018-07-16:
The problem seems to be in this line:

> x_sc = np.sign(z_sc) * np.sqrt(z_sc ** 2 - r_sc ** 2)

and all other warnings are follow up issues of this problem. I would suggest debugging to this line and figuring out why this operation fails. I assume either there is a negative value under the square root or there is an nan somewhere in the transformers.

Poldi1912, 2018-07-16:
Thank you, lthurner and jhmenke, I was able to identify the transformer parameter that was inducing the issue. Unfortunately, the powerflow still does not converge.

The console detects the following:
"""
_____________ PANDAPOWER DIAGNOSTIC TOOL _____________ 

Checking for overload...

Overload check failed: Power flow still does not converge with load scaled down to 0.1 percent.
Overload check failed: Power flow still does not converge with generation scaled down to 0.1 percent.

 --------

Checking switch configuration...

Power flow still does not converge with all switches closed.

 --------

_____________ END OF PANDAPOWER DIAGNOSTIC _____________ 
Traceback (most recent call last):

  File "<ipython-input-4-b3a6a2dff4e9>", line 1, in <module>
    runfile('C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes/GridmodelA.py', wdir='C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes')

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes/GridmodelA.py", line 144, in <module>
    pp.runpp(net, calculate_voltage_angles=True, algorithm='nr', check_connectivity=True, max_iteration=20, init="dc")

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\run.py", line 252, in runpp
    _powerflow(net, **kwargs)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 79, in _powerflow
    "{1} iterations!".format(algorithm, max_iteration))

LoadflowNotConverged: Power Flow nr did not converge after 20 iterations!
"""

I tried to play with the number of iteration steps but this does not have any impact as well. 
Are there any other options you can think of that make my powerflow not converge?

Thank you for your help so far!!!

lthurner, 2018-07-17:
Do you have unsupplied buses in the grid? You can check that with the function:

```
pandapower.topology.unsupplied_buses(net)
```

Poldi1912, 2018-07-18:
Hi lthurner, apparently yes, i identified 3 unsupplied buses while my data input file seems to consider them. Thank you for this new hint!

I will try to figure out what is issuing that the buses aren't recognised to be supplied.

Poldi1912, 2018-07-19:
i am still having the problem that the powerflow is not converging.  Apparently the buses [25, 26, 356] are bein unsupplied. However, when i am using the following function for each of the buses i can identify the bus [428] in the list of connected components and see that my external grid is connected to:
`    connected_component = pp.topology.connected_component(mg, 25, notravbuses=[])` `print(connected_component)`
Generating the output list:
[25, 26, 356, 27, 12, 4, 0, 1, 303, 2, 3, 349, 11, 13, 7, 8, 5, 6, 333, 334, 351, 19, 18, 17, 353, 20, 21, 9, 10, 16, 304, 352, 14, 15, 313, 312, 314, 350, 22, 29, 30, 354, 23, 24, 335, 336, 358, 305, 355, 28, 34, 33, 32, 31, 37, 36, 359, 35, 38, 41, 44, 43, 59, 71, 77, 81, 80, 370, 427, 78, 93, 103, 102, 101, 92, 91, 306, 104, 308, 105, 106, 107, 112, 124, 126, 127, 302, 332, 381, 129, 128, 300, 324, 383, 120, 121, 317, 284, 135, 134, 301, 141, 142, 140, 132, 133, 125, 138, 137, 136, 327, 328, 386, 130, 131, 329, 330, 388, 139, 145, 144, 143, 389, 146, 148, 147, 151, 152, 153, 390, 150, 149, 156, 158, 159, 160, 392, 154, 155, 161, 165, 164, 393, 162, 163, 391, 157, 168, 172, 170, 169, 177, 190, 185, 186, 179, 180, 397, 191, 194, 192, 193, 197, 196, 187, 184, 183, 182, 181, 398, 188, 189, 322, 323, 399, 198, 199, 200, 337, 338, 401, 201, 202, 285, 203, 204, 402, 343, 400, 195, 205, 211, 217, 286, 234, 235, 236, 237, 241, 242, 221, 207, 208, 213, 214, 404, 222, 223, 225, 224, 405, 245, 248, 252, 255, 246, 247, 415, 256, 258, 257, 261, 262, 264, 263, 275, 276, 278, 281, 280, 432, 430, 429, 431, **428**, 279, 283, 282, 420, 277, 309, 421, 273, 274, 269, 268, 419, 270, 271, 272, 418, 267, 265, 266, 417, 259, 260, 319, 320, 411, 344, 410, 253, 254, 289, 290, 339, 340, 416, 249, 250, 251, 414, 347, 426, 238, 240, 239, 412, 346, 425, 287, 288, 299, 311, 423, 296, 297, 298, 218, 220, 219, 229, 228, 227, 406, 407, 230, 233, 243, 244, 413, 232, 231, 408, 212, 216, 215, 226, 345, 409, 292, 291, 424, 206, 210, 209, 403, 178, 176, 175, 395, 171, 173, 174, 396, 167, 166, 394, 310, 422, 331, 387, 318, 384, 321, 385, 114, 113, 115, 294, 293, 341, 342, 379, 116, 117, 118, 119, 380, 122, 123, 382, 96, 88, 87, 86, 85, 371, 98, 97, 325, 326, 373, 99, 100, 378, 110, 111, 376, 108, 109, 375, 94, 307, 377, 95, 90, 89, 374, 79, 82, 74, 75, 65, 66, 63, 64, 295, 62, 61, 365, 68, 67, 366, 69, 70, 368, 76, 73, 72, 369, 83, 84, 372, 348, 367, 60, 57, 58, 363, 45, 46, 360, 42, 48, 47, 49, 52, 56, 55, 364, 53, 54, 362, 50, 51, 315, 316, 361, 40, 39, 357]
Thus, i do not understand why the system detects bus[25] as unsupplied.

Poldi1912, 2018-07-24:
Hello again, apparentlyit the issue was not imposed by the buses. However, I tried to run the powerflow with a different algorithm (bfsw) and got a new error message saying:

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/Paul Kessler/Documents/Studium/Master/Masterarbeit/E.ON Masterarbeit/03_Analyse/03_Use cases/Use case A/SimulationToolA/Classes/GridmodelA.py", line 172, in <module>
    pp.runpp(net, calculate_voltage_angles=True, algorithm='bfsw', check_connectivity=True, max_iteration=100, init="flat")

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\run.py", line 252, in runpp
    _powerflow(net, **kwargs)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 70, in _powerflow
    result = _run_pf_algorithm(ppci, net["_options"], **kwargs)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 95, in _run_pf_algorithm
    result = _run_bfswpf(ppci, options, **kwargs)[0]

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\pf\run_bfswpf.py", line 409, in _run_bfswpf
    ppci, DLF, buses_ordered_bfs_nets = _get_bibc_bcbv(ppci, options, bus, branch, G)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\pf\run_bfswpf.py", line 177, in _get_bibc_bcbv
    DLF, buses_ordered_bfs_nets = _make_bibc_bcbv(bus, branch, graph)

  File "C:\ProgramData\Anaconda3\lib\site-packages\pandapower\pf\run_bfswpf.py", line 148, in _make_bibc_bcbv
    shape=(nobranch, nobranch))

  File "C:\ProgramData\Anaconda3\lib\site-packages\scipy\sparse\compressed.py", line 51, in __init__
    other = self.__class__(coo_matrix(arg1, shape=shape))

  File "C:\ProgramData\Anaconda3\lib\site-packages\scipy\sparse\coo.py", line 191, in __init__
    self._check()

  File "C:\ProgramData\Anaconda3\lib\site-packages\scipy\sparse\coo.py", line 243, in _check
    raise ValueError('negative column index found')

ValueError: negative column index found

Do you have any hint, what could be the issue for this negative index?

lthurner, 2018-07-24:
> Do you have any hint, what could be the issue for this negative index?

Not really, but the backward-forward sweep is not as well tested as the Newton-Raphson, so i would really stick with the NR instead of alternate solvers.

You are using very peculiar parameters for runpp:

>  pp.runpp(net, calculate_voltage_angles=True, algorithm='bfsw', check_connectivity=True, max_iteration=100, init="flat")

typically if voltage angles are calculated you would want to initialize with a dc power flow instead of a flat start, which also the default. If you have convergence issues, i would advice to not calculate voltage angles and see if you can get the power flow to converge like that.

Poldi1912, 2018-07-25:
While debugging I noticed that pandapower apparently confuses my the switches between several buses. Therefore, I always receive "isolated_nodes" and "bus_not_reachable" when the connection between two "nodes"/ "buses" is realized with a switch. Is this a common problem?

As a help, I initiated all my switches as "lines" within pandapower and set very small values for "r_ohm_per_km","x_ohm_per_km" and "length_km" etc... In the end, my powerflow is still not converging but pandapower is at least recognizing the buses as reachable and not isolated.

As I am a bloody beginner in programming, I would be happy if anyone has an idea for the problem.

lthurner, 2018-07-25:
> While debugging I noticed that pandapower apparently confuses my the switches between several buses. Therefore, I always receive "isolated_nodes" and "bus_not_reachable" when the connection between two "nodes"/ "buses" is realized with a switch. Is this a common problem?

If you can make a small reproducable example of the problem we can look into it. But I doubt pandapower is confusing any buses, because the switch model is actually pretty well tested. I am pretty sure this is a problem with how you input the data. Are you aware of the difference between bus-bus and bus-line switches? 

> As I am a bloody beginner in programming, I would be happy if anyone has an idea for the problem.

It is really not possible for anyone to debug your code for you. Even if we had the time to do that, its not possible without having the data base you are working off. You will have to provide small reproducable examples of the problem, otherwise there is no way for us to know what is wrong with your code or network.

Poldi1912, 2018-07-25:
First of all, thank you for your effort and fast replies regarding my questions and the overall issue!

I identified the issue of my problem. For the sake of completeness and future newcomers to pandapower, I want to describe the cause of my issue a bit further:

After a bit of debugging, I  descried that whenever a switch connected two buses or a bus and a node the connected node/ bus was isolated. I checked my input file and could not locate why this was the case, as the topology is directly drawn from a SCADA system of a DSO.

In the end, it was the input data that was loaded into the pandapower inducing the powerflow to not converge. The DSO handed over the load data including large reactive power values. Setting these values to zero made the powerflow converge and yield in reasonable results.

I still can not explain why these values caused the issue that some buses were isolated tho.

rbolgaryn, 2021-11-11:
The issue seems to be still unsolved - the calculation is working as expected in the example shown here, but in a different example the following unexpected behavior can be observed:

* with recycle=True, continue_on_divergence=False the calculation fails silently (Loadflow not converged), the rest of the simulation after the error has wrong results
* with recycle=False and continue_on_divergence=False, the calculation fails at error, as expected

The recycle feature must be investigated to find the root cause of this issue.

closed 2025-03-19 by vogt31337