e2nIEE/pandapower #137: NaN results in branch current

WinfriedL, 2018-07-26:
Running a powerflow with oos buses  (BUS_TYPE=NONE) in the ppc results in warnings when calculating the branch results because theses buses have no voltage results:

```
RuntimeWarning: invalid value encountered in reduce
    return umr_maximum(a, axis, None, out, keepdims)
```

A possible fix is in https://github.com/WinfriedL/pandapower/commit/0f09276c3f2bb1544acb8bd2d42c9980b15902e5

Not sure if this is the best solution...

lthurner, 2018-07-27:
I also remember there was some inconsistency in the branch results . I think out_of_service lines have a loading of zero percent, while disconnected lines have nan or something like that? Is this issue related?

WinfriedL, 2018-07-30:
> I think out_of_service lines have a loading of zero percent, while disconnected lines have nan or something like that? Is this issue related?

Yes I think this exactly the issue.

Islolated Lines have no voltage result (nan) at the from/to bus, so dividing s_ft (which is zero) by u_ft gives nan results for i_ft and loading_percent.

The question is if nan is a valid powerflow result for isolated buses/lines. If not, maybe there should be no nan result for the bus voltage in the first place. However, a 'correct' voltage result for isolated buses would be zero, but this would also lead to nan results for lines.

Maybe a better defined approach would be 
`i_ft = np.where(s_ft==0., 0., s_ft * 1e-3 / u_ft / np.sqrt(3))`
 instead of brute force filling all nan and inf values...

If nan results are valid, maybe the warning should be suppressed with np.seterr()?

lthurner, 2018-07-30:
After talking to @SteffenMeinecke , the issue that I remembered was that lines with two open switches at both sides have a loading of zero, while all other disconnected/out of service lines have nan. This is a different issue that comes from the fact that we explicitely deal with the "open switches at both sides of the line" case as a special case. So its not exactly the same issue, but related.

> The question is if nan is a valid powerflow result for isolated buses/lines. If not, maybe there should be no nan result for the bus voltage in the first place. However, a 'correct' voltage result for isolated buses would be zero, but this would also lead to nan results for lines.

Its true that it would be physically correct to say a disconnected bus has zero voltage, but one could also argue that the power flow doesn't calculate a voltage value for that bus, therefore no number is assigned. From a practical perspective, I like the clear differentiation between "result" (a number) and "no result" (not a number). It wouldn't be very practical always having to filter out the 0 values when calculating for example the minimum voltage or medium line loading in a grid. 

> If nan results are valid, maybe the warning should be suppressed with np.seterr()?

That however changes the behaviour numpy-wide even outside pandapower. What version of numpy are using btw, because I can't reproduce the warning with numpy 1.14.5. Does this:
```
np.array([1,np.nan,3]) / np.array([3,4,np.nan])
```
give a warning for you? Or am looking at the wrong case here?

WinfriedL, 2018-07-30:
> Does this:
np.array([1,np.nan,3]) / np.array([3,4,np.nan])
give a warning for you? Or am looking at the wrong case here?

No, the line giving the warning is in fact this one:
https://github.com/e2nIEE/pandapower/blob/5516e064f6e807ab38e4bc2f28f5239ea4627341/pandapower/results_branch.py#L66

I should have mentioned that before, sorry. 

> I like the clear differentiation between "result" (a number) and "no result" (not a number)

So the issue would be how to get np.max()  to return nan, but not issue a warning about that...
Of course this is not high priority, since in principle everything works, but always having a RuntimeWarning when everything works as intended seems kind of awkward.

lthurner, 2018-07-30:
>  Of course this is not high priority, since in principle everything works, but always having a RuntimeWarning when everything works as intended seems kind of awkward.

Definetly. Seems like this warning was recently introduced, at least with numpy 1.11 I don't get that warning.

Instead of using np.seterr, I suggest using np.errstate to supress the warning only in this line:

```
with  np.errstate(invalid='ignore'):
    i_ka = np.max(i_ft[f:t], axis=1) 
```

lthurner, 2018-07-30:
I pushed the fix as suggested above. I also checked on the two-open switches at one line issue as descibed above, and it seems like that has since been fixed and now also returns nan results, so everything seems to be consistent. So if the fix works for you, this issue can be closed?

WinfriedL, 2018-07-31:
Yes this can be closed, thanks!

WinfriedL, 2018-07-31:
I closed too early. Not surprisingly the same happens now for transformer branches. np.max() is not used for impedances, so there should be no problem. But switches use it, I guess only if switches have a resistance?

The fix should be straightforward, should I do it and make a pull request or do you want to add that yourself?

PR #138 (2018-07-31): silence warning for transformers and switches without results
Since I already added the fix anyway...
Closes #137
closed 2018-07-31 by lthurner