e2nIEE/pandapower #140: Change of patch_type in create_bus_trace show no effect

julianw00, 2018-08-17:
Hi all,

I wanted to change the type of how buses are represented in pf_res_plotly.py. For that, I changed the patch_type as explained in the documentation of create_bus_trace.

`bus_trace = create_bus_trace(net, net.bus.index, size=bus_size, patch_type='star', infofunc=hoverinfo, cmap=cmap, cbar_title='Bus Voltage [pu]', cmin=0.9, cmax=1.1)` 

It shows no effect in the figure and buses are in fact still represented as circles. Can you reproduce the problem? 

When I change the standard patch_type in create_bus_trace, simple_plotly excepts the change, res_plotly, however, does still not.

I am not interested in changing the standard symbol in res_plotly. I just tried to add further elements to the loadflow figure, such as loads, using bus_traces with different symbols. 

Thanks a lot,
Julian

jhmenke, 2018-08-21:
After the related pull request has been processed, custom markers should work even when the cmap parameter is not None.

jhmenke, 2018-08-21:
if the cmap parameter is given, the Marker is

closed 2018-08-21 by jhmenke