e2nIEE/pandapower #146: STATCOM, UPFC and capacitor banks in pp

thomaswolgast, 2018-08-21:
For a voltage control algorithm, I want to implement Static Synchronous Compensators (STATCOM), Unified-Power-Flow-Controller (UPFC) and capacitor banks in a pandapower network. My general question is: what are the best solutions to implement these three elements?

Capacitor banks seem to be the easiest ones, because of the "create_shunt_as_capacitor()" method. But it appears that it is not possible to define a fixed "vm_pu". Instead the reactive power "q_kvar" is defined and the voltage is a result after power flow calculation. Is it possible to turn this around? To define a fixed "vm_pu" and get the "q_kvar" as a result? Or would it be the best to use just a Generator with a fixed active power of 0? 

For the other two I didn't find any prefabricated elements in pp. But again I think that for the STATCOM a simple Generator could be the easiest solution. But maybe these problems occured before and someone did already implement these elements in pandapower? Are there any experiences with this case?

lthurner, 2018-08-22:
>For a voltage control algorithm, I want to implement Static Synchronous Compensators (STATCOM), Unified-Power-Flow-Controller (UPFC) and capacitor banks in a pandapower network. My general question is: what are the best solutions to implement these three elements?

Always depends on the control scheme: what value is given and what value has to be calculated? Accordingly use load/sgen for PQ type elements, gens for PV type elements and shunt for impedance type elements. ZIP loads can also be used to model constant current sources. For an element with for example a droop control, you will need to implement a loop that adapts the element settings until the power flow result is in line with the droop curve.

>  Or would it be the best to use just a Generator with a fixed active power of 0?

Yes that would be the way to go in this case.

> But maybe these problems occured before and someone did already implement these elements in pandapower? Are there any experiences with this case?

At the moment we don't plan to add any elements, but you can propose one if you think a certain element would be helpful in a specific use case.

friederikemeier, 2018-09-10:
Hi, sorry for the late response.
So, yes, It is in per unit. For one entry of the matrix it is power in per unit divided by voltage in per unit. The unit value for power is 1 MVA by default and can be changed by using the parameter net.sn_kva.
If you do not change it, you can as well work with it like it was power in MW divided by voltage in per unit.

friederikemeier, 2018-09-10:
`PV elements results in a constant voltage and therefore a sensitivity value of 0.` I don't get it. If you want this bahvour: `To define a fixed "vm_pu" and get the "q_kvar" as a result?`, then you should use a PV element. I think it is quite unefficient to implement your own loop to emulate a PV element with a PQ element. Maybe you can tell a little more about what exactly you are tryinign to optimize/regulate,

thomaswolgast, 2018-09-26:
Thank you for your answer anyway,
actually I decided to use PV elements in the meantime. The reason I wanted to use PQ-elements was that I evaluate different voltage control concepts. And some of these concepts require PQ-elements, and some PV-elements. That's why I wanted to unify it, so that I can apply all concepts to the same network with the same elements. But you are right, using PV-elements is much easier and less prone to errors.

By the way, I think it is confusing to use MVA here, when you use kVA in most other parts of the pandapower code.

From my side this issue can be closed. Thank you.