e2nIEE/pandapower #153: pandapower 1.6

lthurner, 2018-09-07:
A lot has changed since the last release, so I suggest releasing pandapower 1.6 soon. What do we still want to include in this release? I suggest:

- [ ] #122
- [ ] #117 / #126 
- [ ] #113
- [ ] Update changelog

And optional:

- [ ] #147 
- [ ] #85 / #127

lthurner, 2018-09-12:
@SteffenMeinecke: I think we had the "the transformer parameter has changed" warning for the adopted standard type parameters now for a long enough time that it can be deprecated, can you please do that?

SteffenMeinecke, 2018-09-13:
> 
> 
> @SteffenMeinecke: I think we had the "the transformer parameter has changed" warning for the adopted standard type parameters now for a long enough time that it can be deprecated in 1.6, can you please do that?

Yes, I thought about that, too. I agree.

lthurner, 2018-09-18:
released

closed 2018-09-18 by lthurner