e2nIEE/pandapower #154: data type mismatch error in from_json

jkupka, 2018-09-07:
Right now `to_json` doesn´t ensure right data types and sometimes it´s not possible to load the saved network with `from_json` again.

The following example will raise a **data type mismatch** error because the reference apparent power will be saved as int instead as float and `from_json` is expecting **sn_kva** to be a float.

```
import pandapower as pp
import pandapower.networks as nw

net = nw.case9()
net.sn_kva = 1000
pp.to_json(net, "net.json")
net = pp.from_json("net.json")
```

rbolgaryn, 2018-09-07:
we need to keep in mind that int(10.5) is 10

rbolgaryn, 2018-09-07:
so net shouldn't have int fields

rbolgaryn, 2018-09-07:
We shouldn't be casting types because we risk losing informatioon in a case like in my previous comment.
If the user changes sn_kva to int, it is the responsibility of the user.
We can discuss this on Monday, now I will make the pull request so that the above example works and net can be loaded from json.

closed 2018-09-10 by lthurner