e2nIEE/pandapower #173: ValueError: shape mismatch when simulating generators with different lengths to piecewise linear cost curves

dzimmanck, 2018-09-30:
I am simulating a system with PV and Storage.  The piece-wise linear cost function for PV had 2 points, while the piece-wise linear cost function for my storage had 4 points.  When I attempted to run, I saw the error below.  I was able to get the program to run by adding dummy points to the PV cost curve so that the array was the same size as the Storage array.  I think there my be a bug in the cost aggregation code that runs internally.

--------------------------------------------------------------------------------------------------
  File "C:\Python27\lib\site-packages\pandapower\run.py", line 443, in runopp
    _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)
  File "C:\Python27\lib\site-packages\pandapower\optimal_powerflow.py", line 40, in _optimal_powerflow
    ppc, ppci = _pd2ppc(net)
  File "C:\Python27\lib\site-packages\pandapower\pd2ppc.py", line 124, in _pd2ppc
    ppci = _make_objective(ppci, net)
  File "C:\Python27\lib\site-packages\pandapower\opf\make_objective.py", line 132, in _make_objective
    n_piece_lin_coefficients:2] = p
ValueError: shape mismatch: value array of shape (3,4) could not be broadcast to indexing result of shape (3,3)
--------------------------------------------------------------------------------------------------------

lthurner, 2018-10-02:
Can you provide a minimal example to reproduce the bug?

dzimmanck, 2018-10-03:
@lthurner 

Attached is a minimalist example of the bug.  It simulates a 1 bus system with a utility, 3kW of solar, 10kW of storage, and a 1kW load.  It only runs if you insert 2 dummy points in the PWL cost curve for PV so that its length matches the PWL curve used for storage.

`import pandapower
import numpy as np

# create an emty network
house = pandapower.create_empty_network(name="House",
                                      f_hz = 60.0,
                                      sn_kva = 1e-3)
# create buses
main_panel = pandapower.create_bus(house,
                      name = "Main Panel",
                      vn_kv = 0.24,
                      in_service = True)

# grid connection
PGandE = pandapower.create_ext_grid(house, main_panel, min_p_kw = -1e9, max_p_kw = 0)  # Don't allow export ("self-supply")

# add a 3kW PV inverter
pv_inverter = pandapower.create_sgen(house, 
    bus = main_panel, 
    p_kw = -3, 
    q_kvar = 0, 
    sn_kva = 300e-3, 
    name = "IQ8", 
    type= "PV", 
    in_service=True, 
    max_p_kw=0, 
    min_p_kw=-300e-3, 
    max_q_kvar=300e-3, 
    min_q_kvar=-300e-3, 
    controllable=True, 
    k=2, 
    rx=1.0)

# add a 10kW Battery inverter
battery_inverter = pandapower.create_storage(house, 
  bus = main_panel, 
  p_kw = 0, 
  max_e_kwh = 3.3, 
  q_kvar = 0, 
  sn_kva = 4*300e-3, 
  soc_percent = 50.0, 
  min_e_kwh=0.33, 
  name="ACB3", 
  type = "Lithium Ion", 
  in_service=True, 
  max_p_kw=4*300e-3, 
  min_p_kw=-4*300e-3, 
  max_q_kvar=4*300e-3, 
  min_q_kvar=-4*300e-3, 
  controllable = True)

# add load
pandapower.create_load(house, 
                       bus = main_panel, 
                       p_kw = 1, 
                       q_kvar=0, 
                       const_z_percent=0, 
                       const_i_percent=0, 
                       sn_kva=1, 
                       name = 'Washer Machine', 
                       scaling=1., 
                       index=None, 
                       in_service=True, 
                       type=None, 
                       max_p_kw=5, 
                       min_p_kw=0, 
                       max_q_kvar=1, 
                       min_q_kvar=0, 
                       controllable=False)

# add costs
pv_pwl_cost = np.array([[-300e-3, 0],[0, 0]])  # sunlight is free (THIS DOES NOT WORK)
#pv_pwl_cost = np.array([[-300e-3, 0],[-2e-3, 0], [-1e-3, 0], [0, 0]])  # add some dummy points so the PWL lengths match (THIS DOES WORK)
battery_pwl_cost = np.array([[-4*300e-3, 12],[-10e-3, 0.01], [10e-3, -0.01], [4*300e-3, -10]])
pandapower.create_piecewise_linear_cost(house, pv_inverter, "sgen", pv_pwl_cost, type="p")
pandapower.create_piecewise_linear_cost(house, battery_inverter, "storage", battery_pwl_cost, type="p")
PGandE_cost = pandapower.create_polynomial_cost(house, 0, 'ext_grid', np.array([-50, 0]))  # 50 cents per kWh utillity cost

# calculate optimal power
pandapower.runopp(house, verbose=True)

# print results
print house.res_sgen
print house.res_storage
print house.res_ext_grid
print house.res_cost/1e6`

dzimmanck, 2018-10-03:
Here is the code in a zip file.
[pandapower_pwl_bug.zip](https://github.com/e2nIEE/pandapower/files/2440191/pandapower_pwl_bug.zip)

friederikemeier, 2018-10-04:
Hi,

This is not a bug. This is a requirement of the pypower OPF and we see pandapower only as an interface of the pypower OPF.
We would have to insert the dummy points internally as well to keep all PWL costs in the same length, but this is implicit and not desired. Kind regards!

closed 2018-10-04 by friederikemeier