e2nIEE/pandapower #191: Wrong generator lookup with isolated buses

WinfriedL, 2018-10-18:
The following net has an issue with the generator lookup:

```
net=pp.create_empty_network()

gen_bus=pp.create_bus(net,vn_kv=1., name='gen_bus')
slack_bus=pp.create_bus(net,vn_kv=1., name='slack_bus')
gen_iso_bus=pp.create_bus(net,vn_kv=1., name='iso_bus')

pp.create_line(net, from_bus=slack_bus, to_bus=gen_bus, length_km=1, std_type="48-AL1/8-ST1A 10.0")

pp.create_ext_grid(net, bus=slack_bus, vm_pu=1.)        

pp.create_gen(net, bus=gen_iso_bus, p_kw=-1, vm_pu=1., name='iso_gen')
pp.create_gen(net, bus=gen_bus, p_kw=-2, vm_pu=1., name='gen')

pp.rundcpp(net)
print(net.gen[['name','p_kw']].join(net.res_gen[['p_kw']], rsuffix="_res"))
print(net["_pd2ppc_lookups"]['gen'])
print(net['_ppc']['gen'][:,[1]])
```

The result of the (not isolated) generator is -1 kw instead of -2 kw. From the ppc it seems like the lookup is wrong.

WinfriedL, 2018-10-18:
I think the problem is (not surprisingly) in
https://github.com/e2nIEE/pandapower/blob/67823ae7a60af529b7fae9ad863c5e57cb079fba/pandapower/pd2ppc.py#L198-L221

The `new_gen_positions` still contains all generators from the ppc, even isolated ones, but `eg_end ` and `gen_end ` only count in_service ones. That means the indices in the generated lookup are shifted.

WinfriedL, 2018-10-18:
I have added a testcase in https://github.com/WinfriedL/pandapower/commit/ec594088bde5d94c0bda12b42299596bace20f4c and a temporary (ugly) fix in https://github.com/WinfriedL/pandapower/commit/94838e4adf4984e8ea805f531c63e0356053ffbd

ascheidl, 2018-11-05:
In you fix, can you please alter net[element][net[element]['in_service']].index.values in net[element].index.values[net[element]['in_service'].values] and np.sum(net['gen']['in_service']) in np.sum(net['gen']['in_service'].values) for performance reasons and open a pull request ?

WinfriedL, 2018-11-06:
Yes of course, I will do that (later).
I was not sure if that fix is the right way to fix the problem, it was more meant as a proof of concept.

WinfriedL, 2018-11-08:
I have opened the pull request. It is only tested for explicit generators so far. I expect the same issue exists at least for xwards as well.

lthurner, 2018-12-04:
I completely refactored how generators are added. There was a lot of copy pasted code in build_gen.py between powerflow and OPF that I replaced with one consistent function that works for OPF as well as PF. I also refactored the way the pd2ppc lookups for generators are created.

With regards to isolated generators: the assumption for creating the lookup was, that all generators are in service (because out of service generators wouldn't be added to the gen matrix in the first place). When the connectivity check set some generators out of service, the mapping of the generators was messed up. I solved this by moving the _build_gen function after the connectivity check in pd2ppc:

https://github.com/e2nIEE/pandapower/blob/a4c6f05866947e34b42b9ed843da659b41c5ffc9/pandapower/pd2ppc.py#L80-L110

So now the connectivity check is performed first and disconnected generators are marked as out of service before the gen matrix is initialized. Disconnected generators are then treated the same as out of service generators, in that they aren't added to the gen matrix in the ppc in the first place. This should solve the mapping issues described here without any workarounds. I also added a test with isolated xwards in test_scenario, and everything seems to work fine.

closed 2018-12-04 by lthurner