e2nIEE/pandapower #192: Tap range in Synthetic Voltage Control LV Networks

thomaswolgast, 2018-10-18:
I evaluate voltage control concepts in distribution grids. For that, I use the Synthetic Voltage Control LV Networks which are included in "pandapower.networks". But I noticed that the range of the transformer's tap positions is very small (from -2 to +2). Why were these values chosen?  Or where do they come from?

Were that values chosen by Mr. Lindner who wrote the paper where the networks were worked out? Or were they chosen by the pandapower team? If so, on what basis? Are these small ranges normal in low voltage networks?

It is important for me, because the small range of tap positions hinders my algorithms from working. And I need more information to decide if I can change the allowed range by myself or if that would distort my results. Of course I want to use preferably close to reality networks.

Thank you very much

SteffenMeinecke, 2018-10-23:
The paper which is the origin of these grids do not give complete information about the transformer parameter. Therefore, most parameters, not sn_kva or vsc_percent, are derived from standard pandapower types. Because of the reasons which are already mentioned by @ascheidl No Load Tap Changers are assumed here. But there are certainly reasons for deviating from these assumptions for specific investigations.

thomaswolgast, 2018-10-23:
Thank you very much. That helps me.

closed 2018-10-23 by thomaswolgast