e2nIEE/pandapower #201: Falsche Zuweisung in res_bus?

ZhengLiu1119, 2018-11-02:
Beispiel case39: Knoten Name 23 (index 22) setze ich out of service. Hinter diesem Konten liegt ein Konten (Beispiel: index 35) mit Generator (Index 5), der für runpp abgetrennt erscheinen müsst. 

Was bei runpp passiert, ist jedoch, dass die Knoten, die zu dem in net.gen hinter dem Eintrag des isolierten Generators stehenden Generator gehören, geändert p/q-Leistungswerte bekommen. 

++++++++++++++++++++      Beispiel    +++++++++++++++++++
import pandapower as pp
import pandapower.networks as pn

net = pn.case39()
net.bus.in_service[22]=False
pp.runpp(net)
print(net.res_bus.loc[22])
print(net.res_bus.loc[35:38])
print("############ nachher ###########")
net.bus.in_service[35]=False   
pp.runpp(net)
print(net.res_bus.loc[22])
print(net.res_bus.loc[35:38])
+++++++++++++++++++++++++++++++++++++++++++++++++

Unterschiede gibt es dann bei den p/q-Werten der Knoten-Indizes 36...38. Also p/q-Werte kommen zu falschen nachfolgenden Konten.

WinfriedL, 2018-11-02:
Might be related to#191.
What happens if you add pp.set_isolated_areas_out_of_service(net) before running the powerflow?

ZhengLiu1119, 2018-11-02:
> Might be related to #191.
> What happens if you add pp.set_isolated_areas_out_of_service(net) before running the powerflow?

Hey, 
thanks for your replay. With the "pp.set_isolated_areas_out_of_service(net) " both powerflow are identical to each other.

WinfriedL, 2018-11-02:
Then I guess it is indeed related to #191, I observed the same behaviour.

WinfriedL, 2018-11-08:
I tried it with #204, and I think the results are correct with that fix.

closed 2018-11-09 by lthurner