e2nIEE/pandapower #208: Changing the signing system for generators

lthurner, 2018-11-12:
pandapower was traditionally developed as a tool for distribution system analysis, so that our initial decision was to keep all power system values from the load viewpoint. This wasn't a bad decision at the time, but since pandapower has moved to higher voltage levels, this is becoming increasingly impractical. For example, constraints for a generator with a maximum power of 1MW have to be defined as min_p_kw=-1000 and max_p_kw=0. Because of the negative sign of generated power, costs also have to be entered as negative values.

All this is obviously not very intuitive, but of course changing this would not be backwards compatible and break a lot of code, so that we have not tackled this problem. This issue now came up again with the PowerModels interface #207, and it might be time to adress this issue even if it will make a major release necessary.

The suggestion is to change the power values in gen, sgen and ext_grid to the generation viewpoint and leave everything else like it is now. Of course the definition of constraints and cost values would also have to be adapted accordingly.

maddinonline, 2018-11-13:
What about storages?
They would be the only element with negative active power "allowed" then. Otherwise there needs some work to be done to distinguish their load and generation behaviour.

lthurner, 2018-11-13:
There are some elements that can have both negative and positive active power. Storage is one of them, but there are also shunts, wards/xwards and ext_grids. To have as least changes as possible, I would only change the signing system in the cases where it is clear that it should behave like a generator (which from my perspective is gen, ext_grid and sgen, but of course this is up for debate) and leave all other elements as they are now.

lthurner, 2018-11-13:
I created a branch dev_2.0 where I started to implement this. Obviously many tests will fail because the signing system changed, but I already got all the power flow tests to pass again, shouldn't be a huge issue to get the rest of the tests to pass again also.

lthurner, 2018-11-15:
I now changed the signing system for sgens, ext_grids and gens, and all tests are passing again. This means costs can now be defined with positive values for generators, ext_grids and sgens. For loads or storages, the signing system is still the same as before. With positive costs, higher power values will be penalized, so that the OPF would try to minimize the storage/load power. 

So to get the same behaviour as before, this:
https://github.com/e2nIEE/pandapower/blob/19b4af5662bbd1634fa1fe0a8ad0fbfc49178818/pandapower/test/opf/test_basic.py#L585-L588

is now changed to:

https://github.com/e2nIEE/pandapower/blob/3ba09d64e6ed6d50dce739715c08bf90b3647d41/pandapower/test/opf/test_basic.py#L586-L589

@maddinonline (and everyone else), please check out the new version and test if this implementation makes sense

lthurner, 2018-11-26:
Implemented on branch dev_2.0, all tests are passing again. Please reopen if there is some issue with this

lthurner, 2018-11-26:
Reopened because documentation has not been adapted yet

WinfriedL, 2018-12-13:
> I would only change the signing system in the cases where it is clear that it should behave like a generator

Just from my point of view, I see wards and especially xwards more like generators. At least the way they are used in our cases is more like generators. And since an extended ward is internally modeled as a generator maybe it would make sense to change those too. 

Since I don't use storage elements I do not really have a strong opinion there, but intuitivly I would treat them like generators as well. If I model a storage, I always use gens or sgens...

But it is not clear by definition so either way would be fine, as long as it's documented.

lthurner, 2018-12-13:
> Just from my point of view, I see wards and especially xwards more like generators. At least the way they are used in our cases is more like generators. And since an extended ward is internally modeled as a generator maybe it would make sense to change those too.

Yes I can see that point. What I don't like about it is that the shunt power of the ward/xward would then have the opposite sign compared to the power of a regular shunt element (which I believe should stay in load signing system, since it is typically a passive element). But there is probably nothing that can be done about that.

>Since I don't use storage elements I do not really have a strong opinion there, but intuitivly I would treat them like generators as well. If I model a storage, I always use gens or sgens...

>But it is not clear by definition so either way would be fine, as long as it's documented.

As long as there is no real argument for changing the signing system I would go with the current rule.

miek770, 2018-12-13:
If I may add an outsider's perspective:

1. Having negative power for generators is counter-intuitive;
2. Having generators and motors share the same table (DataFrame) is also counter-intuitive, and leads to either being forced to have a negative power for the *normal* flow of power.

If pandapower's focus really is transmission systems, then item 2 appears acceptable (i.e. keeping gens and motors in the same table), and favoring a signing system that leads to generators having positive power would be best.

If pandapower's focus includes distribution / industrial systems, then my humble opinion is that motors should ideally be split into their own category, with positive power for consumption.

So either way, the initial suggestion would still be an overall improvement. It just feels like motors are not *officially* supported but can easily be *hacked* by reversing a generator:

> The suggestion is to change the power values in gen, sgen and ext_grid to the generation viewpoint and leave everything else like it is now.

WinfriedL, 2018-12-13:
>What I don't like about it is that the shunt power of the ward/xward would then have the opposite sign compared to the power of a regular shunt element (which I believe should stay in load signing system, since it is typically a passive element)

I think I am convinced ;)

WinfriedL, 2018-12-13:
> Having generators and motors share the same table (DataFrame) is also counter-intuitive, and leads to either being forced to have a negative power for the normal flow of power.

Maybe it's my background, but for me the sign is kind of arbitrary anyway, I have no problem modelling loads as static generators with negative power, or loads as sgens with negative power. In the end it is just one bus value in the ppc matrix.
For dclines I use generators with positive and negative active power anyway.

And from that point of view, splitting all that in too many different dataframes just adds inconvinience for analyizing the network because there are more tables where loads/power injections  can be in.

But I never dealt with opf formulation or shortcircuits, so maybe the sign has more meaning there.
And I can also imagine that for 'smaller' distribution / industrial systems it makes more sense to distinguish between the elements.

lthurner, 2018-12-13:
> That's exactly it, for distribution and industrial systems, there are a lot of motors and they are treated differently in short-circuit analysis and different types of power flow studies. But if pandapower's focus really is the transmission level, then please ignore this comment.

Actually pandapower was originally developed specifically for power flow calculations in distribution systems. We have since included other analysis methods (OPF, state estimation, short-circuit analysis) and added support for transmission system elements (trafo3w, xward, gens etc.). We are now correcting some design choices in pandapower 2.0, so that pandapower can be better used for transmission system analysis and optimization, but distribution systems analysis is still one of the main use cases of pandapower.

>If pandapower's focus includes distribution / industrial systems, then my humble opinion is that motors should ideally be split into their own category, with positive power for consumption.

The reason we don't yet have this, is that in a power flow study it is usually just modeled as a load, so a seperate element wasn't necessary. You are right however that in short-circuit calculation the motor modeling is not very intuitive. Until now, there just wasn't any major push to introduce a motor element, but I absolutely see your point of view. I see that this is related to signing systems, but it is still a different topic, so can you please open up a seperate issue for us to discuss this?

> But I never dealt with opf formulation or shortcircuits, so maybe the sign has more meaning there.

Not only the sign, but also the type of element (a motor behaves differently in a short-circuit than a household load or a generator).

> And I can also imagine that for 'smaller' distribution / industrial systems it makes more sense to distinguish between the elements.

Yes I think that is the differnce in point of view: in transmission systems most of the loads/generators are an aggregation of underlying elements anyhow, so it doesn't make much sense to differentiate between load/sgen/storage/motor.  But if you model a specific element that exists in a low-voltage grid in detail, it makes a lot more sense to differentiate the elements.

Ultimately, we want pandapower to be able to be used in transmission systems as well as distribution systems. We always have to balance between adding functionality that is needed and not adding to much features so that clarity is lost and the tool gets to crowded and complicated. To that end, these discussions are very helpful to understand the different use cases.

lthurner, 2018-12-13:
> I think I am convinced ;)

One more thought: are xwards also used for optimization, for example in an OPF that the reactive power would be optimized? Because then the Q-constraints would have to be defined in load signing system for xwards, but in generator signing system for gens, which would also be confusing. I don't think that is currently supported by the OPF in pandapower, but it could be easily included.

WinfriedL, 2018-12-13:
>I don't think Q-constraints for xwards are currently supported by the OPF in pandapower, but it could be easily included.

Well, I have never seen limits for xwards. In our cases they are mostly added to supply unlimited amount of reactive power.

miek770, 2018-12-13:
> I see that this is related to signing systems, but it is still a different topic, so can you please open up a seperate issue for us to discuss this?

Done: #244. Thanks for considering this.

closed 2019-02-19 by lthurner

marhofmann, 2019-07-11:
Please also change the description of the signing system under "Unit System and Conventions" 

[https://pandapower.readthedocs.io/en/v2.1.0/about/units.html](url)

The signing system of gen, sgen, and ext_grid is described as follows:
"There are three elements that have power values based on the generator viewpoint (positive active power means power consumption), which are (...)"

This is quite confusing!

gvdwel, 2019-11-27:
I am confused by the documentation as well. For example:

https://pandapower.readthedocs.io/en/v2.1.0/elements/sgen.html

"All elements in the grid are modelled in the consumer system, including generators! If you want to model the generation of power, you have to assign a negative active power to the generator. Please pay attention to the correct signing of the reactive power as well."

Is this correct or outdated?