e2nIEE/pandapower #209: pandapower 2.0

lthurner, 2018-11-12:
pandapower has up until now mostly been about power system analysis. While the OPF interface to PYPOWER allowed for some optimzations, this was not a major focus of pandapower. We want to change this in the future, but this will make some major changes necessary. This is why the next release is probably going to be a major non-backwards-compatible release. This should include the following isssues:

- [ ] Include an Interface to PowerModels #207 
- [ ] Change the signing system of generators #208 
- [ ] We didn't persue this further #73 because it seemed to big of a change, but if we are already making a non-backwards compatible release, it might make sense now? In any case we should make a final decision on this
- [ ] Change some names of variables, like tp_pos --> tap_pos (one letter more and easier to read) or vsc_percent --> vk_percent?
- [ ] Include an converter that allows to import pypsa grids to allow to get some models that are made for optimization (e.g. the [EUR-30](https://www.pypsa.org/animations/pypsa-eur-30/) example or generic distribution grids from [ding0](https://github.com/openego/ding0))

Any more changes that might make sense, but we put them off because we didn't want to make incompatible changes?

ccoffrin, 2018-11-12:
You might have a look at Github's milestone feature.  It's great for tracking collections of issues leading up to a release.

jhmenke, 2018-11-21:
if we are doing incompatible changes, i would like to update to a new measurement structure, which i already use for quite some time internally. it is more descriptive, especially the "side" column ("from" / "to" for lines, "hv"/"mv"/"lv" for transformers)

> [("name", dtype(object)),
>                 ("measurement_type", dtype(object)),
>                 ("element_type", dtype(object)),
>                 ("element", "uint32"),
>                 ("value", "float64"),
>                 ("std_dev", "float64"),
>                 ("side", dtype(object))]

instead of the previous:

> 
> [("name", dtype(object)),
>                         ("type", dtype(object)),
>                         ("element_type", dtype(object)),
>                         ("value", "f8"),
>                         ("std_dev", "f8"),
>                         ("bus", "u4"),
>                         ("element", dtype(object))]
> 

anyone against it?

friederikemeier, 2018-11-21:
Regarding those trafo parameters, I have the suggestions that we add a check into the diagnostic function, that checks if the tap parameters are set correctly. If I forget to define e.g. tp_side, the tap is completely ignored in the powerflow.

lthurner, 2018-11-26:
> anyone against it?

We had "element_type" in switch, but changed it to "et" because element_type was producing long lines in the code. In the new cost functions, there is also an "et" column. So I would suggest also naming it "et" instead of element_type to be consistent. Other than that, go ahead and add the changes on the dev_2.0 branch (but don't forget to include it in the convert_format function). 

> Regarding those trafo parameters, I have the suggestions that we add a check into the diagnostic function, that checks if the tap parameters are set correctly. If I forget to define e.g. tp_side, the tap is completely ignored in the powerflow.

:+1:

jhmenke, 2018-11-26:
> We had "element_type" in switch, but changed it to "et" because element_type was producing long lines in the code. In the new cost functions, there is also an "et" column. So I would suggest also [...]

okay, will do it this week.

closed 2019-02-19 by lthurner