e2nIEE/pandapower #211: cost function definition

lthurner, 2018-11-13:
Currently the cost function is defined like this:

> pp.create_polynomial_cost(net, 0, 'ext_grid', np.array([-10, 0]), type=p)
   pp.create_polynomial_cost(net, 0, 'ext_grid', np.array([-1, 0]), type=q)

Apart from the signing system #208, this isn't really in line with the rest of pandapower. How about something like this instead:

> pp.create_polynomial_cost(net, 0, 'ext_grid', c_eur_per_kw=10, c_eur_per_kw2=0, c_eur_per_kvar=1, c_startup_eur=100)

friederikemeier, 2018-11-13:
I think that is a extremely more user friendly definition.

friederikemeier, 2018-11-13:
If n is the number of generators in the ppc, the gencost array for reactive power starts in row n+1 of ppc["gencost"]. So for p and q, the conversion is exactly the same (except some signing system madness).
That's what the shift_idx variable in pandapower/opf/make_objective.py is for:

`                    if cost_type == "q":
                        shift_idx = ng
                        sign_corr = -1
                    else:
                        shift_idx = 0
                        sign_corr = 1`

closed 2019-02-18 by lthurner

Misbahrani, 2019-07-04:
hello
Kindly tell me how to apply the n-polynomial cost function for n=2:
fpol(q)=c2q2+c1q+c0
in the given format
pp.create_polynomial_cost(net, 0, 'sgen', np.array([0,1, 0]) is given
if I go for np.array[1 1 1] then how to incorprate c2, c1, c0

commit de47185c (2018-11-21): pandas legacy fix #211