e2nIEE/pandapower #214: Transformer voltage regulator

miek770, 2018-11-18:
Is there any plan to add transformer voltage regulators in the future? It doesn't appear to be in the [2.0 milestone](https://github.com/e2nIEE/pandapower/issues/209) and I didn't find any mention of it in the documentation.

It's a simple control loop, we added it to GridCal [here](https://github.com/SanPen/GridCal/blob/8116566b749966e47fd12949e10df3c45e9ba937/UnderDevelopment/GridCal/Engine/PowerFlowDriver.py#L835). The downside is that it has to happen in an outer control loop because the power flow has to be fully resolved before the voltage regulator is allowed to take action on the tap changer.

It could also easily be hacked around pandapower, but it's such a useful and common device that it seems like it should be part of the base code.

I can prepare a PR for the 2.0 branch if you agree with its addition.

Thanks.

WinfriedL, 2018-11-19:
Just as a comment: As far as I know it does not have to be added to an outer control loop, it should be possible to add them directly in the power flow solution by adding the derivatives of the voltages with respect to the tap positions.
The drawback would be that the powerflow would consider continuous and unlimited tap  positions and one needs to round (and limit) them afterwards. 

So depending on the situation one would still need at least two powerflows, one with continuous/unlimited tap positions and afterwards one with rounded and fixed tap positions.

miek770, 2018-11-19:
Indeed, I was thinking about this last night and it can definitely be reduced to 2 (or 3) power flows. Either by using the method you described or by skipping the intermediary taps in an outer loop to get to the final tap right way.

lthurner, 2018-11-26:
> It could also easily be hacked around pandapower, but it's such a useful and common device that it seems like it should be part of the base code.

We view this as an application of pandapower until now, because there can be lots of different implementations of such controllers (e.g. local/decentral/central voltage control, continous/discrete tap changers etc.), so that there is not one universally valid "tap changer" function. Thats why we do not view it as part of the pandapower core code for now, but we are discussing this further internally.

So instead of a Pull Request maybe sharing a possible implementation of this as a code snippet on gists as a starting point for other users would be a better place for this?

> As far as I know it does not have to be added to an outer control loop, it should be possible to add them directly in the power flow solution by adding the derivatives of the voltages with respect to the tap positions.

Yes that would be a clean solution, so if someone would implement this, a PR would be very welcome.

miek770, 2018-11-26:
Thanks for the response. I don't entirely agree with the comment about different implementations, but it's great you're looking into it internally. I believe taps are always discrete, but I understand that the control loop implemented by the voltage regulator may be variable in complexity (ex.: single setpoint; line drop compensation; centralized control; enhanced VAR capability for unit transformers [see this paper for ex](https://www.researchgate.net/publication/3659581_Capability_chart_for_generator-transformer_units)).

I would recommend implementing the standard behavior (i.e.: single setpoint + option for LDC) in the base code as @WinfriedL suggested, and leaving complex controls to individual applications like @lthurner suggested above.

lthurner, 2018-11-26:
> I believe taps are always discrete

Yes real life taps are always discrete, but it is common to assume continous steps for studies in my experience. For example in studies where a load/generation scenario is assumed for +20 years, the voltage profile could look very different depending on if you are just below the deadband of the transformer or just above. Since you can't accurately forecast the tap position in 20 years anyway, it can be more realistic to allow continous steps to not get large differences in the results for different scenarios depending on the tap changer position.

miek770, 2018-11-26:
I see, so you could have a realistic tap changer for power system studies, and an ideal (continuous) tap changer for long-term energy studies. I get its use for non-linear OPF if it reduces simulation time.

Thanks.

lthurner, 2019-09-02:
This is now included in the control module. See continous tap control: https://github.com/e2nIEE/pandapower/blob/develop/pandapower/control/controller/trafo/ContinuousTapControl.py and discrete tap control: https://github.com/e2nIEE/pandapower/blob/develop/pandapower/control/controller/trafo/DiscreteTapControl.py. Tutorials will follow, see #490

closed 2019-09-02 by lthurner