e2nIEE/pandapower #216: pp.runpp fails with "Generators with different voltage setpoints connected to the same bus", BUT all setpoints are equal in grid model.

vogt31337, 2018-11-23:
Hi,

in build_gen.py (Line 463) an equality check is made. But due to some conversions made before, this check fails:
``` python 
values = [1.00999999 1.00999999 1.00999999 1.00999999 1.         1.01 1.        ]
values_equal = [1.00999999 1.00999999 1.00999999 1.00999999 1.         1.00999999 1.        ]
```

Attached is the problematic grid in pickle, using pandapower 1.6.0 develop commit b7136d72ca66a1fcfdcf2460d40c35dac38f02a0 and python 3.7

```
Traceback (most recent call last):
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\run.py", line 294, in runpp
    _powerflow(net, **kwargs)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\powerflow.py", line 66, in _powerflow
    ppc, ppci = _pd2ppc(net)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\pd2ppc.py", line 114, in _pd2ppc
    _check_voltage_setpoints_at_same_bus(ppc)
  File "C:\Anaconda3\envs\py37\lib\site-packages\pandapower\build_gen.py", line 437, in _check_voltage_setpoints_at_same_bus
    raise UserWarning("Generators with different voltage setpoints connected to the same bus")
UserWarning: Generators with different voltage setpoints connected to the same bus
```

BR V3

jhmenke, 2018-11-23:
Try replacing

`return not np.array_equal(values, values_equal)
`
with 

`return not np.allclose(values, values_equal, 1e-7)`

lthurner, 2018-11-26:
@jhmenke  can you make a PR for this?

PR #221 (2018-12-04): fix #216
numpy's array equal had and has some issues (checking address of object instead of contents, nan != nan ...). using allclose ensures the contents of two arrays are equal with a precision.

this commit fixes #216 

i will cherry-pick the from_dill / to_dill, please don't merge yet.
closed 2018-12-04 by jhmenke