e2nIEE/pandapower #224: Drop Python2.7 support

jhmenke, 2018-11-30:
Python3 has a number of interesting features, which are currently not used because we want to keep Python2 compatibility. I propose a discussion whether we should drop Python2 support and i suggest to do this in pandapower 2.0 due to other compatibility-breaking changes.

Advantages:
* type hints can make development easier. the code will be more readable for new developers
* async library for concurrent code (e.g., asynchronous generators)
* easier string formatting with f strings
* Python2 support ends 2020 anyway, for security reasons alone the switch to Python3 should occur in 2020 at the latest
* no additional work required since we are already Python3 compatible

Disadvantages:
* potential Python2 users are forced to update their codebase

What do you think?

lthurner, 2018-12-04:
> easier string formatting with f strings

this was only introduced in Python 3.5, so we can't use that even if we drop support for 2.7

Apart from that, I agree with this. Python 2 will only be supported for [one more year](https://pythonclock.org/), and pandas already drops their Py2 support [next month](https://www.analyticsvidhya.com/blog/2018/04/pandas-will-drop-all-support-for-python-2-next-year/). So either way there won't be support for Python 2 much longer, and if we are already planning a major release this seems to be good timing to migrate to Python 3 only.

miek770, 2018-12-04:
F-strings were actually added in Python 3.6.

But out of curiosity, is there a specific reason why you'd support Python 3.5 and lower? I've seen the argument that 3.5 was still the default version for some devices such as Raspberry Pis, is this the case here?

jhmenke, 2018-12-05:
> F-strings were actually added in Python 3.6.
> 
> But out of curiosity, is there a specific reason why you'd support Python 3.5 and lower? I've seen the argument that 3.5 was still the default version for some devices such as Raspberry Pis, is this the case here?

i don't think there is a reason other than a broad compatibility

lthurner, 2018-12-05:
Yes, there might be users that can't upgrade to Python 3.6 because they use a package that is not available in 3.6 or they don't have the possibility of easily upgrading (e.g. in a corporate setting where the installation is provided by IT). It doesn't seem worth it to deny these users access to pandapower just because of f-strings. That's also the reason we support Py2 until now, but it seems like people will be forced to switch now anyhow.

jhmenke, 2019-01-09:
pandas is also dropping python2 support: https://twitter.com/jakevdp/status/1080583192803823616

lthurner, 2019-01-28:
pandapower 2.0 will only support Python 3, travis was adapted accordingly

closed 2019-01-28 by lthurner