e2nIEE/pandapower #225: PQ capability load flow possible

hariramtud, 2018-12-02:
Is it possible to do a complete PQ capability of static generator at the grid connection point as load flow at different nominal voltage with pandapower ?

lthurner, 2018-12-04:
I am not quite sure what you mean by that. Are you talking about a continuation power flow?

hariramtud, 2018-12-04:
Hi
I have a PQ curve capability of renewable generators like PV inverters,
which are mainly symmetric PQ curve.

At the grid connection to check the grid code requirement to see if there
are enough reactive power etc, I want to check how the entire PQ shifts I
want to do a load flow with all inverters, cable and Trafos to see the PQ
capability of the power plant at the grid connection.
I think this means the panda power takes in as input generator PQ and
provide PQ capability of entire plant at grid connection.

As an example I am attaching a required pq at grid connection to simulated
capability. This was done with Digsilent. Want to know if this can be done
with Panda power.

Thanks a lot
Best regards
Hariram

lthurner, 2018-12-04:
You will have to upload the attachement directly on github, the attachement from mail isn't processed.

From what you are describing, this sounds like you would only need to model your inverters as sgen elements in pandapower, change the P/Q values of the sgens according to the PQ curve, run a power flow and check the resulting Q value at the external grid connection.

miek770, 2018-12-04:
I also have this requirement (reactive power capability demonstration for grid interconnection studies) and from what I understand from your description, there are two main issues:

1. Generators have either a 2 or 3 dimensional capability curve (active power and sometimes voltage dependant), but never a static limit as implemented in most open-source software (AFAIK). Commercial softwares usually support this.

2. You want to trace the complete farm's capability curve as seen at the POI.

I think you'll have to write a script that generates the required simulations, calculate reactive power limits for your generators depending on their curve(s), and show the resulting reactive power at the POI for each simulation. That's what I do. It gets especially tricky when your generators have a voltage dependant reactive power curve, but that's the beauty of working with Python+OSS: you can do it!

PS.: I don't see any attachment, perhaps a link to Dropbox or other would work better.

hariramtud, 2018-12-04:
@miek770 : You have nailed it. Thats exactly i want to do in the load flow studies. 
Can you tell me a bit more about the Python and OSS.

You can see those requirements for PQ compliance at grid connection point for example in Germany with the grid code VDE AR 4120 or VDE AR 4110 and in variety of country code. 

How can i do this kind of analysis with pandapower. Then this will be very useful to check grid compliance of a power plant.

hariramtud, 2018-12-04:
you can see some example in the following link on the PQ curves
[](https://www.researchgate.net/figure/P-Q-capability-curve-for-one-inverter_fig4_276268238)

hariramtud, 2018-12-04:
[https://ieeexplore.ieee.org/document/8301197](https://ieeexplore.ieee.org/document/8301197)

hariramtud, 2018-12-04:
on in this link some PQ curve ( for those who cannot access IEEE )
[https://www.researchgate.net/figure/P-Q-capability-curve-for-one-inverter_fig4_276268238](https://www.researchgate.net/figure/P-Q-capability-curve-for-one-inverter_fig4_276268238)

hariramtud, 2018-12-04:
> You will have to upload the attachement directly on github, the attachement from mail isn't processed.
>
> From what you are describing, this sounds like you would only need to model your inverters as [...]

But i need to give the P/Q as a matrix or array or set of PQ points, and I get only one value of resulting Q at the external grid connection and not a matrix or vector of data.

miek770, 2018-12-04:
Hi @hariramtud,

By Python + OSS, I meant open-source software. For example, using pandapower makes it easy to expand it using custom scripts. I regularly use commercial softwares and the lack of API makes adaptations... possible but far less elegant.

What @lthurner and I meant is that you have to automate the evaluation of each point of your capability curve at the POI (point of interconnection, aka point of common coupling) yourself using an external Python script. Something like this (**pseudocode**, for 2-dimensional capability curve):

```
x = 0 to 6 MW
q_min = empty list
q_max = empty list
for p in x:
    q_min += run_pandapower( q_static_generators = min_q_at_p() ).get_q_at_poi()
    q_max += run_pandapower( q_static_generators = max_q_at_p() ).get_q_at_poi()
plot(x, q_min, q_max)
```

Sorry for the messy pseudocode, I think it the first time I use this in at least 10 years!

If you need to consider voltage in the Q limit determination (3-dim curve), then it gets a bit more complicated but not insurmountable. You'll have to iterate for each solution though (or some, depending on your grid).

For plotting, I recommend looking here: [https://python-graph-gallery.com/](https://python-graph-gallery.com/). I find it really helpful, but don't forget to add the show() statement at the end to make it visible, and add the inline statement if you're in a Jupyter notebook:

```
%matplotlib inline
import matplotlib.pyplot as plt
```

It would be nice to have a builtin way to do this, but I think there are too many variables. We're looking into implementing capability curves in [GridCal](https://github.com/SanPen/GridCal) but we've only had preliminary discussions about it so far. And from my experience different states have vastly different reactive power capability requirements, so the same methodology won't suit all of them. Still, if you make a nice script/Jupyter notebook to generate the curve, it could serve as a great example for others.

Hope this helps.

Michel

hariramtud, 2018-12-10:
thanks a lot for your feedback. 
I dont think the "runpp" ( or run pandapower ) can take the arguments as you have mentioned.

I think everytime the P and Q data have to be initialized for each iterations in the create_sgen. 
But this creates a new static generator.

I also tried some methods but this does not work as i thought. Any idea from anyone on this ? 

p = [ 1*400.51e+3, 0.95*400.51e+3, 0.8*400.51e+3, 0.6*400.51e+3, 0.4*400.51e+3, 0.2*400.51e+3, 0.1*400.51e+3]
q = [ 0.228*400.51e+3, 0.33*400.51e+3, 0.33*400.51e+3, 0.33*400.51e+3, 0.33*400.51e+3, 0.33*400.51e+3, 0.1*400.51e+3]
for p_value, q_value in zip(p, q): 
    pp.create_sgen(net, bus4, p_kw=-1*p_value, q_kvar=-1*q_value,  type="PV",  name = "PVInverter", in_service = True, controllable=True)
    pp.runpp(net, algorithm="gs",calculate_voltage_angles='true')
    resultsQ.append(net.res_ext_grid.q_kvar[0])
    resultsP.append(net.res_ext_grid.p_kw[0] )

plt.figure(1)
ax.plot(resultsQ, resultsP)
ax.set(xlabel='q', ylabel='p',title='pq')
ax.grid()
fig.savefig("test.png")
plt.show()
        

Thanks

lthurner, 2019-02-19:
Closed for inactivity

closed 2019-02-19 by lthurner

hariramtud, 2019-09-08:
has this datastructure " net.sgen.p_kw.at[sgen_index] = -p_value"   changed ?.
I tried to change .p_kw to .p_mw ( respectively for .kvar to .mvar ) ; 
But gets a convergence error. 
IT used to work previously with the above sample code.

Abdullah2969, 2023-08-24:
ha

> Hi I have a PQ curve capability of renewable generators like PV inverters, which are mainly symmetric PQ curve. At the grid connection to check the grid code requirement to see if there are enough [...]

Hi Mate,
          It seems you have done the reactive power capability studies at the pcc on DidSCILENT powerfactory
          I am a master's student and currently working on it it would be really help full if you could share the pfd file with me