e2nIEE/pandapower #226: from_json is not working as intended

jhmenke, 2018-12-03:
On current pandapower-develop, run the following code in debug mode:

```
import pandapower.networks as ppn
grid = ppn.mv_oberrhein()
```

With many grids (maybe all?), the *from_json*-function will fall back to the deprecated function *from_json_dict*. This is because in *dicts_to_pandas* there will be a UserWarning raised because the pandapowerNet contains some elements, which are not dicts (e.g., strings, int):
https://github.com/e2nIEE/pandapower/blob/f01d0efea8b6d46146077d2ea64c18142eb15291/pandapower/io_utils.py#L95

I suspect the issue arose due to some refactoring in *from_json*? I think *dicts_to_pandas* should only be called for the actual dicts?
@rbolgaryn can you take a look?

Another issue is that, by default, the warnings will not be shown to the user. I only noticed this by looking at tests for an entirely different issue.

rbolgaryn, 2018-12-19:
Part of it is old code. It seems it works but it does not look very reliable. 
Actually, the function to save net to json is deprecated, too. 

It is not necessary to iterate over DF anymore, it's enough to have json.dump(net, path, cls=PPJSONEncoder). So, the pp.to_json can be simplified.

The loading would be net = json.load(filename, cls=PPJSONDecoder)
from_json can be simplified to just have this line and "convert_format".
It would not work until grids are resaved again, that's why a from_json_dict function should still be kept in a try block.

I suggest simplifying the to_json and from_json, as well as removing the code for loading of json files from the old implementation (dicts_to_pandas, from_dict_of_dfs), unless we want backwards compatibility in case someone still has old json files from before.

rbolgaryn, 2019-01-11:
Actually, dict_to_pandas is deprecated, and not from_json_dict

jhmenke, 2019-01-11:
Are you sure? Then it works correctly, but the call to dict_to_pandas is unnecessary and should be deleted.

rbolgaryn, 2019-01-11:
as well as dicts_to_pandas

rbolgaryn, 2019-01-11:
I am changing the to_json function, so that the file has "_module", "_class", "_object", and then it will be loaded as pandapowerNet already. But I'll leave the part of "from_json_dict" for older files.

rbolgaryn, 2019-01-11:
I am removing the "convert" argument from "from_json_dict", because "convert_net" is then done in from_json anyways

jhmenke, 2019-01-18:
pull request was merged, so this can be closed.

closed 2019-01-18 by jhmenke