e2nIEE/pandapower #231: Explanation misaligned for generator input in doc

miek770, 2018-12-07:
Hi,

The generator inputs explanation as misaligned in the latest doc: [https://pandapower.readthedocs.io/en/latest/elements/gen.html](https://pandapower.readthedocs.io/en/latest/elements/gen.html)

In the **Input Parameters** table, `rdss` is explained to be *Rated voltage of the generator*, for example.

Also, `xdss` and `rdxx` don't follow the [Unit conventions](https://pandapower.readthedocs.io/en/latest/about/units.html). They could be either in percent or in per unit, so I would rename them to show the unit.

Thanks,

Michel

lthurner, 2018-12-07:
You are right about both of this, thanks for reporting. I fixed it on the dev_2.0 branch, see: https://pandapower.readthedocs.io/en/dev_2.0/elements/gen.html.

The unit of the resistance/reactance values are per unit, so I added the unit to the parameter name.

closed 2018-12-07 by lthurner

miek770, 2018-12-07:
Great thanks!