e2nIEE/pandapower #233: Bus sometimes not showing on plot

miek770, 2018-12-08:
Hi,

I have a weird issue with 2 buses sometimes not showing on my plots for a particular network. They are definitely modelled and active; they are considered in the short circuit calculation and show up in the Excel network export. The code is all in a Jupyter notebook, and ~80% of the time it doesn't show these buses, but ~20% they show up. I use the "Restart the kernel and rerun the whole notebook" each time.

Here's the plot with all buses: [https://drive.google.com/open?id=1Wty-yaFP6xiFEyHSB73sLlza3Hpl3w3m](https://drive.google.com/open?id=1Wty-yaFP6xiFEyHSB73sLlza3Hpl3w3m)

Here's the plot with missing buses: [https://drive.google.com/open?id=1pZfacIdvaVlpwbmLofqGpB1LC6IHbPHN](https://drive.google.com/open?id=1pZfacIdvaVlpwbmLofqGpB1LC6IHbPHN)

I could share my notebook but with a limited number of people; it's for a private project. Considering this, how could I share it with the appropriate people? But basically I call:

```
import pandapower.topology as topo
from pandapower import plotting
mg = topo.create_nxgraph(net)
net = plotting.create_generic_coordinates(net=net,
                                          mg=mg,
                                          library="igraph",
                                          respect_switches=True)
plotting.plotly.simple_plotly(net)
```

Sometimes I get this error (so no plot at all), sometimes I get a similar one that complains about width instead of height:
```
ValueError                                Traceback (most recent call last)
<ipython-input-16-148b1eae6409> in <module>
----> 1 plotting.plotly.simple_plotly(net)

C:\ProgramData\Anaconda3\lib\site-packages\pandapower-1.6.0-py3.6.egg\pandapower\plotting\plotly\simple_plotly.py in simple_plotly(net, respect_switches, use_line_geodata, on_map, projection, map_style, figsize, aspectratio, line_width, bus_size, ext_grid_size, bus_color, line_color, trafo_color, ext_grid_color)
    166 
    167     draw_traces(line_traces + trafo_trace + ext_grid_trace + bus_trace,
--> 168                 aspectratio=aspectratio, figsize=figsize, on_map=on_map, map_style=map_style)

C:\ProgramData\Anaconda3\lib\site-packages\pandapower-1.6.0-py3.6.egg\pandapower\plotting\plotly\traces.py in draw_traces(traces, on_map, map_style, showlegend, figsize, aspectratio, filename)
    643 
    644         aspectratio = np.array(aspectratio) / max(aspectratio)
--> 645         fig['layout']['width'], fig['layout']['height'] = ([ar * figsize * 700 for ar in aspectratio])
    646 
    647     # check if called from ipynb or not in order to consider appropriate plot function

C:\ProgramData\Anaconda3\lib\site-packages\plotly\basedatatypes.py in __setitem__(self, prop, value)
   3727         if match is None:
   3728             # Set as ordinary property
-> 3729             super(BaseLayoutHierarchyType, self).__setitem__(prop, value)
   3730         else:
   3731             # Set as subplotid property

C:\ProgramData\Anaconda3\lib\site-packages\plotly\basedatatypes.py in __setitem__(self, prop, value)
   2808             # ### Handle simple property ###
   2809             else:
-> 2810                 self._set_prop(prop, value)
   2811 
   2812         # Handle non-scalar case

C:\ProgramData\Anaconda3\lib\site-packages\plotly\basedatatypes.py in _set_prop(self, prop, val)
   3044                 return
   3045             else:
-> 3046                 raise err
   3047 
   3048         # val is None

C:\ProgramData\Anaconda3\lib\site-packages\plotly\basedatatypes.py in _set_prop(self, prop, val)
   3039         validator = self._validators.get(prop)
   3040         try:
-> 3041             val = validator.validate_coerce(val)
   3042         except ValueError as err:
   3043             if self._skip_invalid:

C:\ProgramData\Anaconda3\lib\site-packages\_plotly_utils\basevalidators.py in validate_coerce(self, v)
    717             if self.has_min_max:
    718                 if not (self.min_val <= v <= self.max_val):
--> 719                     self.raise_invalid_val(v)
    720         return v
    721 

C:\ProgramData\Anaconda3\lib\site-packages\_plotly_utils\basevalidators.py in raise_invalid_val(self, v, inds)
    252             typ=type_str(v),
    253             v=repr(v),
--> 254             valid_clr_desc=self.description()))
    255 
    256     def raise_invalid_elements(self, invalid_els):

ValueError: 
    Invalid value of type 'numpy.float64' received for the 'height' property of layout
        Received value: 4.5677898305882701e-301

    The 'height' property is a number and may be specified as:
      - An int or float in the interval [10, inf]
```

jhmenke, 2018-12-10:
the generic coordinates are created with an algorithm with a more-or-less random layout each time. maybe the buses are there but have overlapping coordinates.
i can see the two switches exist in both plots.

as a workaround you can create generic coordinates once and then save them to the grid if the layout looks acceptable.

miek770, 2018-12-10:
Ah yes I appear to have uploaded the wrong image, sorry about that. There are images where 2 of the 4 switches don't show up, but I'm on a different computer today and they show up each time, so I can't reproduce.

But your explanation makes sense, these buses are fed from an island (not ext grid, but static gens) and their position is somewhat random.

Considering the workaround you suggested and the fact that I can't even reproduce this behavior on this PC, I'll close the issue.

closed 2018-12-10 by miek770