e2nIEE/pandapower #234: Short circuit failure with in_service = False

miek770, 2018-12-08:
Hi,

I'm doing a maximum, 3ph sym. short-circuit on a commercial distribution network (see #233 for the plotly and plot of the network) and I'm getting values that don't make sense once I start changing transfer switches position. The normal short circuit (i.e. power coming from the utility) are fine, but in emergency mode (i.e. power from generators) short circuit values make no sense for some buses. I'm doing some testing to find the root cause, but I encountered this strange behavior during the latest test:

Firstly, I'm normally using this to change from the normal mode to the emergency mode:

`net.switch["closed"] = net.switch["closed"].apply(func=lambda x: not x)`

Since the results don't make sense, I tried this instead (closed all switches, put associated lines out of service):

```
net.switch["closed"] = net.switch["closed"].apply(func=lambda x: True)
for l in 51, 52:
    net.line.at[l, "in_service"] = False
```

But then, trying to run the short-circuit, I get the following error:

```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-26-00991481076a> in <module>
----> 1 sc.calc_sc(net=net)
      2 results = pd.merge(results, net.res_bus_sc, left_index=True, right_index=True)
      3 results.rename(columns={"ikss_ka": "ikss_ka (lines 51, 51 open)"}, inplace=True)
      4 results

C:\ProgramData\Anaconda3\lib\site-packages\pandapower-1.6.0-py3.6.egg\pandapower\shortcircuit\calc_sc.py in calc_sc(net, fault, case, lv_tol_percent, topology, ip, ith, tk_s, kappa_method, r_fault_ohm, x_fault_ohm, branch_results)
    120                     consider_sgens=False, branch_results=branch_results)
    121     if fault == "3ph":
--> 122         _calc_sc(net)
    123     if fault == "2ph":
    124         _calc_sc(net)

C:\ProgramData\Anaconda3\lib\site-packages\pandapower-1.6.0-py3.6.egg\pandapower\shortcircuit\calc_sc.py in _calc_sc(net)
    145     _add_kappa_to_ppc(net, ppci)
    146 #    t4 = time.perf_counter()
--> 147     _calc_ikss(net, ppci)
    148     if net["_options"]["ip"]:
    149         _calc_ip(net, ppci)

C:\ProgramData\Anaconda3\lib\site-packages\pandapower-1.6.0-py3.6.egg\pandapower\shortcircuit\currents.py in _calc_ikss(net, ppc)
     27         ppc["bus"][:, IKSS1] = c / z_equiv / \
     28             ppc["bus"][:, BASE_KV] / 2 * ppc["baseMVA"]
---> 29     _current_source_current(net, ppc)
     30 
     31 

C:\ProgramData\Anaconda3\lib\site-packages\pandapower-1.6.0-py3.6.egg\pandapower\shortcircuit\currents.py in _current_source_current(net, ppc)
     66     i_sgen_pu = sgen.sn_kva.values / net.sn_kva * sgen.k.values
     67     buses, ikcv_pu, _ = _sum_by_group(sgen_buses_ppc, i_sgen_pu, i_sgen_pu)
---> 68     ppc["bus"][buses, IKCV] = ikcv_pu
     69     ppc["bus"][:, IKSS2] = abs(
     70         1 / np.diag(Zbus) * np.dot(Zbus, ppc["bus"][:, IKCV] * -1j) / baseI)

IndexError: index 66 is out of bounds for axis 0 with size 66
```

I'm running pandapower 1.6.0 (from github develop installation) on anaconda, python 3.6.7, win10 64b).

As for #233, this is unfortunately a confidential network so I could share it but with a limited number of people. Please let me know if you need it and how to share it.

Thank you,

Michel

lthurner, 2018-12-10:
This seems to have something to do with the current source calculation of the sgens. Have you tried setting the sgens out of service and checking if the results make sense then?

> As for #233, this is unfortunately a confidential network so I could share it but with a limited number of people. Please let me know if you need it and how to share it.

Can you first try to reproduce the problem on a smaller (and non-confidential network)?

lthurner, 2018-12-10:
P.S.: you can also achieve this behaviour:

> net.switch["closed"] = net.switch["closed"].apply(func=lambda x: not x)

with:

> net.switch["closed"] = ~net.switch["closed"]

miek770, 2018-12-11:
Thanks for the suggestion and feedback. I'm trying to break down the issue, as you might have noticed, and will likely eventually get back to this one but I'm finding other things along the way. I'll leave this one on hold until the other, simpler ones are resolved.

miek770, 2018-12-19:
Should've already closed this, it was cause by my misunderstanding of sgen behavior during short-circuit (see #245 and #240).

closed 2018-12-19 by miek770