e2nIEE/pandapower #236: unexpected behaviour if init=results

WinfriedL, 2018-12-10:
Since the results for auxiliary buses (e.g. t3w transformer buses, xward buses) are not stored in the pandapower network, their results cannot be used for powerflow initialization.

Instead they are initialized from the results of the buses they are connect to. While this probably works in most cases, it is not the expected behaviour.
If one does two powerflows in series with init='results' one would expect the same solution without any iteration the second time. But instead the NR has to iterate and in one extreme case it even converged to a different (instable) solution when running the second time.

While I see the reason for keeping the auxiliary buses out of the pandapower network, it might be a better approach to store those and their results (maybe in a new dataframe 'aux_bus'), so they could be used again.

lthurner, 2018-12-11:
> If one does two powerflows in series with init='results' one would expect the same solution without any iteration the second time. 

I fully agree and I am currently working on exactly this issue.

First of all, I am trying to avoid [creating the auxiliary buses as pandapower buses](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/powerflow.py#L127) which takes a considerable amount of time just for pandas indexing. I am looking into adding them directly in _build_bus without having them in the bus table. This already works and gives correct results in all test cases, apart from one weird numba problem that crashes the kernel in some specific situations (working on it).

The second issue is initialization. I agree the current solution is not ideal. The reason why we have this version is that we wanted to allow using the init="res" option not only to initialize from results, but also to initialize with an arbitrary voltage vector. So if I would want to initialize from any voltage vector, the behaviour would be to write it in res_bus and then initialize from results. In this case, there wouldn't be any results for the auxiliary buses, which is why we chose an implementation that guesses these voltages as good as possible.

However, after #113 there is now the possibility of handing a voltage vector directly to the power flow function. So the init="res" function can concentrate completely on initializing from actual results, as you have proposed. While I am refactoring adding the aux buses, I want to implement this properly.

> While I see the reason for keeping the auxiliary buses out of the pandapower network, it might be a better approach to store those and their results (maybe in a new dataframe 'aux_bus'), so they could be used again.

I have thought about to versions to achieve this. **Option 1** is similar to what you describe. My idea was:

- create new result entries res_xward.vm_internal_pu and res_trafo3w.vm_internal_pu (for degree also). Use those values to initialize.
- create new result entries res_line.vm_from_pu / res_line.vm_to_pu (accordingly also for trafo & trafo3w). These values will normally be equal to the voltages of the from/to bus, but in case of an open switch they will have the voltage value of the auxiliary bus. Use those values to initialize the auxiliary buses of open switches.

**Option 2** would be to not take the values from pandapower, but to reuse the voltage vector directly from the ppc from the last iteration. The problem with this is, that it has to be ensured that the buses are the same. So my suggestion would be to save the ppc["bus"][:, [VM, VA]] values of the last iteration, and also save the pd2ppc_lookup. Then, build the pd2ppc_lookup for the new grid and check if it is the same is the previous pd2ppc_lookup. If it is the same, use the previous voltage vector, if not revert to flat start.

What I like about Option 1 is that the user has full control about what values are used (and could theoretically also manipulate them in the res_ tables to the desired values). However, the user is responsible to use init="res" only when the results match the grid. For example if the switching state is changed, the initialization might have to be changed explicetely. With Option 2, the power flow would check if the network has changed and only use the initialization if the network topology has not changed. 

In the end, maybe its best to have both options. For example use Option 2 as default behaviour in init="auto" and option 1 for init="res"?

lthurner, 2018-12-17:
We have discussed this in our developers meeting today and decided that Option 1 would be better. 

Option 2 has the advantage that changes in the topology are automatically detected and the results are only used if the topology doesn't change. However, if a topology change is made that the user knows to be minor (for example closing a switch that doesn't have any impact on the power flow), the results still couldn't be reused, because the dimensions of the ppc voltage vector isn't the same anymore and the mapping of the buses is unclear. With Option 1, the results can always be reused, and the user has full control over all values.

lthurner, 2018-12-19:
This now works with internal buses of xwards and trafo3w as well as open switches at lines, trafos and trafo3w. Running a second power flow with init="results" allows convergence with 0 iterations in all those cases:

https://github.com/e2nIEE/pandapower/blob/89dcd128cad4a50504ea02bf6da60a0658808e2b/pandapower/test/loadflow/test_runpp.py#L957-L986

New result parameters are:

- res_line:
      - vm_from_pu
      - va_from_degree
      - vm_to_pu
      - va_to_degree
- res_trafo:
      - vm_hv_pu
      - va_hv_degree
      - vm_lv_pu
      - va_lv_degree
- res_trafo3w:
      - vm_hv_pu
      - va_hv_degree
      - vm_mv_pu
      - va_mv_degree
      - vm_lv_pu
      - va_lv_degree
      - vm_internal_pu
      - va_internal_degree
- res_xward:
      - vm_internal_pu
      - va_internal_degree

closed 2018-12-19 by lthurner