e2nIEE/pandapower #237: Incorrect 3ph Scc at xfo secondary with wrong vbus

miek770, 2018-12-10:
Hi,

I modeled a very simple network to investigate weird results I was getting for 3ph short circuit simulations at 600V and lower. Along the way I noticed that the short circuit current in kA is reported incorrectly when a bus voltage differs from its feeder transformer's voltage.

This is the code I used:

```
import pandapower as pp
import pandapower.shortcircuit as sc
import pandas as pd

net = pp.create_empty_network(name="IUSMQ", f_hz=60.0)

bus0 = pp.create_bus(net, vn_kv=0.6, name="bus0", type="n")
#bus1 = pp.create_bus(net, vn_kv=0.6, name="bus1", type="n")
bus1 = pp.create_bus(net, vn_kv=0.208, name="bus1", type="n")
print(f"Buses:\n\n{net.bus}\n")
    
pp.create_ext_grid(net=net, bus=bus0, s_sc_max_mva=999, rx_max=1/10, rx_min=1/10)
print(f"Grid:\n\n{net.ext_grid}\n")

s_kva = 15
z_percent = 2.3
xr = 1.13
vsrc_percent = (z_percent**2 / (1 + xr**2))**0.5
pp.create_transformer_from_parameters(net=net,
                                      hv_bus=bus0,
                                      lv_bus=bus1,
                                      sn_kva=s_kva,
                                      vn_hv_kv=0.6,
                                      vn_lv_kv=0.208,
                                      vscr_percent=vsrc_percent,
                                      vsc_percent=z_percent,
                                      pfe_kw=1e-20,
                                      i0_percent=1e-20)
print(f"Transformer:\n\n{net.trafo}\n")

sc.calc_sc(net=net)
results = pd.merge(net.bus, net.res_bus_sc, left_index=True, right_index=True)
results
```

Using simple calculations (MVA method, see [https://ieeexplore.ieee.org/document/4157585?arnumber=4157585](https://ieeexplore.ieee.org/document/4157585?arnumber=4157585)), I would expect the symmetrical short circuit current to be around 1.8 kA at 208V:

`Icc = 15 kVA / (2.3% * 3**½ * 208V) = 1.81 kA`

Using Pandapower, I get:

- 1.92 kA when bus1's voltage is set to 208V;
- 5.55 kA when bus1's voltage is set to 600V (see commented line, the xfo remains 0.6-0.208 kV).

The expected behavior would be either:

a. Raise an error because the bus' voltage greatly differs from its supply voltage (from its transformer);
b. Report 1.92 kA, the bus' nominal voltage is irrelevant from a physical perspective.

For reference, Etap 18.1.1, I get to 1.809 kA using ANSI standards (see link below), and 1.92 kA using IEC standards.

[https://drive.google.com/open?id=1RJ3Dpd9-XZQO6ar-_sVGjwlRiviOrCv_](https://drive.google.com/open?id=1RJ3Dpd9-XZQO6ar-_sVGjwlRiviOrCv_)

This also raises concerns for situations where the transformer's voltage would only slightly differ from the bus voltage. If a transformer's voltage is slightly different from the bus' nominal voltage, will this be considered adequately?

Tested on win7, python 3.7.0 64 bit (anaconda3). Please let me know if you need more information to reproduce.

Michel

lthurner, 2018-12-19:
Thanks for bringing this up!

> Along the way I noticed that the short circuit current in kA is reported incorrectly when a bus voltage differs from its feeder transformer's voltage.

From the example you have given, I think that pandapower actually calculates the correct result. The short-circuit calculation according to IEC assumes that a short-circuit occurs at rated voltage +- security margins (I am not familiar with ANSI but I assume it is similar). So if you have a 600V bus, the assumption for maximum short-circuit current is that the short-circuit occurs at 600V*1.1 = 660V. If you have a 208V bus, the assumption would be that the short-circuit occurs at 208V*1.1=228.8V. Obviously the short-circuit current will be much lower in this case. 

You say the current is incorrect, but I would actually argue it is the correct result for the input that your provided. The result is unrealistic, because the input parameters are unrealistic, not because the method is wrong.

> This also raises concerns for situations where the transformer's voltage would only slightly differ from the bus voltage. If a transformer's voltage is slightly different from the bus' nominal voltage, will this be considered adequately?

Yes I think pandapower does correctly consider this, from your description it sounds more like ETAP would not consider this correctly if the rated voltage of the bus is ignored?

This:

> a. Raise an error because the bus' voltage greatly differs from its supply voltage (from its transformer)

already exists, but not directly in the pd2ppc function and rather in the diagnostic function that is called with:

> pp.diagnostic(net)

If I run this with the example network you provided, the parameterization is correctly "diagnosed":

> Trafo 0: Nominal voltage on lv_side (0.208 kV) and voltage_level of lv_bus (bus 1 with voltage_level 0.6 kV) deviate more than +/- 30.0 percent.

miek770, 2018-12-19:
> From the example you have given, I think that pandapower actually calculates the correct result. The short-circuit calculation according to IEC assumes that a short-circuit occurs at rated voltage [...]

Margins aside (I am not saying 1.92 kA is wrong; ETAP and pandapower reach the same result), and notwithstanding the *incorrectness* of the input parameters, if you were to test this in real life, bus1's nominal voltage would have absolutely no impact. The short-circuit current would be close to 1.92 kA, no matter the voltage rating of bus1.

I understanding this is an extreme example, but when I design substations tied to the transmission network, the transformer's nominal voltages often differ from the buses nominal voltages. For example in one substation the transformer is 110-12.48 kV and the HV bus is operated at 115 kV (utility). The medium voltage bus' nominal voltage is 12.5 kV, but its operating voltage is going to be closer to 13.0 kV because of this mismatch (ignoring taps). An if there's a short circuit at the MV bus, its base voltage should be 13.0 kV, not 12.5 kV or 12.48 kV.

To summarize, I believe that apart from voltage regulated buses and swing buses, a bus' nominal voltage should have absolutely no impact in most calculations, as in real life.

> ```pp.diagnostic(net)```
> 
> If I run this with the example network you provided, the parameterization is correctly "diagnosed":

Ah thanks, I didn't know about this function. I'll add it to my scripts.

lthurner, 2018-12-19:
> Margins aside (I am not saying 1.92 kA is wrong; ETAP and pandapower reach the same result), and notwithstanding the incorrectness of the input parameters, if you were to test this in real life, [...]

I think there is a difference between how we interpret the value. There is a nameplate parameter of the actual physical equipment, which says "this busbar has a rated voltage of 0.6kV, do not operate this busbar above 0.8kV". Then there is a the rated voltage of the network area, which says "this network is a 600V network which is operated in a range of 540V-660V". One is the rated voltage of the physical equipment, one is the nominal operating voltage of teh grid. The vn_kv parameter gives the nominal operating voltage of the grid, not the nominal voltage of the equipment.

> I understanding this is an extreme example (the one from my first post), but when I design substations tied to the transmission network, the transformer's nominal voltages often differ from the buses nominal voltages. For example in one substation the transformer is 110-12.48 kV and the HV bus is operated at 115 kV (utility). The medium voltage bus' nominal voltage is 12.5 kV, but its operating voltage is going to be closer to 13.0 kV because of this mismatch (ignoring taps). An if there's a short circuit at the MV bus, its base voltage should be 13.0 kV, not 12.5 kV or 12.48 kV.

I know that they can differ, that is actually taken into account when the ratio of the transformer is calculated. But the question is: at what voltage do we expect a short-circuit to occurr? If the rated voltage of the bus is 0.6kV, we would expect a short-circuit to ocurr at 0.6kV, not at 0.2kV (because we expect the grid to operated at 0.6kV). For example in this case:

bus 1                                          bus2                        bus3
x  ----------trafo------------------ x --------line---------- x 
0.6kV          0.6kV/0.2kV               0.6kV                      0.6kV

You are saying the rated voltage at bus2 should be ignored, and it should be expected that a voltage at bus2 ocurrs at 200V. But what should happen at bus3? There is no rated transformer voltage there, so you could only take the 600V rated bus voltage. Then you would have a short-circuit current that is 3 times higher at bus3, even though the two buses are only seperated by a line? That wouldn't reflect the reality either.

miek770, 2018-12-19:
Ok, I get your point. Basically, pandapower trusts that the user inputs an adequate vn_kv, and provides a diagnostic function to detect invalid vn_kv. Thanks for the thorough explanation, I appreciate it.

Would you be open to changing the description for vn_kv from "The grid voltage level" to "The grid nominal operating voltage level"?

P.S.: I checked on EasyPower and it behaves exactly like pandapower. ETAP does the diagnostic beforehand and refuses to simulate a short circuit because of the voltage difference.