e2nIEE/pandapower #238: Unit convention - Fault kA *sym*

miek770, 2018-12-10:
Hi again,

Please note that I am not very familiar with the IEC 60909 standard (I usually use ANSI/IEEE), but I would like to suggest adding either `sym` or `asym` to the `ikss_ka` variable, in `net.res_bus_sc`. IMHO, it should never be implicit. I believe it would also be more in line with pandapower's [Unit System and Conventions](https://pandapower.readthedocs.io/en/v1.6.0/about/units.html).

From what I've gathered from the doc, it looks like ikss_ka is a symmetrical value, so I'd simply suggest renaming it to:

`ikss_sym_ka`

Thanks,

Michel

julffers, 2020-11-06:
@rbolgaryn any news regarding this issue?

vogt31337, 2025-07-23:
Will be closed, since pandapower will be updated soon with asym powerflow.