e2nIEE/pandapower #243: State estimation and zero injection constaints

ALRSIE, 2018-12-12:


jhmenke, 2018-12-13:
great to hear you like pandapower!

regarding your problem: zero injections are not modelled differently in pandapower right now, therefore the current workaround is to use the "virtual measurement" with 0 as the measurement values - just like you did. i found a std. dev. of 1. kW/kVar to be sufficiently small for my problems, as it does not change the results significantly, but then again most of my measurements are pseudo measurements anyway.

i'd try different values (1, 0.1, 1e-2, 1e-3) to see where there is a good trade-off between an ill-conditioned matrix and varying estimation results.

ALRSIE, 2018-12-13:
Thanks for the reply!

Well, I have real measurements and of really good quality... :-)

I have gone through all these tests and trials before and I have unfortunately confirmed that I need zero injection constraints functionality for this particular problem I’m working with. Therefore I did my own estimator in Matlab. My problem is that it is just too much work to expand my functionality and maintain the code. Therefore I started to look for an alternative and found pandapower.

So I can see two alternatives:
1.	I keep my code (not a preferred option) and use pandapower for other things.
2.	I can try to implement a zero injection constraint in pandapower.

Would the second alternative be of any value for you guys?

Best,
Anders

jhmenke, 2018-12-13:
> 2\. I can try to implement a zero injection constraint in pandapower.

Sure, i think it would help many people if you could implement it!
If you want to do it, i would refer you to the dev_2.0 branch, where some breaking changes regarding the estimation module have been made for the pandapower 2.0 release.

ALRSIE, 2018-12-13:
OK…

Now  let me tell you! I was probably finished at university before you were born. The first computer I used at university was available my second year (only word processing). We were not allowed to use a calculator before the third year. I am a physicist by education. So when you say “2.0 branch” or even understanding github is already complicated for me. But I know some about matrixes, grid measurements and grid models.

Before I can say anything about actually being able to help and create more value than problems, I would like to look at the specific code in more detail to try to understand how your stuff is build up. Could you perhaps direct my to the function(s) where the gain matrix is constructed?

Sorry for being a crappy programmer… but I'm not so old that I stopped learning (yet).
Anders

jhmenke, 2018-12-13:
no worries, i just assumed you were familiar with Github!

The relevant code block can be seen here: https://github.com/e2nIEE/pandapower/blob/9fc64ac8e5934e930e639a2e0aaeb6f37af162b3/pandapower/estimation/state_estimation.py#L297
the whole state estimation iteration loop lasts from line 283 to line 311.

do you have information about the required changes? i can make an estimate how much work it would be and where the code would have to be changed.

ALRSIE, 2018-12-13:
It all looks familiar! 

I'm trying to find the paper with the details. It should be in this paper:
Aschmoneit, Peterson, Adrian, “State Estimation with Equality Constraints”, 10th IEEE PICA conference, May 1977.  (1977 is more my "prime time"...)

I think it will be fairly simple. At least a test. It is all about "expanding the gain matrix" with the constraints.  I will come back after having examined the code some more.

You got mail

ALRSIE, 2018-12-13:
Getting there slowly.

Perhaps a new "create_jacobian" can be made for the "zero injection constaints" or put that in that same definition. One has to find the buses with zero injection constaints. Perhaps type="n" when "create_bus" can be a solution (in "my world" a node has no power injection...perhaps you use this differently?)

jhmenke, 2018-12-14:
good morning and thanks for the mail! i looked at your comments and the code you sent me and it does not seem to be so complicated to integrate this into the pandapower estimation.

there are two steps to integrating this: 

1. writing a new function to create the matrix C. this is done in *wls_matrix_ops.py*. i have created a sample for the first version cx_11:

```
    def create_zero_injections(self, v, delta, zero_inj):
        n = len(self.ppc["bus"])
        G = self.G
        B = self.B

        cx_11 = np.zeros(len(zero_inj), n - 1)
        for i in range(len(zero_inj)):
            for j in range(n - 1):
                k = zero_inj[i]
                if j + 1 == k:
                    cx_11[i, j] = -v[k]**2 * B[k, k]
                    for l in range(delta):
                        cx_11[i, j] = cx_11[i, j] + v[k] * v[l] * (-G[k, l] * np.sin(delta[k] - delta[l]) 
                                                                   + B[k, l] * np.cos(delta[k] - delta[l]))
                else:
                    cx_11[i, j] = v[k] * v[j + 1] * (G[k, j + 1] * np.sin(delta[k] - delta[j + 1]) 
                                                     + B[k, j + 1] * np.cos[delta[k] - delta[j + 1]])
```

2. in *state_estimation.py*, the iteration loop has to be extended to include the new matrices, which is very similar to the Matlab code. 

(3. since using for-loops in Python is very inefficient and results in a high computation time, the code can then be optimized)

From your Matlab code is assumed that Theta are the voltage angles of all buses? What exactly is Ninj (i have called it zero_inj in the sample code).
I am happy to help out extending this code and make sure it works.

ALRSIE, 2018-12-14:
Ninj are just list of the nodes with zero injection. I managed to find the formualtion that I used. If you have the book "Power System State estimation" by Ali Abur you can find different formulations. I used "Equality-Constrained WLS State Estimation" chaper 3.5

I looked att your creation of jacibian and it is som much nicer looking than my code. The jacobian for the contraints should be possible to write in the same nice manner but just looping over nodes with zero injection. After the operation the gain matrix is expanded as shown.

So how do we indicate that a node is has zero injection so that we can find the nodes? Perhaps it is possible to use "type" when creating a bus? Any idea?

lthurner, 2018-12-14:
> Regarding the word node: in the estimation code, node is used as a general term for bus, regardless of any power injections or not.

How about adding the zero injection constraint for all buses that do not have a load/sgen/shunt element?

jhmenke, 2018-12-14:
> > Regarding the word node: in the estimation code, node is used as a general term for bus, regardless of any power injections or not.
> 
> How about adding the zero injection constraint for all buses that do not have a load/sgen/shunt element?

good idea. also gen.

is ward/xward modeled as a branch or as a bus power injection (zip load)? maybe it should be included as well.

ALRSIE, 2018-12-14:
I guess you can always use zero injection when you know it is. It will be a larger matrix and will perhaps slow down calculations.

I want to discuss the names a little further. When creating a bus you can assign a type "b", "n" or "m". For me who works with the transmission grid a "b" is a busbar that can distribute power to several objects through switches and other stuff. A "node" is a simple fixed connection point that splits power to a very limited number of items. What was the original reason to choose different types when creating a bus?

Personally I for example refer to nodes when you go from overhead line to cable or when you split a overhead line to go in different directions. (But I'm not really power engineer). On the other hand. This is not so important if we are only very clear on what we are talking about.

jhmenke, 2018-12-14:
> I guess you can always use zero injection when you know it is. It will be a larger matrix and will perhaps slow down calculations.

Is there a reason to not use a zero injection "measurement" if you know there is nothing connected at that bus?

ALRSIE, 2018-12-14:
I just had a short discussion with my friend at the Swedish TSO. He was not really sure. It is probably better. The condition of the matrix will be better and the results more accurate. I guess an implementation and some performance tests will clear that question. I guess it is not so hard to implement a "switch" that decides whatever method to use?

ALRSIE, 2018-12-15:
>> I want to discuss the names a little further. When creating a bus you can assign a type "b", "n" or "m".

> This is really more of a general pandapower discussion and may be best suited in a separate issue if important.

Not important for me. Just curious. Let’s drop that for now.

>No, that should not be an issue. I can try to make the rough function and structure of the code next week and then the individual matrices could be coded by you maybe?

Yes, that sounds like a plan. I'm looking at it right now to warm up.

jhmenke, 2018-12-18:
So i have created the base code to include zero bus injections. See the new required function here:

https://github.com/jhmenke/pandapower/blob/89ca5d5eb0c7fb067a30ce6c75288847312ff635/pandapower/estimation/wls_matrix_ops.py#L307

And then the iterative process to find the estimate here:

https://github.com/jhmenke/pandapower/blob/89ca5d5eb0c7fb067a30ce6c75288847312ff635/pandapower/estimation/state_estimation.py#L304

If you have never used Git you can download the code as a zip file: https://github.com/jhmenke/pandapower/archive/est_zero_injections.zip
But it would be preferred to use the versioning system git. Install it and then you can check out the code with 

> git clone https://github.com/jhmenke/pandapower.git
> git checkout est_zero_injections

ALRSIE, 2018-12-18:
Great!

I will have a long Christmas holiday and I'm not sure how much I can work on the problem. I will do my best. I will figure out and use the Git version system.

Merry Christmas

ALRSIE, 2018-12-26:
OK, now I think I have a working code. At least a code that seems to be working on my problems. I had to change your concept a little. I look for nodes without P and Q injections and add constraints for those buses only. Also I added code inside the “create_jacobian” function because everything is already calculated there and adding another function will just slow down the code (I think).

Personally I only use measurements and never use generators or loads etc. I have no good understanding if my changes will be affected by components like generators, loads and others. This has to be tested properly.

Since I still struggle with learning Python I think there is “room for improvement” in my code. I have not yet got very friendly with Git and will send my work by mail for you to look at (sorry). During the rest of my holiday I will make some more test to verify that everything is working.

Happy new year!

ALRSIE, 2018-12-30:
Well, you are probably right.

I tried to implement a static generator and a load. The code for state estimation just do not seem to care about these objects at all. How are the objects modeled? For example if a load is modeled as a “constant Z” it will be part of grid as a shunt somewhere(?) If it is treated as a “measurement” it needs to have some kind of uncertainty to it in order for the WLS estimator to work(?) (or implemented as a constant constraint). What is it I do not understand? Have I misunderstood how to implement a load? “ld1=pp.create_load(net, bus=b4, p_mw=10., q_mvar=0., const_z_percent=100)”.

Either way I think that one must also exclude busses with injection measurements from your list of “zero_injection_buses” described above…if that approach is used.

Talk to you later

jhmenke, 2019-01-02:
pandapower elements are completely separate of the measurements. a pandapower load value will be used for the power flow, but the measurement model will only use the elements to create all required matrices for the state estimation. the only relevant inputs are in the measurement table of the pandapower grid. but one can deduce that when there is no element attached to a bus the bus injection measurement would always be zero.

a constant Z load will affect the Y matrix i think (not sure how it's modelled in pandapower without looking it up) and therefore should be included in the estimation process.

>Either way I think that one must also exclude busses with injection measurements from your list of “zero_injection_buses” described above…if that approach is used.

Agreed of course.

jhmenke, 2019-01-07:
it's actually not that easy to code cleanly. i will need a few more days

jhmenke, 2019-01-11:
Okay i finally had some time and implemented it with automatic zero injections. Your example still yields the same results. The version can be found on https://github.com/jhmenke/pandapower/tree/est_zero_injections

```
git clone https://github.com/jhmenke/pandapower.git
git checkout est_zero_injections
```

Could you do some testing if the behaviour is as expected? Thanks!

jhmenke, 2019-02-20:
i guess the issue has been resolved

PR #269 (2019-01-17): Add support true zero injection and capsuled the matrix operation in state estimation
Fixed #243, 
Fixed #268
closed 2019-02-20 by jhmenke