e2nIEE/pandapower #244: Separating motors from generators

miek770, 2018-12-13:
This is an open discussion that was started in #208 about splitting motors from generators, which would be of interest for distribution and industrial systems. The current issue was opened to keep both subjects separated, as @lthurner suggested.

A brief recap:

#208 is about changing the signing convention for generators, which currently uses negative values for generation, and positive values for consumption. This might be a source for confusion, which is currently being mitigated by being well documented.

The main argument in favor of splitting motors from generators, i.e. in different DataFrames, is (from [this post](https://github.com/e2nIEE/pandapower/issues/208#issuecomment-446986548)):

> ...for distribution and industrial systems, there are a lot of motors and they are treated differently in short-circuit analysis and different types of power flow studies. [...]
>
> For example you have load factors and diversity factors that apply to motors and not generators. You may have different rules based on different scenarios. You may want to ignore all motor contributions after a few cycles for short circuit. You may want to simulate motor starting. The list goes on.

[This post](https://github.com/e2nIEE/pandapower/issues/208#issuecomment-446995060) adds more insight for motors in distribution vs transmission systems, and concludes with:

> Ultimately, we want pandapower to be able to be used in transmission systems as well as distribution systems. We always have to balance between adding functionality that is needed and not adding to much features so that clarity is lost and the tool gets to crowded and complicated. To that end, these discussions are very helpful to understand the different use cases.

I thus propose that we start by documenting and discussing the potential pros and cons of splitting motors from generators. I'll update this post based on the following discussion:

***PROS***
- Allow for positive signing for motor loads, even if #208 is integrated.
- Facilitates motor behavior during certain analysis (to be developed).

***CONS***
- Added complexity of creating a new element, documenting it, maintaining it, updating tests, etc.

lthurner, 2018-12-13:
Thanks for summarizing this and in general thanks for getting involved in pandapower development, much appreciated!

My main question would be: what parameters are necessary to model a motor, on how would it be considered in the different analysis functions? For short-circuit calculation, the internal impedance is currently calculated from the rated voltage (vn_kv), the rated power (sn_mva) and the rx ration (rx), so those two would be needed. For power flow calculation, I guess you would specify an active power (p_mw) and a power factor (cos_phi)?

miek770, 2018-12-14:
My pleasure!

About parameters, I would recommend keeping in mind that for industrial systems, information is often scarce. When I work with a HV power transformer or a generator, I expect to be able to get tons of data including a well detailed nameplate and test reports. Of course there are exceptions but the general rule is to have reliable data about the equipment.

But in LV distribution systems, you are often limited to something like: 7.5kW, 575V (3ph) motor (sadly usually in horsepower in Canada...). There are many reasons for this, from the fact that getting more precise values costs money (field survey) for not much overall benefit, to inaccessible or unreadable nameplates and lost drawings. Fortunately IEEE standards and guides offer many reasonable assumptions to estimate other motor parameters based on the known values. There are typical PF and short-circuit ratios that can be used based on the motor type.

In addition, I'd *try* to limit the inputs to nameplate values instead of test and calculated values, at least for the required inputs. But my suggestions might be biased towards IEEE, so they should be validated against IEC which is the predominant standard for pandapower (if I'm not mistaken, at least for short circuit).

Some optional inputs default values, as mentioned above, should depend on the required inputs. It might be awkward to do this in the standard create_motor() function, so I'd suggest adding standard motor types instead. We could create a standard type for each range of motor power that typically share common optional attributes.

Considering this, and leaving the dynamic stuff aside for now, I'd propose the following inputs for the motors created from parameters. This could be expanded A LOT in the future, but I think it'd be a good start:

***Required inputs***

- **vn_kv** (note that it's not necessarily the same as the bus voltage; we use 575V motors on 600V systems)
- **p_kw** (yes there's a discussion in #73 about units but for motors I'd recommend a lower unit, i.e. kW)

***Optional inputs***

- **lra_a** (locked rotor amps; required for short circuit and ultimately motor starting, but could default to ~6 times the calculated FLA which is a conservative value. I'd calculate the internal impedance based on this nameplate value if it was provided instead of requesting it as an input)
- **cos_phi** (i.e.: power factor; required for power flow)
- **eff_pu** (efficiency; sn_kva would be derived from it and the cos_phi; required for power flow)
- **loading_pu** (for example, a 10kW motor could be used on a 8kW mechanical load, resulting in a 0.8 loading factor; used in power flow but with sane default (ex.: 0.9))
- **demand_pu** (for example, the motor could be individually specified as running 50% of the time, which is usually combined with the loading factor for load flow studies. could also be called diversity factor; used in power flow, defaults to 1.0)

And indeed I don't see much benefit for special OPF considerations for motors.

lthurner, 2018-12-14:
> p_kw (yes there's a discussion in #73 about units but for motors I'd recommend a lower unit, i.e. kW)

We can discuss the unit in the other issue. What exactly is this parameter defining, the mechanical rated power? Shouldn't it then be pn_kw?

> lra_a (locked rotor amps; required for short circuit and ultimately motor starting, but could default to ~6 times the calculated FLA which is a conservative value. I'd calculate the internal impedance based on this nameplate value if it was provided instead of requesting it as an input). Also I think you would still need an rx-ratio to calculate the impedance?

Since this value is usually given relative to the rated current, I would suggest Ira_pu instead? Then Ira_pu = 6 would be the value that you mean.  

> eff_pu (efficiency; sn_kva would be derived from it and the cos_phi; required for power flow)

I would find a percentage value more intuitive for efficiency. You didn't mention sn_kva in your list of parameters, I am still not clear with regard to the different parameters p_kw (pn_kw?) and sn_kva. 

>loading_pu (for example, a 10kW motor could be used on a 8kW mechanical load, resulting in a 0.8 loading factor; used in power flow but with sane default (ex.: 0.9))

I would also either use a percentage value here, or maybe just a power value (p_kw) if the other power value is supposed to be the rated power (pn_kw)?

>demand_pu (for example, the motor could be individually specified as running 50% of the time, which is usually combined with the loading factor for load flow studies. could also be called diversity factor; used in power flow, defaults to 1.0)

For load/sgen/gen we have a scaling parameter, that seems similar to this, I would call it scaling here also for uniformity.

Based on this my "counterproposal" for the parameters would be:

- vn_kv
- pn_kw
- p_kw
- cos_phi
- ira_pu
- rx
- efficiency_percent
- scaling

But I am not yet completely clear on the whole pn_kw/p_kw/sn_kva situation. Maybe you can clarifiy from your experience?

> Some optional inputs default values, as mentioned above, should depend on the required inputs. It might be awkward to do this in the standard create_motor() function, so I'd suggest adding standard [...]

Yes I agree. If we want to define some "typical" parameters those should not be default values in the create function, but stored in standard types.

miek770, 2018-12-14:
> > p_kw (yes there's a discussion in #73 about units but for motors I'd recommend a lower unit, i.e. kW)
> 
> We can discuss the unit in the other issue. What exactly is this parameter defining, the mechanical rated power? Shouldn't it then be pn_kw?

Ah yes I should've written pn_kw.

> Since this value is usually given relative to the rated current, I would suggest Ira_pu instead? Then Ira_pu = 6 would be the value that you mean.

It's actually in amperes on the nameplates, but the nominal amperes is also shown so calculating a ratio is rather easy in most cases. Plus it's going to be much easier and elegant when defining standard types. Sounds good.

> > eff_pu (efficiency; sn_kva would be derived from it and the cos_phi; required for power flow)
> 
> I would find a percentage value more intuitive for efficiency. You didn't mention sn_kva in your list of parameters, I am still not clear with regard to the different parameters p_kw (pn_kw?) and sn_kva.

Honestly, I leaned towards pu to save characters. The value is usually given in percent, so ok.

sn_kva is derived from the other parameters:

`sn_kva = pn_kw / (efficiency_percent * cos_phi)`

...but it's not directly relevant to the calculations. Short-circuit will derive an impedance from the lra_pu, and power flow / OPF will use the calculated active power:

`s_kva = sn_kva * loading_percent * scaling`

Just to be clear: short circuit will ignore both the loading_percent and scaling.

You could also check for overloading, i.e. when the operating voltage differs from the motor's nominal voltage, but the relationship is non-linear. The rx ratio changes as the voltage differs from vn_kv. Thus calculating motor overloading would not be straightforward. For the first implementation I would consider that cos_phi is linear and not check for motor overloading.

> I would also either use a percentage value here, or maybe just a power value (p_kw) if the other power value is supposed to be the rated power (pn_kw)?

I would definitely use either a percent or pu value. It's a factor, and p_kw also depends on the demand/scaling factor below. Ok for loading_percent.

> For load/sgen/gen we have a scaling parameter, that seems similar to this, I would call it scaling here also for uniformity.

Ok, we can mention that it can be used to model demand/diversity factors in the description. I prefer demand because the factor represents a time-dependant statistic, whereas scaling implies a load-dependant factor, but if it's documented it's ok.

P.S.: Shouldn't it be scaling_percent?

> Based on this my "counterproposal" for the parameters would be:
> 
> * vn_kv
> * pn_kw
> * p_kw
> * cos_phi
> * ira_pu
> * rx
> * efficiency_percent
> * scaling
> 
> I think having a pn_kw and p_kw would be clearer than having a p_kw and then two different scaling factors that are applied. But I am not yet completely clear on the whole pn_kw/p_kw/sn_kva situation. Maybe you can clarifiy from your experience?

As mentioned above, I would keep pn_kw, loading_percent and scaling (or demand_percent, or diversity_percent). Another reason I didn't mention is that loading_percent should be pretty much fixed for all situations, for a given motor, but scaling (demand/diversity) changes depending on the scenario (maximum load, minimum load, specific operating circumstances, etc.).

So my counter-counterproposal would be (required in **bold**):

* **vn_kv**
* **pn_kw**
* lra_pu (note that the 1st letter is an "L", not a capital "i")
* cos_phi
* efficiency_percent
* loading_percent
* scaling (or scaling_percent?)

lthurner, 2018-12-14:
> Honestly, I leaned towards pu to save characters. The value is usually given in percent, so ok.

If it is usually given in percent, then I think its better for clarity to use percent.

> P.S.: Shouldn't it be scaling_percent?

scaling is 0...1, not 0...100, so it is not a percentage value. It is also not a pu value, since it doesn't have anything to do with the power system per unit system. It is just a relative scaling factor without a unit.

>sn_kva is derived from the other parameters:

>sn_kva = pn_kw / (efficiency_percent * cos_phi)

>...but it's not directly relevant to the calculations. Short-circuit will derive an impedance from the lra_pu, and power flow / OPF will use the calculated active power:

>s_kva = sn_kva * loading_percent * scaling

>Just to be clear: short circuit will ignore both the loading_percent and scaling.

> You could also check for overloading, i.e. when the operating voltage differs from the motor's nominal voltage, but the relationship is non-linear. The rx ratio changes as the voltage differs from [...]

But how exactly is Q calculated for the motor? Let's say we have a motor with
- pn_kw = 100
- efficiency_percent = 95
- cos_phi = 0.9

and want to calculate P/Q for loading_percent = 80 I would then say:

- electrical rated power is 100kW/(95%/100) = 105kW
- active power for power flow input is p = 105kW*80%=84kW
- reactive power for power flow input is q=?

You say that cos_phi is linear, but wouldn't we need to model the motor in more detail to calculate reactive power? Don't the slip and the bus bar voltage also influence the reactive power?

miek770, 2018-12-14:
> But how exactly is Q calculated for the motor? Let's say we have a motor with
> 
> * pn_kw = 100
> * efficiency_percent = 95
> * cos_phi = 0.9
> * loading_percent = 80
> 
> and want to calculate P/Q:
> 
> * electrical rated power is 100kW/(95%/100) = 105kW
> * active power for power flow input is p = 105kW*80%=84kW
> * reactive power for power flow input is q=?

Adding a scaling of 0.5 (motor used 50% of the time):

```
pn_kw = 100 = nominal mechanical active power
pn_kw / efficiency_percent = 100 / 0.95 = 105 kW = nominal electrical active power
sn_kva = 105 kW / cos_phi = 105 kW / 0.9 = 117 kVA = nominal power
qn_kvar = sqrt(sn_kva**2 - pn_kw**2) = 51 kVAR = nominal reactive power
```

I would then apply the loading_percent and scaling factors for PF and OPF:

```
p_kw = 105 kW  * 80% * 0.5 = 42 kW
q_kvar = 20 kVAR
s_kva = sqrt(p_kw**2 + q_kvar**2) = 46.5 kVA
```

Here's ETAP data for the same motor: [https://drive.google.com/open?id=1LfSSuz-FKdSacrj_nj4q4lUEXAIYMosr](https://drive.google.com/open?id=1LfSSuz-FKdSacrj_nj4q4lUEXAIYMosr) (note that I combined loading and demand in the same tab, but they are really split into 2 variables as I recommended).

> You say that cos_phi is linear, but wouldn't we need to model the motor in more detail to calculate reactive power? Don't the slip and the bus bar voltage also influence the reactive power? And if we use a simplified model that doesn't rely on slip, wouldn't a constant reactive power be more realistic than scaling the reactive power with the active power?

Sorry I meant constant. But for a non-linear cos_phi:

scaling shouldn't be affect cos_phi, because it's meant to be time-dependant, not load-dependant. The bus voltage and loading_percent would indeed affect cos_phi. For the effect of loading of cos_phi, we could also simply ask users to input the cos_phi for different loadings; this is what ETAP 18.1.1 does: [https://drive.google.com/open?id=1M2wZ5KYuYFUBWEPog4MumUCbQB2HL_Eu](https://drive.google.com/open?id=1M2wZ5KYuYFUBWEPog4MumUCbQB2HL_Eu) (also prevous link).

Notice that by default ETAP suggests a constant cos_phi for different loadings, supposedly based on manufacturer data. The same principle applies to efficiency, which also varies with loading. The effect of bus voltage is neglected, and my unsterstanding is that it's very hard to estimate. There are rule-of-thumbs, but it's usually considered during motor starting / motor acceleration studies, not SC/PF/OPF, and let to the engineer to determine if the motor succeeds or fails to start, at least for *static* motor acceleration. Dynamic motor acceleration is another beast that requires a lot of mechanical information. That could be left for future enhancements.

So I think the best solution would be to:

1. Add optional cos_phi at 0%, 50% and 75% loading, and interpolate for loading_percent (I'd use a polynomial interpolation, not a linear interpolation).
2. Consider constant cos_phi from 0% to 100% loading, when no other value is provided.

Thus:

* **vn_kv**
* **pn_kw**
* lra_pu (note that the 1st letter is an "L", not a capital "i")
* cos_phi_fl (cos-phi at full load)
* efficiency_percent_fl (efficiency at full load)
* loading_percent
* scaling
* cos_phi (dictionary of cos_phi at different loadings)
* efficiency_percent (dictionary of efficiencies at different loadings)

Alternatively, we could define cos_phi_75, cos_phi_50, etc. or something similar to avoid having a dictionary, which might not look very good in a DataFrame and/or adhere to pandapower's current way of handling data.

miek770, 2018-12-18:
Hi,
I renamed the issue to make it clear that I'm talking about asynchronous motors and static generators in this issue. I've also added ***PROS*** in my first post.
Michel

lthurner, 2018-12-19:
Hi, thanks for the comprehensive PROS list, I am convinced that a motor element would be a good idea! :+1: 

We have talked internally about the cos_phi / cos_phi_n issue. As you said, giving a dictionary in the DataFrame wouldn't be a very elegant way to give this input. Here is another suggestion that emerged instead: the motor would get only a cos_phi value that is applied as a constant. The user would then be responsible for updating the cos_phi value together with the scaling/loading. There could then be helper functions in toolbox that set the cos_phi for example from a dictionary of cos_phi points with linear interpolation. For example:

```
update_motor_loading(net) #set the loading values according to the scenario
pp.set_linear_motor_cos_phi(net, [(0, 0.1), (50, 0.7), (100, 0.9)]) #set the cos_phi values to match the loading
pp.runpp(net)
```

For a motor that has a cos_phi of 0.1 at 0%, 0.7 at 50% and 0.9 at 100%.

With this method, a complicated input in the DataFram would be avoided, and there would be full flexibility for the user (linear interpolation between points would only be one possibility, you can of course also write a function that calculates the cos_phi with any other method). What do you think?

miek770, 2018-12-19:
About cos_phi, I think it's a reasonable compromise. The issue I would have with this is that part of the motor physical data (not scenario data) would live in the python code and not with the rest of the data. If you share your data with someone, you'll have to specify the cos_phi's separately to allow the other user to work with the data.

I would lean towards keeping all of the data at the same place.

lthurner, 2020-08-13:
I have now created a version of this. The motor element now has the following elements:

- pn_mech_mw
- loading_percent --> used for active power calculation for power flow
- efficiency_percent --> used for active power calculation for power flow
- scaling --> used for active power calculation for power flow
- cos_phi --> used for reactive power calculation for power flow
- cos_phi_n --> used for short-circuit calculation
- efficiency_n_percent --> used for short-circuit calculation
- lrc_pu --> used for short-circuit calculation
- vn_kv --> used for short-circuit calculation
- rx --> used for short-circuit calculation

As you can see, I did not add any solution for defining cos_phi as a function of the loading, so it is up to the user now to update cos_phi & efficiency_percent to an appropriate value when loading_percent is updated. I did however include a seperate parameter cos_phi_n/efficiency_n_percent for the cos_phi/efficiency_percent at rated power, which is always used for short-circuit calculation. In that way, cos_phi&efficiency can be set for power flow studies without influencing the short-circuit behaviour.

This is the simplest and least opinioted implementation that I could come up with. It gives the user complete control over how to model the motor behaviour, but is not vrey user friendly, because it is up to the user to update cos_phi & efficiency with the loading. 

I am open to discussing again how to make the motor model more user friendly, but this is a good first solution imho.

@miek770: if you are still following this, I would very much appreciate your input!

miek770, 2020-08-17:
Hi @lthurner, this looks great. I converted a previous short-circuit study performed with pandapower using #865; it went smoothly and was rather intuitive.

A minor suggestion would be to initially set `cos_phi_n` equal to `cos_phi` if the former is not specified (instead of `nan` - likewise for `efficiency_percent` and `efficiency_n_percent`)... although I'm not so sure; changing `cos_phi` later on wouldn't change `cos_phi_n`, and that may or may not be what the user expects. Explicit is better than implicit, but if you only want to run short circuit calculations, `cos_phi` is both required and unused. Maybe `cos_phi` should also be optional, and the software will warn the user when either `cos_phi` or `cos_phi_n` is missing?

Ah, I read why `cos_phi` is required in #865 (power flow parameters should be required for uniformity). Sorry if I'm rambling, but that convinced me that it's probably for the best as it is (`cos_phi` required, `cos_phi_n` optional, but required for short circuit calculations - defaults to `nan`).

Didn't get a chance to test the power flow yet, but the parameters you have retained appear quite sufficient for the scenarios I can think of, and there are enough of them to handle whatever factors one might want to use elegantly (ex.: load factor, demand factor, diversity factor, operating status, etc.). I really don't mind updating `cos_phi` manually with the loading; perhaps a helper function could help in the future but I don't think it's a huge deal - this is already a huge improvement for power system studies at the distribution level.

Thanks for following up on this!

lthurner, 2020-09-01:
Thanks for the initiative and reviewing the implementation!

closed 2020-09-01 by lthurner