e2nIEE/pandapower #247: Speed improvements

lthurner, 2018-12-17:
I have significantly improved performance for elements with auxiliary buses, namely 3wtrafos and xwards by

1. Not adding auxiliary buses in the net.bus DataFrame but adding them directly in the bus_lookup
2. When converting 3W trafos to 3 2W trafos each, not creating a DataFrame to store the parameters in, but to store them in a dictionary of numpy array instead.
3. vectorize the calculation of 3W parameters instead of calculating parameters for each 3Wtrafo in a loop
4. remove some bottlenecks in build_branch and build_bus (mostly pandas indexing or inefficient mask selection)

I have tested the speedups with two networks

Network 1:

Network 2:
```
from pandapower.test.loadflow.result_test_network_generator import add_test_trafo3w
net = pp.create_empty_network()
for _ in range(500):
     add_test_trafo3w(net)
```

| Commit | Description   |   Network  1 | Network  2|
| ---------|--------------|--------------|------------|
| 5234059ca65a6c8e1bdb27ddbbd57cceb620a960 | Current implementation |  54ms (100%) | 194ms (100%) |
| cfdb83c476fb5106f72df8991abe51dea5db7f16 | Auxiliary buses in build_bus |  42ms (-22%) | 178ms (-8%) |
| 8d7cbc15114c74ab30c7d0d1dcaaca8e169ea8e6 | Trafo3w parameters in dict |  35ms (-35%) | 97ms (-50%) |
| f85706402da4977db39c780a9373677c9a46b29d | Vectorized calculation of trafo3w parameters |  31ms (-42%)  | 41ms (-78%) |
| | Removed some bottlenecks in _pd2ppc |  26ms (-52%)  | 39ms (-80%) |

wangzhenassd, 2018-12-18:
Adding some more test results on grids with trafo3ws.
Test environment: i7-4712 MQ 2.3GHz, 16GB RAM, Anaconda Python3.7 

| Item | Network 1 | Network 2| Network 3| Network 4|
| --|--|--|--|--|
| Bus | 3794 | 2563 | 3047 | 3802 |
| Trafo3w | 59 | 26 | 18 | 20 |
| Trafo | 82 | 85 | 120 | 131 |
| Line | 214 | 174 | 270 | 318 |

Testing result in ms: 

| Commit | Network 1 | Network 2| Network 3| Network 4|
| -- | -- | -- | -- | -- |
| 5234059 | 57.4 | 48.5 | 56.9 | 57.9 |
| cfdb83c | 51 | 40.2 | 48.2 | 44.7 |
| 8d7cbc1 | 40.7 | 36.1 | 45.2 | 38.3 |
| f857064 | 35.5 | 31.5 | 41.2 | 36.2 |
| b728239 | 34.3 | 29 | 38.9 | 33.2 |

WinfriedL, 2019-01-15:
I finally had the time to test with this my network. I hope the formating works, my old firefox does not show the preview anymore:

Commit | Description | time [s] | relative
---------|-------------|--------|---------
5234059 | Current implementation | 1.795 | 	100%
cfdb83c | Auxiliary buses in build_bus | 1.125 | -37%
8d7cbc1 | Trafo3w parameters in dict | 1.019 | -43%
f857064 | 	Vectorized calculation of trafo3w parameters | 0.911 | -49%
b728239 | Removed some bottlenecks in _pd2ppc | 	0.864 | -52%
89dcd12 | Trimmed some fat in result extraction | 0.843 | -53%
              | remove duplicate row  | 0.840 | -53%

The last result is when removing the duplicate wye_delta_vector() call in build_branch. Not a huge impact but should be fixed...

So great work!

lthurner, 2019-01-16:
Thanks for the timings, seems like ~50% speedup is realistic, so it was definitely worth the effort!

> The last result is when removing the duplicate wye_delta_vector() call in build_branch. Not a huge impact but should be fixed...

Thanks for noticing that, I removed it now.

closed 2019-01-16 by lthurner