e2nIEE/pandapower #251: OPF solver logger

daetzm, 2018-12-20:
Hi everyone,

I am trying to make the transition from matpower to pandapower. That is why I am trying to solve a system in pandapower that showed convergence in matpower before. So far I haven't reached convergence in pandapower when using the runopp() command. I am trying to debug my system and find out if I have not created all of the components in the network correctly. What I am missing is a command to see what the solver is doing. The solver is iterating just not converging. I would like to be able to see what the solver is doing with the buses, branches and generators. 
In matpower the command: "mpopt = mpoption('OPF_ALG', 580,'VERBOSE', 1, 'OUT_ALL', 1); % use IPOPT"   gave me information about what the solver is doing. I attached a log-file to this issue so you can see how the structure of that file is. (I had to remove the values, due to confidentiality)
[SolverLog_1.txt](https://github.com/e2nIEE/pandapower/files/2699156/SolverLog_1.txt)
Is there a way to also create such an output with pandapower or pypower?
I already tried using "pp.diagnostic(net,report_style='detailed',warnings_only=False,return_result_dict=True)"
But this only tells me that my system has a wrong_switch_configuration, even though I didn't create any switches. 

Thank you very much for your help 
Max

lthurner, 2018-12-20:
I have looked into this, and the logging mentioned above is possible with PYPOWER, but was not yet connected to pandapower. I have included this in PR #252. If you run an OPF with verbose=True on the newest version, you will get the logging. Reproduce with:

```
import pandapower as pp
import numpy as np
net = pp.create_empty_network()
#create buses
bus1 = pp.create_bus(net, vn_kv=220.)
bus2 = pp.create_bus(net, vn_kv=110.)
bus3 = pp.create_bus(net, vn_kv=110.)
bus4 = pp.create_bus(net, vn_kv=110.)
#create 220/110 kV transformer
pp.create_transformer(net, bus1, bus2, std_type="100 MVA 220/110 kV")
#create 110 kV lines
pp.create_line(net, bus2, bus3, length_km=70., std_type='149-AL1/24-ST1A 110.0')
pp.create_line(net, bus3, bus4, length_km=50., std_type='149-AL1/24-ST1A 110.0')
pp.create_line(net, bus4, bus2, length_km=40., std_type='149-AL1/24-ST1A 110.0')
#create loads
pp.create_load(net, bus2, p_kw=60e3, controllable = False)
pp.create_load(net, bus3, p_kw=70e3, controllable = False)
pp.create_load(net, bus4, p_kw=10e3, controllable = False)
#create generators
eg = pp.create_ext_grid(net, bus1, min_p_kw = -1e9, max_p_kw = 1e9)
g0 = pp.create_gen(net, bus3, p_kw=-80*1e3, min_p_kw=-80e3, max_p_kw=0,vm_pu=1.01, controllable=True)
g1 = pp.create_gen(net, bus4, p_kw=-100*1e3, min_p_kw=-100e3, max_p_kw=0, vm_pu=1.01, controllable=True)
costeg = pp.create_polynomial_cost(net, 0, 'ext_grid', np.array([-1, 0]))
costgen1 = pp.create_polynomial_cost(net, 0, 'gen', np.array([-1, 0]))
costgen2 = pp.create_polynomial_cost(net, 1, 'gen', np.array([-1, 0]))
pp.runopp(net, verbose=True)
```
Gives you:

```
================================================================================
|     System Summary                                                           |
================================================================================

How many?                How much?              P (MW)            Q (MVAr)
---------------------    -------------------  -------------  -----------------
Buses              4     Total Gen Capacity   1000180.0       -3000000000.0 to 3000000000.0
Generators         3     On-line Capacity     1000180.0       -3000000000.0 to 3000000000.0
Committed Gens     3     Generation (actual)    140.1              -1.4
Loads              3     Load                   140.0               0.0
  Fixed            3       Fixed                140.0               0.0
  Dispatchable     0       Dispatchable           0.0 of 0.0        0.0
Shunts             0     Shunt (inj)              0.0               0.0
Branches           4     Losses (I^2 * Z)         0.09              3.85
Transformers       4     Branch Charging (inj)     -                5.3
Inter-ties         0     Total Inter-tie Flow     0.0               0.0
Areas              1

                          Minimum                      Maximum
                 -------------------------  --------------------------------
Voltage Magnitude   0.998 p.u. @ bus 1          1.000 p.u. @ bus 3   
Voltage Angle      -3.89 deg   @ bus 1          0.00 deg   @ bus 0   
P Losses (I^2*R)             -                  0.08 MW    @ line 0-1
Q Losses (I^2*X)             -                  3.83 MVAr  @ line 0-1

================================================================================
|     Area Summary                                                             |
================================================================================
Area  # of      # of Gens        # of Loads         # of    # of   # of   # of
 Num  Buses   Total  Online   Total  Fixed  Disp    Shunt   Brchs  Xfmrs   Ties
----  -----   -----  ------   -----  -----  -----   -----   -----  -----  -----
  1       4       3      3       3      3      0       0       4      4      0
----  -----   -----  ------   -----  -----  -----   -----   -----  -----  -----
Tot:      4       3      3       3      3      0       0       4      4      0

Area      Total Gen Capacity           On-line Gen Capacity         Generation
 Num     MW           MVAr            MW           MVAr             MW    MVAr
[... 62 lines cut ...]

Which gives you:

``` 
hp.pandapower.toolbox - INFO: These missing columns in ext_grid are considered in OPF as +- 1000 TW.: ['min_q_kvar' 'max_q_kvar']
hp.pandapower.toolbox - INFO: These elements have missing power constraint values, which are considered in OPF as +- 1000 TW: ['gen']
hp.pandapower.toolbox - INFO: min_vm_pu is missing in bus table. In OPF these limits are considered as 0.0 pu.
hp.pandapower.toolbox - INFO: max_vm_pu is missing in bus table. In OPF these limits are considered as 2.0 pu.
hp.pandapower.toolbox - INFO: 
Cotrollables & Costs:
  Ext_Grid
    all 1 Ext_Grid with p costs
  Gen
    all 2 Gen with p costs
Constraints:
  External Grid Constraints
    at all Ext_Grid [min_p_kw, max_p_kw, min_q_kvar, max_q_kvar] is [-1000000000.0, 1000000000.0, nan, nan]
  Generator Constraints
    at Gen 0 [min_p_kw, max_p_kw, min_q_kvar, max_q_kvar] is [-80000.0, 0.0, nan, nan]
    at Gen 1 [min_p_kw, max_p_kw, min_q_kvar, max_q_kvar] is [-100000.0, 0.0, nan, nan]
```

This might also help you with debugging to see which constraints and costs are defined.

Please let me know if it works for you.

daetzm, 2018-12-21:
Thank you very much for your help. This is exactly what I wanted. My case is till not converging, but now I can at least see where problems occur. 

I wish you a Merry Christmas

lthurner, 2019-01-07:
You are welcome! I would still advise you to switch to pandapower 2.0 (dev_2.0 branch). If you are coming from MATPOWER, this will me be much easier because the signing system is the same (no negative power for generators) and the units are the same (MW instead of kW). It is very possible that you made a mistake with these assumptions in pandapower 1.x that are different from MATPOWER.

closed 2019-01-07 by lthurner