e2nIEE/pandapower #253: State Estimation error with bus-bus switches and measurements on each bus

wangzhenassd, 2018-12-20:
Hi,
i am currently working on a high voltage network with lots of measurements. The bus of the pandapower network represents a busbar in the substation in the real world and the buses were connected with each other through bus-bus switches. The measurement of bus (p, q) represents the power flow on a busbar. However the state estimation module (and similarly in power flow calculation) will merge the buses connected through bb-switch to a single bus and which was not in the _add_measurement function the measurement correctly merged. 

I have tried to connect the bus with measurement (and also load or sgens) with lines instead of bb-switches as small impedance. This work around doesn't work well, since from one side the large admitance value leads to error within the linalg block in numpy and on the other side increases the buses and branches in ppc which will then require more measurements and more time for the computation. 

I will post one example of this problem later on.

wangzhenassd, 2019-01-03:
I am thinking of a possible fix to the issue with minimum change of the original process. Since the reason for the problem is that on the ppci level, multiple measurements (and also load, sgens) are fused together onto one single bus. My idea is to identify such cluster (ppci bus) with many-to-one bus mapping and using auxiliary buses on such a cluster with mupltiple devices to achieve the minimum code adjustment. 

1.  if a cluster has more than one buses with devices, add auxiliary buses for all the buses except the first one (to avoid generating a bus with zero p,q) to substitute the original bus in the ppci cluster
2. set all the bb switch onto the original bus to out of service and add auxiliary bb switches onto the newly added buses with the same configuration
3. add lines with small impedance between the auxilliary buses and the buses, which were substituted by the auxilliary buses
4. run the normal state estimation
5. remove all auxiliary buses, bb switches and lines, restore the in_service state of the original bb-switch
6. remove state estimation results for the auxiliary elements

The results will thus mapping correctly onto the original configuration.

jhmenke, 2019-01-04:
fixed in pull request

PR #258 (2019-01-04): Fix #253 Avoid auto fusing of buses with elements and bb switches in state estimation
Fix #253 

- in case of bb switches, with the option fuse_all_bb_switch disabled, the module will find such auto fused buses with elements on them, and create auxiliary buses to substitute the bus in the original topology with bb switches, the original bus will then connect to the auxiliary bus with a short piece of line, thus avoiding
auto fusing, this process will be done in the estimation function before the pd2ppc conversion and and cleaned at the end of the function ofr the unnecessary information

- Add [... cut]
commit c5a21a95 (2019-01-04): Fix #253 with substitution of the originally fused bus with elements with line and auxiliary buses,
added test case
commit f12f7371 (2019-01-04): Merge pull request #258 from wangzhenassd/dev_2.0

Fix #253 Avoid auto fusing of buses with elements and bb switches in state estimation
closed 2019-01-04 by jhmenke