e2nIEE/pandapower #259: Suggestion on handling bus-bus switches in pd2ppc

wangzhenassd, 2019-01-07:
Hi,
regarding the modelling of a high-voltage grid with substation with multiple busbars, these busbars are connected with each other through bus-bus switches (bb switch). As default these switches will be fused to one single bus in ppc and ppci, and there is the alternative of handling it "r_switch", which modeled the bus-bus switch as small impedance (all of them).

While i am working on a high-voltage grid with such configuration, that a substation has multiple busbars (>=5) modeled as bus in the pandapower network, while only e.g. two of them has an element connected as used for testing in #247. Supposing that both elements are properly measured, fusing them together (default behaviour) will lose information on that and cause failure in state estimation. While with "r_switch" activated, it will produce too many buses with zero injection in ppc and ppci, which would cause problem mentioned in #243. And for the power flow, the increment in the size of matrix will somehow lower the efficiency of calculation. 

My suggestion is to change the behaviour of the "r_switch" to which only necessary bb switches (e.g. between the buses with elements) will be replaced by impedance (line) to garantie the rightfulness in state estimation (as mentioned in #248 and fixed in PR #258).  These method was realized on the pandapower level with manuelly adding auxiliary bus. This kinds of process could be also implemented directly within _pd2ppc. What's your opinions on the topic guys?

Thanks in advance!

wangzhenassd, 2019-01-08:
Since it does not bring a lot of difference in power flow calculation and would be a little confusing to the behaviour of 'r_switch', a feasible method of doing this would be:
1.  Add columns for impedance in the switch table, if impedance is 0 then the buses will be fused as by default, if impedance is larger than zero, a impedance will be created in ppc and ppci
2.  Create a function to identify such configuration and add impedance in the switch table

ascheidl, 2019-01-08:
I think this is a good solution

wangzhenassd, 2019-01-08:
Thanks! I am starting working on it.

wangzhenassd, 2019-01-09:
> Maybe, wait for our next pp meeting. So we can determine how to implement it in the best way and what things have to be considered.

Sure @ascheidl , i will think about how to do this before the meeting and update the changes to be made in this thread.

lthurner, 2019-02-18:
Discussed today:

- introduce mandatory net.switch.z_ohm column for switches (add in convert_format for legacy support)
- switches with z_ohm=0 are fused, switches with z_ohm > 0 are modelled as branches
- r and x for each "branch" switch are calculated with a fixed r/x ratio that can optionally be specified as a parameter in runpp. Open question: what would be a good default value? It cannot be 1, because the DC runpp doesn't work with x=0 (see above)

lthurner, 2019-03-20:
Closed with #335

closed 2019-03-20 by lthurner