e2nIEE/pandapower #260: Add ideal transformers to impedances

WinfriedL, 2019-01-08:
I have impedances with ideal transformers at the end in my data. I have added this to pandapower by adding the columns 'ur1_kv' and 'ur2_kv' to the impedance table and calculate an off nominal tap ratio from these. Would you be interested in adding such functionality? In that case I am going to open a pull request.
The column names are of course open for discussion, one could also add a single columns for the tap ratio. 

I have tested/compared this with Integral. I would be interested if other tools like powerfactory have this kind of impedance as well?

WinfriedL, 2019-01-10:
I have created the pull request, with this it might be clearer what I have in mind.

If you consider this addition there are some open points which have to be discussed before merging:

- Integral does not allow setting an off nominal tap ratio for an asymmetric impedance. This might make sense, but I don't know if this should also be enforced by pandapower
- I am not totally sure about the impedance pu conversion with this addition. Integral uses Ohm for R and X. Usually when I convert this to pu I would just divide by `vn1_kv ** 2 / sn_mva` where vn1_kv is the nominal voltage of the from_bus. But with an additional voltage converter I have to correct the impedance values for the off nominal tap ratio (squared). Currently the user has to take care of this, but I am not sure if this is the right approach.
- Currenty I calculate the taps only if there are any ur1_kv or ur2_kv set, but in that case for all impedances, using the nominal voltages for missing values. This might be bad if there are lots of impedances, but only few have an off nomial ratio because in that case there are lots of ones calculated. I don't know what the best numpy approach would be.
- Also in the current implementation the missing ur in the impedance dataframe would be filled with the nominal bus voltages after the powerflow. This might be considered bad, since the powerflow should not modify the pandapower net.

lthurner, 2019-01-16:
> I have tested/compared this with Integral. I would be interested if other tools like powerfactory have this kind of impedance as well?

Yes this exists in PowerFactory as well. In PowerFactory it is a tap ratio, not vn1_kv & vn2_kv.

> I am not totally sure about the impedance pu conversion with this addition. Integral uses Ohm for R and X. Usually when I convert this to pu I would just divide by vn1_kv ** 2 / sn_mva where vn1_kv [...]

Not sure also, would have to look into it. Is there a reason it is from_bus and not to_bus or just by convention? 

> Also in the current implementation the missing ur in the impedance dataframe would be filled with the nominal bus voltages after the powerflow. This might be considered bad, since the powerflow should not modify the pandapower net.

Yes it should also be possible to read the voltage value directly from the ppc, so that the pandas table ist not changed and it is also more performant (using numpy instead of pandas).

> Currenty I calculate the taps only if there are any ur1_kv or ur2_kv set, but in that case for all impedances, using the nominal voltages for missing values. This might be bad if there are lots of [...]

If we decide to use a tap_ratio instead of vn1/vn2 it could just default to 1, whereas having vn1/v2 doesn't really have a logical default. If there is no significant advantage in specifically definining vn1/vn2, I would preferr this over having some case by case distinction.

WinfriedL, 2019-01-17:
I have updated the implementation to use tap_ratio as parameter. 

One still open question is, on which side of the branch the voltage converter should be. This does not matter for tap_ratio=1.
- At the from side (current implementation) 
- At the to side (Integral implementation)
- Configurable (like for transformers)

I have no idea if there is a convention or if most tools use one specific side, or if it is more or less random.
I would recommend one of the first two options (along with clear documentation) and avoid the complexity of the third.