e2nIEE/pandapower #262: init='results' does not check if there are results for aux buses

WinfriedL, 2019-01-10:
init='results' does not check if there are results for aux buses.

Example:

```
    net=pp.create_empty_network()
    
    pp.create_bus(net=net,name="slack_bus", vn_kv=1)
    pp.create_bus(net=net,name="slack_bus", vn_kv=0.1)
    
    pp.create_impedance(net, from_bus=0, to_bus=1, rft_pu=0, xft_pu=1, sn_mva=1)

    pp.create_ext_grid(net=net, bus=0, vm_pu=1.0)
    pp.create_xward(net, bus=1, ps_mw=0, qs_mvar=0, pz_mw=0, qz_mvar=0, r_ohm=0, x_ohm=6, vm_pu=1.0)

    pp.runpp(net, init='results')

RuntimeWarning: invalid value encountered in true_divide
  dVm_x, dVa_x = dSbus_dV_numba_sparse(Ybus.data, Ybus.indptr, Ybus.indices, V, V / abs(V), Ibus)
```

One can consider this as a user error, but might still be better to check for it.

WinfriedL, 2019-01-10:
I think the PR now passes all testcases and my rather specific "mis"treatment of init="results", which I still use in my code. Let me know if you see a more elegant way to deal with this.

lthurner, 2019-01-16:
Thanks for bringing this up, this should definitely be checked for. For the initialization of normal buses, we just check if the net.res_bus table exists, and if not reverts to flat start:

https://github.com/e2nIEE/pandapower/blob/e67072206348803635ff6de98c8048dea9a9c16e/pandapower/build_bus.py#L181-L184

The way you implemented it now in #263 checks which buses have results and only uses the buses with results. I think we should keep both cases consistent. Is there a valid use case where some buses have results and others don't? If not, I would prefer the binary solution that we also use for regular buses...

WinfriedL, 2019-01-17:
> Is there a valid use case where some buses have results and others don't?

Yes. Aux buses which belong to out_of_service elements or disconnected elements have no results. So my first implementation which checks if all results are finite would not work. One could check if any result is finite, I am not sure anymore if there was a reason why I did not do that. 

> I think we should keep both cases consistent.

I agree. Maybe the easiest way would be to check the sanity of init='results' in the beginning, maybe in runpp() or _powerflow()? In a way this would be consistent with other options like voltage_depend_loads in https://github.com/e2nIEE/pandapower/blob/db8681ed63c26b4ca13fed680e94d6e2341c974b/pandapower/run.py#L239-L241
And then the individual checks in build_bus could be removed?

lthurner, 2019-02-26:
> I agree. Maybe the easiest way would be to check the sanity of init='results' in the beginning, maybe in runpp() or _powerflow()?

Yes that is the most elegant solution, thanks for looking into it. I implemented it this way in #311 and also added a test that uses init="results" on a test network with trafo3w, xward and open line switch, so that all possible cases should be covered.

closed 2019-02-26 by lthurner