e2nIEE/pandapower #266: time series example

schmocker, 2019-01-12:
hi,
I'm using paneapower for time series load flow simulation. I'm trying to figure out the fastest way to loop over the time series. I was able to get down to 0.009 seconds per iteration with my current grid, but I'm not really sure if that's the limit :)
It would be very nice of you to provide an example or even a some methods to efficiently iterate over time series.
Attached you'll finde my current code. I read data from a database into the pd dataframe `load_df` where I have the index set to the loads id's and columns for date_time (for each time step) and for p_kw and q_kvar (for each time step and id) . Further I have my pandapower net soared into `pp_grid`.

---
1. Grouping them per `date_time `
```python
# split data frame into data frames per date time
dfs = [(date_time, df_per_dt) for date_time, df_per_dt in load_df.groupby('date_time')]
```
2. Setting some acceleration variables - even thought I don't really understand them - but its much faster
```python
# -> recycle internal pandapower variables for simulation acceleration
recycle = {'_is_elements': True, 'ppc': True, 'Ybus': True}
init = "results"
```
3. Preparing the results to be stored
```python
res = {'date_time': [], 'bus': [], 'trafo': []}
```
4. Looping **!! this part needs to be as fast as possible !!**
```python
for dt, cur_df in dfs:
     try:
             pp_grid['load'].update(cur_df)
             pp.runpp(pp_grid, init=init, recycle=recycle)
     except Exception as e:
            logger.warn(f"Power Flow @ {dt} raised an error\n-> {e}")
     else:
             res['date_time'].append(dt)
             res['bus'].append(pp_grid['res_bus'].copy())
             res['trafo'].append(pp_grid['res_trafo'].copy())
```
5. Concat the result tables back together....
```python
res_bus = pd.concat(res['bus'], keys=res['date_time'], names=['date_time', 'fk_ bus'])
res_trafo = pd.concat(res['trafo'], keys=res['date_time'], names=['date_time', 'fk_ trafo'])
```
---
I found that using `update` with pandas is quite fast. I hope I could provide a good first example. I think I'm not the only one having spent hours on looping pandapower time series simulations efficiently. So it would be very nice to get a feedback on my method and to provide an example or methods/functions for this problem. It would also be very good to describe the advantages and disadvantages of using `recycle` and `init` for simulation acceleration.

Thank you very much!

BTW: I could think about splitting the the time series up into blocks of lets say of 5'000 time steps (I have 35'040 of 15 min over a year) and run them in parallel. But I'm not (yet) that advanced in python programming...

jhmenke, 2019-01-14:
>It would also be very good to describe the advantages and disadvantages of using recycle and init for simulation acceleration.

recycling of ppc/Ybus is possible and will improve the speed tremendously for time series. therefore, it should be used if applicable. however, if the grid topology or grid elements change in the timeseries (not only load/feed-in value), the Ybus/ppc arrays change and cannot be recycled. I'm not sure if that's automatically handled (with higher simulation times) or the results are just faulty as it is.

FlorianShepherd, 2019-02-27:
> however, if the grid topology or grid elements change in the timeseries (not only load/feed-in value), the Ybus/ppc arrays change and cannot be recycled. I'm not sure if that's automatically handled (with higher simulation times) or the results are just faulty as it is.

Not it is not checked and cannot be without higher simulation times, since checking if the topology (Ybus) changes requires to somehow "track" the difference and that is not implemented.

FlorianShepherd, 2019-08-06:
Please checkout the latest develop branch with the control module + time series module for time series calculations. 
If you have any more questions, feel free to reopen the issue.

closed 2019-08-06 by FlorianShepherd