e2nIEE/pandapower #268: Suggestion on structure optimization for module State Estimation

wangzhenassd, 2019-01-15:
The core of state estimation is to use the measurement redundancy to estimate the real grid status (Vm, Va, power injection on buses and power flow on branches).  In current implementation, the process can be concluded as follows:

1.  pd2ppc: transform pandapower networks to ppc and ppci format
2.  add_measurement_to_ppc: adding the measurement form pandapower network to the ppci 
3.  matrix initialization and calculation : do all the wls mathematical operation -> calculate Vm, Va
4.  calculate branch flow
5.  extract result ppci -> ppc -> pandapower network

As referd in #243, in order to solve the issue, a second estimator formulation, which is quite different as the standard wls estimator, is needed to formulate the constraint.  Thus i would suggest that, it would be nice to pack all the matrix operation based on the ppci into single functions (objects) so that the behaviour of the modul (the estimator) can be freely selected and expanded with e.g. other estimator formulations.

jhmenke, 2019-01-16:
I agree that **if** new estimation methods are going to be added we should capsule each one into separate functions.

>As referd in #243, in order to solve the issue, a second estimator formulation, which is quite different as the standard wls estimator, is needed to formulate the constraint.

I'd argue that the addition of zero injection constraints just supplements the WLS SE and is not a different method. It is always desirable to include the zero injection constraints as they increase the estimation quality, an option to disable them is not required (or is it?).

wangzhenassd, 2019-01-16:
> I'd argue that the addition of zero injection constraints just supplements the WLS SE and is not a different method. It is always desirable to include the zero injection constraints as they increase the estimation quality, an option to disable them is not required (or is it?).

I think it would be nice to make this option selectable. I am working on a case might require the original functionality. The situation is, i have a converted network with some parameter error (possibly tap pos) on the trafo3w which leads to a large difference on the estimation result. I want to identify the tap position of the the trafo with the v measurement on bus and pq measurement on trafo3w by a subnet from the original net. To make it work, the bus with/without injection status is unknown and thus the assumption that bus without p,q measurement and element on it to be zero injection bus must be disabled.

jhmenke, 2019-01-17:
due to your pull request
- other estimators can be included easily
- zero injection handling can be manually deactivated if need be

therefore, i close this issue.

PR #269 (2019-01-17): Add support true zero injection and capsuled the matrix operation in state estimation
Fixed #243, 
Fixed #268
closed 2019-01-17 by jhmenke