e2nIEE/pandapower #272: Short-Circuit Calculation ERROR

popwin, 2019-01-23:
Hello
I'm doing a Short-Circuit Calculation Test following the short-circuit example of official Documents with pandapowerv1.6.0. But I get a Error:

```
Traceback (most recent call last):
  File "/home/xyz/Desktop/powerCal/shortCC.py", line 13, in <module>
    sc.calc_sc(net, case="min")
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py", line 122, in calc_sc
    _calc_sc(net)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py", line 134, in _calc_sc
    ppc, ppci = _pd2ppc(net)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/pd2ppc.py", line 90, in _pd2ppc
    _add_gen_impedances_ppc(net, ppc)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/build_bus.py", line 463, in _add_gen_impedances_ppc
    _add_gen_sc_impedance(net, ppc)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/build_bus.py", line 516, in _add_gen_sc_impedance
    phi_gen = np.arccos(gen.cos_phi)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandas/core/generic.py", line 4376, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'DataFrame' object has no attribute 'cos_phi'

```
Can you Help me with that error?
Thank You very Much.

popwin, 2019-01-27:
> "k" is an optional parameter in the data structure but a necessary parameter for the short-circuit calculation. (see [documentation](https://pandapower.readthedocs.io/en/v1.6.0/elements/sgen.html#input-parameters)). This:
> 
> ```
> import pandapower.networks as nw
> 
> net = nw.mv_oberrhein()
> net.ext_grid["s_sc_min_mva"] = 100
> net.ext_grid["rx_min"] = 0.1
> net.sgen["k"] = 1.1
> 
> net.line["endtemp_degree"] = 20
> sc.calc_sc(net, case="min")
> print(net.res_bus_sc)
> ```
> works fine for me. We should definitely include a specific UserWarning here (something like "please specify k to run a short-circuit calculation") instead of just raising the pandas error.
> 
> I am not sure what example you are running in the first example, can you specify?

lthurner,Thanks for your replying.
I have added the k value  of static generator as you said when I run the **mv_oberrhein** example. And it  is worked. Thank you for your help.

And the first example is case5 of official documents, Networks Part.

But when I run it ,it will show the error just as I have published above. `AttributeError: 'DataFrame' object has no attribute 'cos_phi'`

So, I set `net.gen["cos_phi"] = 0.8`  and run  short-circuit  calculation of the case5 again 
and it showed some errors like 
```
has no attribute 'xdss'
has no attribute 'vn_kv'
has no attribute 'rdss'
```  
Finally, I set all the  *vn_kn xdss and rdss* values   just like 

```
import pandapower as pp
import pandapower.networks as pn 
import pandapower.plotting as Pplot
from pandapower.plotting.plotly import simple_plotly
import pandapower.shortcircuit as sc

net = pn.case5()
net.ext_grid["s_sc_min_mva"] = 100
net.ext_grid["rx_min"] = 0.1
net.line["endtemp_degree"] = 20
net.gen["cos_phi"] = 0.8
net.gen["vn_kv"] = 230
net.gen["xdss"] = 0.2
net.gen["rdss"] = 0.1
sc.calc_sc(net, case="min")
print(net.res_bus_sc)
```

But some errors still happened 

```
Traceback (most recent call last):
  File "/home/xyz/Desktop/powerCal/code/shortCC.py", line 16, in <module>
    sc.calc_sc(net, case="min")
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py", line 122, in calc_sc
    _calc_sc(net)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py", line 142, in _calc_sc
    raise(e)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/shortcircuit/calc_sc.py", line 139, in _calc_sc
    _calc_zbus(ppci)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/pandapower/shortcircuit/impedance.py", line 48, in _calc_zbus
    ppc["internal"]["Zbus"] = inv(Ybus.toarray())
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/scipy/linalg/basic.py", line 945, in inv
    a1 = _asarray_validated(a, check_finite=check_finite)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/scipy/_lib/_util.py", line 239, in _asarray_validated
    a = toarray(a)
  File "/home/xyz/anaconda3/envs/GriCP/lib/python3.7/site-packages/numpy/lib/function_base.py", line 498, in asarray_chkfinite
    "array must not contain infs or NaNs")
ValueError: array must not contain infs or NaNs

```
Am I wrong in somewhere?

lthurner, 2019-03-14:
Better warnings are now implemented with #331

closed 2019-03-14 by lthurner