e2nIEE/pandapower #276: Include module 'packaging' in the pandapower requirements

SteffenMeinecke, 2019-01-25:
There are recurrent issues concerning version handling. This is why I propose to include 'packaging' in the pandapower requirements. Let's consider pandas examples:
Since pandas changes function names like reindex_axis() to reindex(), we have code like

try:
    net["trafo3w"] = net["trafo3w"].append(dd).reindex(...)
except TypeError: # legacy for pandas <0.21
    net["trafo3w"] = net["trafo3w"].append(dd).reindex_axis(...)

This works but using packaging should be more elegante.

from packaging import version
if version.parse(pd.__version__) >= version.parse("0.21.0"):
    net["trafo3w"] = net["trafo3w"].append(dd).reindex(...)
else:
    net["trafo3w"] = net["trafo3w"].append(dd).reindex_axis(...)

Moreover, using packaging enables distinguishing more than 2 version cases without creating try-except-cascades. For example:

if version.parse(pd.__version__) >= version.parse("0.23.0"):
    net["trafo3w"] = net["trafo3w"].append(dd, sort=False).reindex(...)
elif version.parse(pd.__version__) >= version.parse("0.21.0"):
    net["trafo3w"] = net["trafo3w"].append(dd).reindex(...)
else:
    net["trafo3w"] = net["trafo3w"].append(dd).reindex_axis(...)

jhmenke, 2019-01-28:
packaging is already used to detect plotly - it's optional there, but another reason to include packaging in the requirements

lthurner, 2019-02-18:
agreed, let's include packaging in the requirements.txt and use it to check versions

commit 30271627 (2019-03-20): added packaging to requirements and adapt all version checks to use packaging closes #276
closed 2019-03-20 by lthurner