e2nIEE/pandapower #277: Enhancement in State Estimation

wangzhenassd, 2019-01-28:
After discussion with @jhmenke , because of the complexity in the topology and potential errors in measurement and digital communication in real grid, following advices as milestones for the enhancement onto the state estimation module are proposed. Any contribution is welcome and please feel free to add and discuss in this thread.

1. Function parameter and Measurement Preprocessing:
   1. Parameter Zero Injection:
       - Since in some cases the injection status of buses is not certain, it would be logical to give flexibility in selectin the behaviour of the function. The proposol would be similar to add a parameter similar to the v_start in the power flow, which accepts string, boolean and also a collection of selected buses. For example: "auto", "aux_bus_only" (only aux buses will be considered zero injection), "all" (True, all bus without any element and any p,q measurement), "None" (False, all buses considered possibly with injection).
   2. Handling duplicated Measurement (Same type of measurement at the same point in net):
       - In a real-world network, it's often to have a substation with multiple busbars connected by bb switches and with loads on them, if they both have p,q measurements, while share the same ppci bus in the calculation, then the both measurements have to be properly handled (e.g. adding them together) to avoid offering wrong measurement to the estimator, as an alternative to handle the bb switch as a small piece of impedance. And it should also raise warning to the user, that this process happened, since the results on each buses cannot be identified individually. In other cases e.g. with two p measurements at the same ends of a line, a warning will be raised while one of them will be randomly deleted.

 2. Enhancement in estimation process:
    1. Structure optimization and add interface to the SciPy Optimization Tool
    2. Implement robust estimator e.g. SHGM or LAV as supplement to the classic WLS method

3. Adding functionality in identifing topological errors

jhmenke, 2019-04-26:
is this issue closed by PR #370 ?

wangzhenassd, 2019-04-26:
> is this issue closed by PR #370 ?

I think we could close the issue now. The job is done.
There is a point from my point of view, which might make sense to make the state estimation module better:

- Observability analysis, since "2n-k" measurements won't guarantee the full observability of the grid. Some isolated branches/islands are still possible to exist. A better way to solve the issue would be to apply the numerical process to analysis the observability and add virtual measurements/ raise warning, which avoids algorithms collapse in such case.

closed 2019-04-26 by wangzhenassd