e2nIEE/pandapower #289: Adapting colorbars

dlohmeier, 2019-02-15:
When plotting with colormaps, the settings of how to draw colorbars are pretty much fixed in pandapower. If you call `draw_collections()`, you pass the argument plot_colorbars with just True/False, and then it is pretty much hard coded, how this is done (i.e. concerning size, fontsize etc.). For this reason I wrote a small auxiliary function to adapt the settings, as calling `pyplot.colorbar()` offers quite a few of them. The function looks like this:

        def adapt_cbar_ax(ax, shrink=None, aspect=None, fraction=None, ylabel=None,
                          label_fsize=None, tick_fsize=None):
        	args = dict()
        	if shrink is not None:
        		args["shrink"] = shrink
        	if aspect is not None:
        		args["aspect"] = aspect
        	if fraction is not None:
        		args["fraction"] = fraction
        	for c in ax.collections:
        		if c.colorbar is not None:
        			if len(args) > 0:
        				cb = c.colorbar
        				cb.remove()
        				extend = c.extend if hasattr(c, "extend") else "neither"
        				plt.colorbar(c, extend=extend, ax=ax, **args)
        			cb_new = c.colorbar
        			if hasattr(c, "cbar_title"):
        				yl = c.cbar_title
        				if ylabel is not None:
        					cb_new.ax.set_ylabel(ylabel)
        					yl = ylabel
        				if label_fsize is not None:
        					cb_new.ax.set_ylabel(yl, fontsize=label_fsize)
        				if tick_fsize is not None:
        					cb_new.ax.tick_params("y", labelsize=tick_fsize)

It takes the ax of the network plot, checks for colorbars and adapts all of them with the given settings. I actually only tested it with one colorbar, so maybe it doesn't work too well for more. I would still check on this. Do you think this would be helpful to include in pandapower somewhere?

I am not too sure if this function does exactly what is required, but first of all wanted to ask if this makes sense anyway, and then try to test it more in detail.

lthurner, 2019-02-18:
I agree the current solution is not very elegant, as you mentioned it is added in a draw_collections subfunction if the flag is set. There are some possibilities to add a title and specify the extension, but it is all very implicit by adding a parameter to the collection (c.extend, c.bar_title):
https://github.com/e2nIEE/pandapower/blob/9a7046ccfb4d46ce724d7fc41fcd57dc53048b25/pandapower/plotting/collections.py#L931-L939
Alternatively to adapting the colormap after it is drawn, could we also remove the plot_colorbar parameter and instead make a second function that only draws the colorbars (with the possibility to specify the parameters)?

rbolgaryn, 2020-10-14:
Can the issue be closed?

closed 2025-03-28 by vogt31337