e2nIEE/pandapower #308: convert_format order of execution

julffers, 2019-02-25:
I discovered a minor flaw regarding the implementation of convert_format. In convert_format there is a code block that converts the dtypes of all DataFrame columns to the correct type.

```new_net = create_empty_network()
    for key, item in net.items():
        if isinstance(item, pd.DataFrame):
            for col in item.columns:
                if key in new_net and col in new_net[key].columns:
                    if set(item.columns) == set(new_net[key]):
                        try:
                            net[key] = net[key].reindex(new_net[key].columns, axis=1)
                        except TypeError:  # legacy for pandas <0.21
                            net[key] = net[key].reindex_axis(new_net[key].columns, axis=1)
                    if int(pd.__version__[2]) < 2:
                        net[key][col] = net[key][col].astype(new_net[key][col].dtype,
                                                             raise_on_error=False)
                    else:
                        net[key][col] = net[key][col].astype(new_net[key][col].dtype,
                                                             errors="ignore")```

This conversion however is done **before** trafo parameter names are converted to the pp 2.0 naming scheme.

If you convert a network with old trafo parameter names ("tp_st_degree" etc.) **and** invalid dtypes in these columns this leads to a situation where the the dtypes of some trafo columns will not be converted due to the order of code execution in convert_format.

In this minimal example there is a network with an old trafo column name "tp_st_degree" and an invalid dtype (Object). Executing convert_format one time renames "tp_st_degree" to "tap_step_degree" but the dtype of this column remains the same. As a consequence pp.runpp(net) throws  an error. Executing convert_format a second time fixes that.

´´´
import networks as nw
import pandapower as pp

net = nw.simple_mv_open_ring_net()
net.trafo.rename(columns={"tap_step_degree": "tp_st_degree"}, inplace=True)
net.trafo.tp_st_degree = None
net = pp.convert_format(net)
pp.runpp(net)
net = pp.convert_format(net)
pp.runpp(net)
```

Is there a particular reason for this order? In my opinion checking the columns' dtypes should be the last step in convert_format.

lthurner, 2019-03-14:
> Is there a particular reason for this order? In my opinion checking the columns' dtypes should be the last step in convert_format.

I don't think there is a particular reason, feel free to change the order.

lthurner, 2019-03-15:
I now completely refactored convert_format and also added the nw.example_multivoltage() network saved with each version of pandapower going back to 1.2.0 in /test/test_files/old_version. There is a test in test/api/test_convert_format.py that loads the network in each version and checks that the power flow can be run and the results are the same as in the old version. This should give the convert_format function more stability. See #336

closed 2019-03-15 by lthurner