e2nIEE/pandapower #314: Three winding transformer model inconsistent with documentation

gvdwel, 2019-02-28:
Hello,

I have been looking into the power flow calculations of a simple network with a 3 winding transformer and found some inconsistencies. 
In the documentation it is nicely explained how the 3 winding transformer is modeled as three 2 winding transformers, including the conversion of the parameters. I decided to try this myself and followed the description to create my own 3w trafo equivalent. When comparing the power flow results with Pandapowers 3w trafo I noticed some differences. Comparing both with commercial software (Neplan) I find that the Neplan results match with my 3w trafo model.

So to me the description in the documentation looks fine, but the implementation of the three winding transformer in the code must be different somehow. I understand that Pandapower is tested against Powerfactory, so I am a bit puzzled by this.
Can someone shed some light on this?

gvdwel, 2019-02-28:
Here is the code I used:

```
import numpy as np
import pandapower as pp

# 3W trafo params
vsc_hv_percent = 12.57
vsc_mv_percent = 16.49
vsc_lv_percent = 10.18
vscr_hv_percent = 0.19
vscr_mv_percent = 0.30
vscr_lv_percent = 0.276
sn_hv_kva = 250000.
sn_mv_kva = 75000.
sn_lv_kva = 75000.
vn_hv_kv = 250.
vn_mv_kv = 165.
vn_lv_kv = 23.7
vnom_hv = 220.
vnom_mv = 150.
vnom_lv = 22.
i0 = 0 #0.313
pfe = 0 #78.2

net = pp.create_empty_network()
b1 = pp.create_bus(net, vn_kv=vnom_hv, name="Bus1")
b2 = pp.create_bus(net, vn_kv=vnom_mv, name="Bus2")
b3 = pp.create_bus(net, vn_kv=vnom_lv, name="Bus3")
pp.create_ext_grid(net, bus=b1, vm_pu=1.1, name="Feeder")
pp.create_load(net, bus=b2, p_kw=25000., q_kvar=10000., name="Load2")
pp.create_load(net, bus=b3, p_kw=25000., q_kvar=10000., name="Load3")

pp.create_transformer3w_from_parameters(
        net, hv_bus=b1, mv_bus=b2, lv_bus=b3,
        vn_hv_kv=vn_hv_kv, vn_mv_kv=vn_mv_kv, vn_lv_kv=vn_lv_kv,
        sn_hv_kva=sn_hv_kva, sn_mv_kva=sn_mv_kva, sn_lv_kva=sn_lv_kva,
        vsc_hv_percent=vsc_hv_percent, vsc_mv_percent=vsc_mv_percent, vsc_lv_percent=vsc_lv_percent,
        vscr_hv_percent=vscr_hv_percent, vscr_mv_percent=vscr_mv_percent, vscr_lv_percent=vscr_lv_percent,
        pfe_kw=pfe, i0_percent=i0)

pp.runpp(net)

# create equivalent circuit with 3 2W trafos and compare:
net2 = pp.create_empty_network()
b4 = pp.create_bus(net2, vn_kv=vnom_hv, name="Bus4")
b5 = pp.create_bus(net2, vn_kv=vnom_mv, name="Bus5")
b6 = pp.create_bus(net2, vn_kv=vnom_lv, name="Bus6")
bX = pp.create_bus(net2, vn_kv=vnom_hv, name="BusX")
pp.create_ext_grid(net2, bus=b4, vm_pu=1.1, name="Feeder2")
pp.create_load(net2, bus=b5, p_kw=25000., q_kvar=10000., name="Load5")
pp.create_load(net2, bus=b6, p_kw=25000., q_kvar=10000., name="Load6")

# conversion of input params
vkh_p = vsc_hv_percent
vkm_p = vsc_mv_percent * sn_hv_kva / sn_mv_kva
vkl_p = vsc_lv_percent * sn_hv_kva / sn_lv_kva
vkt1_p = 0.5 * (vkh_p + vkl_p - vkm_p)
vkt2_p = 0.5 * (vkm_p + vkh_p - vkl_p)
vkt3_p = 0.5 * (vkm_p + vkl_p - vkh_p)
vkt1 = vkt1_p
vkt2 = vkt2_p * sn_mv_kva / sn_hv_kva
vkt3 = vkt3_p * sn_lv_kva / sn_hv_kva

vrh_p = vscr_hv_percent
vrm_p = vscr_mv_percent * sn_hv_kva / sn_mv_kva
vrl_p = vscr_lv_percent * sn_hv_kva / sn_lv_kva
vrt1_p = 0.5 * (vrh_p + vrl_p - vrm_p)
vrt2_p = 0.5 * (vrm_p + vrh_p - vrl_p)
vrt3_p = 0.5 * (vrm_p + vrl_p - vrh_p)
vrt1 = vrt1_p
vrt2 = vrt2_p * sn_mv_kva / sn_hv_kva
vrt3 = vrt3_p * sn_lv_kva / sn_hv_kva

pp.create_transformer_from_parameters(
        net2, hv_bus=b4, lv_bus=bX, sn_kva=sn_hv_kva,
        vn_hv_kv=vn_hv_kv, vn_lv_kv=vn_hv_kv,
        vscr_percent=vrt1, vsc_percent=vkt1,
        pfe_kw=pfe, i0_percent=i0)
pp.create_transformer_from_parameters(
        net2, hv_bus=bX, lv_bus=b5, sn_kva=sn_mv_kva,
        vn_hv_kv=vn_hv_kv, vn_lv_kv=vn_mv_kv,
        vscr_percent=vrt2, vsc_percent=vkt2,
        pfe_kw=0., i0_percent=0.)
pp.create_transformer_from_parameters(
        net2, hv_bus=bX, lv_bus=b6, sn_kva=sn_lv_kva,
        vn_hv_kv=vn_hv_kv, vn_lv_kv=vn_lv_kv,
        vscr_percent=vrt3, vsc_percent=vkt3,
        pfe_kw=0., i0_percent=0.)

pp.runpp(net2)

print('\n 3W trafo:')
print(net.res_bus)
print('\n 3 2W trafos:')
print(net2.res_bus)

```

lthurner, 2019-03-14:
Thanks for bringing this up.  You are right, the equations in the documentation are not quite correct. I think we changed the method at some point without adapting the documentation. This should actually give you the right values:

>vkh_p = vsc_hv_percent * sn_hv_mva / min(sn_hv_kva, sn_mv_kva)
vkm_p = vsc_mv_percent * sn_hv_mva / min(sn_mv_kva, sn_lv_kva)
vkl_p = vsc_lv_percent * sn_hv_mva / min(sn_hv_kva, sn_lv_kva)
vkt1_p = 0.5 * (vkh_p + vkl_p - vkm_p)
vkt2_p = 0.5 * (vkm_p + vkh_p - vkl_p)
vkt3_p = 0.5 * (vkm_p + vkl_p - vkh_p)
vkt1 = vkt1_p
vkt2 = vkt2_p * sn_mv_kva / sn_hv_kva
vkt3 = vkt3_p * sn_lv_kva / sn_hv_kva

For comparison, you can also check the values that are calculated by pandapower internally in the conversion like this:

```
from pandapower.build_branch import _trafo_df_from_trafo3w
import pandas as pd
df = pd.DataFrame(_trafo_df_from_trafo3w(net))
```
I will fix this in the documentation for the next version of pandapower.

lthurner, 2019-03-14:
I now changed the formulas and also added some more information to the documentation: https://pandapower.readthedocs.io/en/develop/elements/trafo3w.html#electric-model

> So to me the description in the documentation looks fine, but the implementation of the three winding transformer in the code must be different somehow. I understand that Pandapower is tested against Powerfactory, so I am a bit puzzled by this.

I believe [this note](https://pandapower.readthedocs.io/en/develop/elements/trafo3w.html#electric-model) given in the documentation is relevant here:

> All short circuit voltages are given relative to the maximum apparent power flow. For example vk_hv_percent is the short circuit voltage from the high to the medium level, it is given relative to the minimum of the rated apparent power in high and medium level: min(sn_hv_mva, sn_mv_mva). This is consistent with most commercial network calculation software (e.g. PowerFactory). Some tools (like PSS Sincal) however define all short circuit voltages relative to the overall rated apparent power of the transformer: max(sn_hv_mva, sn_mv_mva, sn_lv_mva). You might have to convert the values depending on how the short-circuit voltages are defined.

There are just two ways to define the short-circuit voltages: relative or absolute. We chose to follow the relative notation, and I suspect NEPLAN follows the absolute notation (short-circuit voltages relative to the maximum rated power instead of to the maximum flow through the transformer). So when converting a NEPLAN network to pandapower you would actually have to convert the short-circuit voltages accordingly.

closed 2019-03-14 by lthurner