e2nIEE/pandapower #315: Data type of pwl cost points

friederikemeier, 2019-02-28:
Hi,

As it is implemented now, the pwl cost points are a tuple.
But: 'tuple' does not support item assignment!

When migrating my code, where I am changing the slope of pwl costs, I now have to convert the tuple to a list, do stuff and then convert it back to a tuple.

Could we consider changing this?
Before we had an array of lists, where item assignment was no problem. I don't know, what's the ideal way, but using a tuple is inefficient for my use case.

lthurner, 2019-03-02:
Yeah you are right tuples are immutable, which makes it kind of unpractical to change the costs. I don't see why it couldn't be a list of lists.

lthurner, 2019-03-14:
So we just have to change the documentation for this, right? Already seems to work with lists as it is.

closed 2019-03-14 by lthurner