e2nIEE/pandapower #327: pandapower to_excel

heinlmic, 2019-03-12:
Hello,
I have problem with network data exporting to excel file.

```
import os
import pandapower as pp
import pandapower.networks as nw
import pandapower.shortcircuit as sc

net = nw.simple_mv_open_ring_net()

pp.to_excel(net,r'c:\Users\heinlmic\Pracovni\python\LF\net.xlsx')
```
but after that I get error.

```
Traceback (most recent call last):
  File "C:\Users\heinlmic\Pracovni\PyCharmPortable\App\PyCharm\helpers\pydev\pydevd.py", line 1596, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Users\heinlmic\Pracovni\PyCharmPortable\App\PyCharm\helpers\pydev\pydevd.py", line 974, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:/Users/heinlmic/Pracovni/python/LF/pandapower_calc.py", line 8, in <module>
    pp.to_excel(net,r'c:\Users\heinlmic\Pracovni\python\LF\net.xlsx')
  File "C:\python27\lib\site-packages\pandapower\file_io.py", line 93, in to_excel
    dict_net = to_dict_of_dfs(net, include_results=include_results)
  File "C:\python27\lib\site-packages\pandapower\io_utils.py", line 93, in to_dict_of_dfs
    geo = coords_to_df(value, geotype="bus")
  File "C:\python27\lib\site-packages\pandapower\io_utils.py", line 58, in coords_to_df
    coords = value.at[i, 'coords']
  File "C:\python27\lib\site-packages\pandas\core\indexing.py", line 2270, in __getitem__
    return self.obj._get_value(*key, takeable=self._takeable)
  File "C:\python27\lib\site-packages\pandas\core\frame.py", line 2767, in _get_value
    series = self._get_item_cache(col)
  File "C:\python27\lib\site-packages\pandas\core\generic.py", line 3061, in _get_item_cache
    values = self._data.get(item)
  File "C:\python27\lib\site-packages\pandas\core\internals\managers.py", line 941, in get
    loc = self.items.get_loc(item)
  File "C:\python27\lib\site-packages\pandas\core\indexes\base.py", line 2658, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas\_libs\index.pyx", line 108, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\index.pyx", line 132, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\hashtable_class_helper.pxi", line 1601, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas\_libs\hashtable_class_helper.pxi", line 1608, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'coords'
```
Can you give me advice, how can I export network to excel?

I am using python 2.7 and 3.6, same error and pandapower 1.6.1.

Thanks for advice

lthurner, 2019-03-12:
The problem here is that in simple_mv_open_ring_net() is overwritten with a dataframe in an old format:
https://github.com/e2nIEE/pandapower/blob/8bdd475747cb8bab247a89cfc495f7a90cb670f6/pandapower/networks/simple_pandapower_test_networks.py#L212-L213
This will be fixed in the next version. For now, this should work:

```
import os
import pandapower as pp
import pandapower.networks as nw
net = nw.simple_mv_open_ring_net()
pp.convert_format(net)
pp.to_excel(net,r'c:\Users\heinlmic\Pracovni\python\LF\net.xlsx')
```

lthurner, 2019-03-12:
You are welcome. Fixed in #328

closed 2019-03-12 by lthurner